stitchEm/stitchEm #4: Support Homebrew env for macOS build

w-m, 2019-04-23:
Travis CI is using Homebrew in their macOS environments: https://docs.travis-ci.com/user/reference/osx/

Our CMake files currently only deal with a MacPorts environment. Shouldn't be too much work to also support homebrew, but I'm need to figure out how I can test both on the same dev machine.

jeremad, 2019-04-24:
Oh I think I can do that, I at least have a look

w-m, 2019-04-24:
Biggest difference: the `APPLE_MACPORTS` option in CMake currently adds lookup in /opt/local, whereas homebrew puts the files in /usr/local, IIRC.

jeremad, 2019-04-24:
Ah... I misread the description, trickier than I thought. I still can have a look though

jeremad, 2019-04-24:
Shouldn't we just drop external deps for linux and mac ?

jeremad, 2019-04-24:
or may be use a CPP package manager like conan. But I'm not sure every lib we use has a receipe

w-m, 2019-04-24:
> Shouldn't we just drop external deps for linux and mac ?

Yes. The `APPLE_MACPORTS` option was just a crutch to migrate away from external deps, when I wasn't sure whether it would work immediately everywhere.

For macOS and Linux we should use the official CMake find modules, wherever possible. If there's no module, there should be the possibility of adding `/opt/local` and `/usr/local` globally to the CMake search paths, so the paths don't have to be repeated in every find incarnation.

In the end CMake should find everything on its own without setting `APPLE_MACPORTS` or `LINUX_PKG`, but the user should still have the opportunity to override the package location with a manual option. I'm not sure what the current CMake best practices say how to do this, need to look at other projects.

As long as we don't have anything else on Windows, I'd like to keep the external_deps system working there.

w-m, 2019-04-24:
> or may be use a CPP package manager like conan. But I'm not sure every lib we use has a receipe

For macOS and Linux I'd try to keep using the packages from the system package manager for simplicity reasons, wherever it makes sense.

To get a Windows build working conan is definitely an option. When I looked at that earlier I was disappointed to not find ffmpeg in the conan center (which is a huge pain point), but there seem to be some 3rd party conan ffmpeg packages on GitHub, maybe they'll work.

PR #9 (2019-04-26): Add MAC CI
## Summary
Make the codebase compile with homebrew packages, because travis infrastructure uses this

Closes: #4 #7
closed 2019-04-26 by jeremad