stitchEm/stitchEm #16: WIN10 installation?

virtualguide, 2019-04-28:
Hi! I found the link to stitchEm in FB group of 360° videographers, but no one know how to compile it and make running on WIN10. Any hint? Thank you in advance

w-m, 2019-04-28:
Hi, thanks for your interest in stitchEm. We are currently missing a process to install the dependencies (3rd party software) on Windows, which are required to get a working build. It's not as straightforward as on Linux and macOS, so getting this to work may take some time.

I'm tracking the technical details in issue #17 and will update you here when we get to a full Windows compilation.

virtualguide, 2019-04-28:
Thank you VERY MUCH for prompt answer!

Alnondo, 2019-05-14:
Is there an update on a windows 10 installer?

w-m, 2019-05-14:
We're working on it, but we are still far away from a full Windows build.

Help by Windows devs would be very welcome, especially around windows dependencies. If you know your way around chocolatey/conan package manager, installing and using packages like ffmpeg, OpenCV, ceres, get in touch on issue #17!

Otherwise you can subscribe to this issue, I will put updates here when we make significant progress for the Windows platform.

Alnondo, 2019-05-15:
Way above my head unfortunately. Thank you for you work 👍

Alnondo, 2019-07-08:
Hey :) I was wondering if there was an update regarding this thread yet? 
Regards, Alan.

jeremad, 2019-07-09:
Hello, we are working on it. However, I cannot provide any timetable at the moment. Any contributions on that is welcome!

Alnondo, 2019-07-09:
Thank you jeremad, I am currently learning C#. If I can help I will. Any area you think I would be of assistance?

doymcc, 2019-09-13:
Good day Jeremad,

I can usually compile from make files and the like and it seems you were getting close on a windows 10 version. Any thing come up for crossing the Jul 9 comment till now Sep. 13.

Regards,

doymcc

jeremad, 2019-09-25:
Hi,

This is an open source project with absolutely no funding, we are working on this on our free time. lately we all have been busy. The annual gathering of the maintainers will take place in a couple of weeks. We will come up with a plan to achieve a windows build ASAP, since we know this is really missing.

Thank you a lot for your interest in the project

Alnondo, 2019-09-25:
Thank you Jeremad :) This will help me and others a great deal. Solidarity from the plebs brother

doymcc, 2019-09-26:
Dear Jeremad,

 

I’m working with my limited understanding to approach a parallel path for getting the code to compile.

 

Regards,

 

Doyle

w-m, 2019-09-26:
@doymcc feel free to reach out with any questions, even silly ones, we're happy to support anyone working on this.

As written in https://github.com/stitchEm/stitchEm/issues/17#issuecomment-528426199, I think it could be worthwhile to try out vcpkg on Windows for the dependencies. This seems to have the important packages that were blockers until now with the other approaches.

KaptainKCG, 2019-10-15:
I'm keeping an eye on this feed. Dying to get back to using "Video Stitch" on PC again.

solosails, 2019-11-18:
Hi, Also keen to install on Windows 10 - thanks for the work so far!

solosails, 2019-12-08:
Has anyone managed to compile a Windows installer exe? I have been trying for weeks to make one, but not being very codey, it's proving quite tricky! Thanks in advance.

w-m, 2019-12-08:
We've gotten much closer, it is now possible to build on Windows, the steps needed are documented here: https://github.com/stitchEm/stitchEm/blob/master/doc/BUILD.md#building-on-windows

We haven't created an installer yet, as stitching always fails due to a CUDA error (#49), on both of our Windows setups.

solosails, 2019-12-09:
> We've gotten much closer, it is now possible to build on Windows, the steps needed are documented here: https://github.com/stitchEm/stitchEm/blob/master/doc/BUILD.md#building-on-windows
>
> We haven't [...]

Hi WM, 

Thanks for the update. OK, would you like me to test compile on my Win10 machine and see if I get the same error?

w-m, 2019-12-09:
Thanks for the offer, but there's probably not much we could learn from it, as it's likely to throw the same nondescript CUDA error without any further information on where the issue lies. It will rather need some local debugging with Nvidia tools.

solosails, 2019-12-09:
Hi, OK, understood - that would be way beyond me anyway! 

Thanks for developing this again!

cansolo, 2019-12-13:
Hi.

I've been trying to build it in win10(x64) for 3 days and want to share my latest progress

So I first of all I'm using cmake and other components for the first time.

I'm using:

VS 2017 with all compiling, debugging and cpp dependencies (including VS 2015 toolkit)
CUDA 8.0
GnuWin32
LLVM
Perl Strawberry
OpenSSL

Here is the result of trying to compile with MSVC 14.16.27023 Hostx64\x86

[image]

And for finding mspdbcore.dll, trying to compile with MSVC 14.16.27023 Hostx64\x64

[image]

Had same results while trying to compile with my all other MSVC versions and Hostx86.

So, also added my cmakecache file, because I edit it a bit for surpass my earlier problems.

[CMakeCache.txt](https://github.com/stitchEm/stitchEm/files/3959643/CMakeCache.txt)

Hoping for advices!

jeremad, 2019-12-14:
it seems CUDA 8.0 is incompatible with VS 2017, can you try with CUDA 10?
https://stackoverflow.com/questions/43745099/using-cuda-with-visual-studio-2017

jeremad, 2019-12-14:
You cannot compile for x86 unfortunately!

cansolo, 2019-12-16:
Unfortunately here is the result of when I try with cuda 10.2. So Actually I couldn't manage to converting 10.2. Can I get any help, because I'm not familiar to using cmake (I literally don't know what to do right now :( )

[image]

And here is my cmake cache file 

[CMakeCache.txt](https://github.com/stitchEm/stitchEm/files/3967259/CMakeCache.txt)

w-m, 2019-12-19:
It might still be using the nvcc from CUDA 8.0 due to the hardcoded value in the [cuda.version file](https://github.com/stitchEm/stitchEm/blob/master/cuda.version). Change this to 10.2, remove all CMake cache files and try again!

w-m, 2020-05-03:
The build instructions for the full Windows build are now found here:
https://github.com/stitchEm/stitchEm/blob/master/doc/BUILD.md#building-on-windows

And pre-built binaries can be downloaded here: https://github.com/stitchEm/stitchEm/releases/latest

closed 2020-05-03 by w-m