stitchEm/stitchEm #17: Automatically build & install Windows deps

w-m, 2019-04-28:
Windows doesn't have a system-wide package manager for the installation of the library/build dependencies. At VideoStitch, the required packages were manually built and distributed as binaries, a process which we can't replicate for the open source project.

This ticket tracks the progress of automatically installing the required deps on a Windows 10 environment.

The ultimate goal is to have an automatic Windows build. Perhaps on AppVeyor?

https://www.appveyor.com/docs/windows-images-software/

deps.json       | State  |
|---------------|--------|
vlc             |        |
ffmpeg          |        |
portaudio       |        |
lame            |        |
x264            |        |
libtiff         |        |
libpng          |        |
opencv2         |        |
aja             |        |
ladybug         |        |
magewell        |        |
magewellPro     |        |
teledyne        |        |
Ximea           |        |
yuan            |        |
OculusSDK       |        |
GL              |        |
librtmp         |        |
libjpeg         |        |
decklink        |        |
ptgrey          |        |
Sapera          |        |
QCAP            |        |
unikey          |        |
nginx_rtmp      |        |
CL              |        |
clangcl         |        |
openvr          |        |
Intel_Media_SDK |        |
eigen           |        |
ceres           |        |
faac            |        |
faad            |        |
gflags          |        |
jsoncpp         |        |
googleapis      |        |
json            |        |
glog            |        |
curl            |        |
openssl         |        |
youtubeapi      |        |
nvenc           |        |
v4l2            |        |
r8b             |        |

Additionally for the build:

| Package          | State |
|------------------|-------|
Visual Studio 2013 |       |
Qt                 |       |
CUDA               |       |
Python 2.7         |       |
CMake              |       |
Ninja              |       |
Flex               |       |
bison              |       |
Swig               |       |
Doxygen            |       |

w-m, 2019-04-28:
After conan.io, another package manager option for Windows seems to be chocolatey. Referenced in https://help.appveyor.com/discussions/questions/629-opencv-is-it-possible-to-install-opencv-in-appveyor, it comes pre-installed on the AppVeyor Images and has at least OpenCV.

jeremad, 2019-04-29:
There is now a conan receipe for OpenCV. I'll start experimenting with conan someday!
https://github.com/conan-community/conan-opencv

klem23, 2019-04-29:
Chocolatey seems to be the most similar to classic package manager (yum, apt) and it has many of the open source deps we have (qt, rtmp, opencv, ffmpeg), but there is still a lot missing.

Also there is not a pre-installed compiler on windows for fetching gihtub's sources for user that just interested in the runtime.

w-m, 2019-04-30:
> Also there is not a pre-installed compiler on windows for fetching gihtub's sources for user that just interested in the runtime.

What do you mean by runtime?

klem23, 2019-04-30:
Windows user that just want to install the software and run/use it (like many windows app), no compilation, visual etc

w-m, 2019-05-14:
Started the work on the [https://github.com/stitchEm/stitchEm/tree/w-m/windows](w-m/windows) branch.

Using Chocolatey and conan to install the packages, to get the library to build first. Later I/O plugins can be added, then the Qt apps, both of which independently depend on the lib.

I had most success with Visual Studio 2013 so far. (Couldn't get 2015 to work, 2017 has std::map related compilation errors, 2019 may be too bleeding edge for Chocolatey).

w-m, 2019-09-05:
I was recently made aware of https://github.com/microsoft/vcpkg. There are ports for ceres, Eigen, ffmpeg & OpenCV!

jeremad, 2019-10-09:
I made some progress on that last night. I think vcpkg misses a couple of lib like faad and faac, I did not find x264 either

jeremad, 2019-10-12:
I'm almost done find all libraries. I still misses the AJA SDK, it seems that I cannot download it for free
libfaac, libmfxhw64, libmpghip, libvfaad. any idea on this @w-m @klem23

w-m, 2019-10-12:
For libfaac I think we'll need to `wget` the library and sources from http://faac.sourceforge.net, then unzip them. We had version 1.28 in deps.json, so trying the current 1.29.9.2 will hopefully work.

(Even though you get to a weird ad page through the links when using a browser, the forwarding properly works and wget can be used for the download).

I'm unfamiliar with the other libraries. Did you install ffmpeg with all the flags (nonfree stuff and so on?).

jeremad, 2019-10-12:
I installed ffmpeg from vcpkg, without any options, may be I need to use a few flags, I'll look into that

w-m, 2019-10-12:
I saw this https://github.com/microsoft/vcpkg/blob/master/ports/ffmpeg/CONTROL but I don't know how to apply it to select these features.

jeremad, 2019-10-12:
I'm really close for the lib. One major road block:
```
1>------ Build started: Project: vslib-objects-cuda, Configuration: Debug x64 ------
1>readerController.cpp
1>D:\dev\stitchEm\lib\src\core\readerController.cpp(654): warning C4456: declaration of 'source' hides previous local declaration
1>D:\dev\stitchEm\lib\src\core\readerController.cpp(638): note: see declaration of 'source'
1>D:\dev\stitchEm\lib\src\core\readerController.cpp(666): warning C4456: declaration of 'source' hides previous local declaration
1>D:\dev\stitchEm\lib\src\core\readerController.cpp(638): note: see declaration of 'source'
1>D:\dev\Visual Studio 2017\VC\Tools\MSVC\14.16.27023\include\xmemory0(881): error C2280: 'std::pair<const _Kty,_Ty>::pair(const std::pair<const _Kty,_Ty> &)': attempting to reference a deleted function
1>        with
1>        [
1>            _Kty=readerid_t,
1>            _Ty=VideoStitch::Audio::Samples
1>        ]
1>D:\dev\Visual Studio 2017\VC\Tools\MSVC\14.16.27023\include\utility(194): note: see declaration of 'std::pair<const _Kty,_Ty>::pair'
1>        with
1>        [
1>            _Kty=readerid_t,
1>            _Ty=VideoStitch::Audio::Samples
1>        ]
1>D:\dev\Visual Studio 2017\VC\Tools\MSVC\14.16.27023\include\utility(194): note: 'std::pair<const _Kty,_Ty>::pair(const std::pair<const _Kty,_Ty> &)': function was implicitly deleted because a data member invokes a deleted or inaccessible function 'VideoStitch::Audio::Samples::Samples(const VideoStitch::Audio::Samples &)'
1>        with
1>        [
1>            _Kty=readerid_t,
1>            _Ty=VideoStitch::Audio::Samples
1>        ]
1>d:\dev\stitchem\lib\include\libvideostitch\audio.hpp(335): note: 'VideoStitch::Audio::Samples::Samples(const VideoStitch::Audio::Samples &)': function was explicitly deleted
1>D:\dev\Visual Studio 2017\VC\Tools\MSVC\14.16.27023\include\xtree(984): note: see reference to function template instantiation 'void std::_Default_allocator_traits<_Alloc>::construct<_Ty,_Value_type&>(_Alloc &,_Objty *const ,_Value_type &)' being compiled
1>        with
1>        [
1>            _Alloc=std::allocator<std::_Tree_node<std::pair<const readerid_t,VideoStitch::Audio::Samples>,std::_Default_allocator_traits<std::allocator<std::pair<const readerid_t,VideoStitch::Audio::Samples>>>::void_pointer>>,
1>            _Ty=std::pair<const readerid_t,VideoStitch::Audio::Samples>,
1>            _Value_type=std::pair<const readerid_t,VideoStitch::Audio::Samples>,
1>            _Objty=std::pair<const readerid_t,VideoStitch::Audio::Samples>
1>        ]
1>D:\dev\Visual Studio 2017\VC\Tools\MSVC\14.16.27023\include\xtree(983): note: see reference to function template instantiation 'void std::_Default_allocator_traits<_Alloc>::construct<_Ty,_Value_type&>(_Alloc &,_Objty *const ,_Value_type &)' being compiled
1>        with
1>        [
1>            _Alloc=std::allocator<std::_Tree_node<std::pair<const readerid_t,VideoStitch::Audio::Samples>,std::_Default_allocator_traits<std::allocator<std::pair<const readerid_t,VideoStitch::Audio::Samples>>>::void_pointer>>,
1>            _Ty=std::pair<const readerid_t,VideoStitch::Audio::Samples>,
1>            _Value_type=std::pair<const readerid_t,VideoStitch::Audio::Samples>,
1>            _Objty=std::pair<const readerid_t,VideoStitch::Audio::Samples>
1>        ]
1>D:\dev\Visual Studio 2017\VC\Tools\MSVC\14.16.27023\include\xtree(1917): note: see reference to function template instantiation 'std::_Tree_node<_Ty,std::_Default_allocator_traits<_Alloc>::void_pointer> *std::_Tree_comp_alloc<_Traits>::_Buynode<_Ty&>(_Ty &)' being compiled
1>        with
1>        [
1>            _Ty=std::pair<const readerid_t,VideoStitch::Audio::Samples>,
1>            _Alloc=std::allocator<std::pair<const readerid_t,VideoStitch::Audio::Samples>>,
1>            _Traits=std::_Tmap_traits<readerid_t,VideoStitch::Audio::Samples,std::less<int>,std::allocator<std::pair<const readerid_t,VideoStitch::Audio::Samples>>,false>
1>        ]
1>D:\dev\Visual Studio 2017\VC\Tools\MSVC\14.16.27023\include\xtree(1917): note: see reference to function template instantiation 'std::_Tree_node<_Ty,std::_Default_allocator_traits<_Alloc>::void_pointer> *std::_Tree_comp_alloc<_Traits>::_Buynode<_Ty&>(_Ty &)' being compiled
1>        with
1>        [
1>            _Ty=std::pair<const readerid_t,VideoStitch::Audio::Samples>,
1>            _Alloc=std::allocator<std::pair<const readerid_t,VideoStitch::Audio::Samples>>,
1>            _Traits=std::_Tmap_traits<readerid_t,VideoStitch::Audio::Samples,std::less<int>,std::allocator<std::pair<const readerid_t,VideoStitch::Audio::Samples>>,false>
1>        ]
1>D:\dev\Visual Studio 2017\VC\Tools\MSVC\14.16.27023\include\xtree(1943): note: see reference to function template instantiation 'std::_Tree_node<_Ty,std::_Default_allocator_traits<_Alloc>::void_pointer> *std::_Tree<std::_Tmap_traits<_Kty,VideoStitch::Audio::Samples,_Pr,_Alloc,false>>::_Copy_or_move<_Value_type,std::integral_constant<bool,false>>(_Ty &,std::_Tree<std::_Tmap_traits<_Kty,VideoStitch::Audio::Samples,_Pr,_Alloc,false>>::_Copy_tag,_Is_set)' being compiled
1>        with
1>        [
1>            _Ty=std::pair<const readerid_t,VideoStitch::Audio::Samples>,
1>            _Alloc=std::allocator<std::pair<const readerid_t,VideoStitch::Audio::Samples>>,
1>            _Kty=readerid_t,
1>            _Pr=std::less<int>,
1>            _Value_type=std::pair<const readerid_t,VideoStitch::Audio::Samples>,
[... cut ...]
1>            _Kty=readerid_t,
1>            _Ty=VideoStitch::Audio::Samples
1>        ]
1>Done building project "vslib-objects-cuda.vcxproj" -- FAILED.
2>------ Build started: Project: libvideostitch_cuda, Configuration: Debug x64 ------
2>LINK : fatal error LNK1104: cannot open file 'D:\dev\build\lib\vslib-objects-cuda.dir\Debug\readerController.obj'
2>Done building project "libvideostitch_cuda.vcxproj" -- FAILED.
========== Build: 0 succeeded, 2 failed, 7 up-to-date, 0 skipped ==========
```

w-m, 2019-10-13:
Could you try this commit: 11ec46e1234941c7135fa874923b047d8b854e7d

I later undid it because I think it didn't work, but please try this again.

The issue is that we have a move-only object (Audio::Samples) in a std::map, and the MSVC handling of that changed between MSVC 2015 and 2017.

See e.g.

https://stackoverflow.com/questions/47604029/move-constructors-of-stl-containers-in-msvc-2017-are-not-marked-as-noexcept

jeremad, 2019-10-13:
OK I'll try that!I managed to work around that one with obviously wrong code.
The last road block I think is that the linker tells me it does not find the GLEW symbols, which is really weird, everything look fine in the configuration. my branch need a little cleaning, but overall it's promising

jeremad, 2019-10-13:
> Could you try this commit: [11ec46e](https://github.com/stitchEm/stitchEm/commit/11ec46e1234941c7135fa874923b047d8b854e7d)
> 
> I later undid it because I think it didn't work, but please try this again.
> 
> The issue is that we have a move-only object (Audio::Samples) in a std::map, and the MSVC handling of that changed between MSVC 2015 and 2017.
> 
> See e.g.
> 
> https://stackoverflow.com/questions/47604029/move-constructors-of-stl-containers-in-msvc-2017-are-not-marked-as-noexcept

Still not compili,g :(

jeremad, 2019-10-13:
> OK I'll try that!I managed to work around that one with obviously wrong code.
> The last road block I think is that the linker tells me it does not find the GLEW symbols, which is really weird, everything look fine in the configuration. my branch need a little cleaning, but overall it's promising

OK, I actually needed the static glew library. This is fixed then

w-m, 2019-10-13:
> Still not compiling :(

I've read through some of the MS postings and couldn't really figure out whether they were just saying it's broken and to file a bug report or whether it can be worked around, or whether a later MSVC version fixes this.

Could you try whether this snippet compiles with your compiler? It does with the earliest MSVC version in godbolt:

https://godbolt.org/z/Jr3GMa

jeremad, 2019-10-13:
> I've read through some of the MS postings and couldn't really figure out whether they were just saying it's broken and to file a bug report or whether it can be worked around, or whether a later [...]

This builds alright

jeremad, 2019-10-13:
Everything I've done is in here. The audio issue is the last thing to fix before a full lib compilation I think
https://github.com/stitchEm/stitchEm/tree/jrm/win

w-m, 2019-10-14:
> This builds alright

So there's hope! As I still don't understand the issue properly, the only way forward I currently see is to manually dissect the changes between the godboldt example and our Audio::Samples code.

Starting by putting the map code in ReaderController and the MoveOnly struct in audio.hpp, as a class VS_EXPORT. Adding back the Samples implementation until the compilation breaks.

w-m, 2020-05-02:
Fixed in https://github.com/stitchEm/stitchEm/pull/43.

closed 2020-05-02 by w-m