stitchEm/stitchEm #24: Compilation in macos with backend OPENCL asks for CUDA

afrhu, 2019-05-20:
Hello, I want to compile the project for mac using backend OPENCL. 
I used macports to install the dependencies and compiled it with GPU_BACKEND_CUDA=OFF, GPU_BACKEND_OPENCL=ON, and GPU_BACKEND_DEFAULT=OPENCL using cmake-gui. 
It compiles with no errors but when I try to run videostitch-gui I get the message "Your CUDA driver appears to be outdated" and then it closes, the same happens with vahana vr.

cvaldesmol, 2019-05-21:
I have issues as well. Since MAC is dropping support for Nvidia I think by default it should be set to compile for OpenCL

w-m, 2019-05-21:
It's hard to reproduce your build settings if you're using cmake-gui. Please try with a command-line build, then I know better what's happening on your system.

Try these steps:
- clone stitchEm
- `mkdir build && cd build`
- `cmake -DGPU_BACKEND_CUDA=OFF -DGPU_BACKEND_OPENCL=ON -DMACPORTS=ON -DQt5_DIR=~/Qt/5.9.7/clang_64/lib/cmake/Qt5 -DCMAKE_BUILD_TYPE=Release -G Ninja ../stitchEm` (set your own Qt path)
- `ninja`

If the build is successful, `cd bin/x64/release` and `./Studio/videostitch-studio` to launch the app.

afrhu, 2019-05-21:
> It's hard to reproduce your build settings if you're using cmake-gui. Please try with a command-line build, then I know better what's happening on your system.
>
> Try these steps:
>
> * clone stitchEm
> * [...]

I tried with: 
`cmake -DGPU_BACKEND_CUDA=OFF -DGPU_BACKEND_OPENCL=ON -DMACPORTS=ON -DDISABLE_RTMP=ON -DQt5_DIR=/Users/videost/Qt5.12.3/5.12.3/clang_64/lib/cmake/Qt5 -DCMAKE_BUILD_TYPE=Release -G Ninja ../stitchEm`

Then I compile using `ninja`. 
But when I run `./Studio/videostitch-studio` I get the message:
"No CUDA capable GPU detected on your system" and it closes.

w-m, 2019-05-21:
That's strange, this line should not be produced in an OpenCL-only build. 

Was the build folder empty when you ran the lines above? After a clean OpenCL-only build there should be a "libvideostitch.dylib" and a "libvideostitch_opencl.dylib" in the bin/x64/release folder, but *no* "libvideostitch_cuda.dylib". Please check for the existence of libvideostitch_cuda.dylib.

w-m, 2019-05-21:
Btw, what system are you running: what's your GPU and OS version?

Please run `bin/x64/release/videostitch-cmd -d` and paste the result here. It should print a list of the GPU devices discovered by the software.

afrhu, 2019-05-21:
I'm using Macos High Sierra version 10.13.6
Graphics NVIDIA GeForce 320M 256 MB (it's an old card)

Yes, it was a new build.
It generates libvideostitch_opencl.dylib but not libvideostich_cuda.dylib

It seem it's not recognizing any gpu. with `bin/x64/release/videostitch-cmd -d`:
`
Could not find any compatible GPU device on this computer.
Host OpenCL allocated memory:
Device OpenCL allocated memory:
`

w-m, 2019-05-21:
This setup won't work. The software requires at least 2 GB of GPU memory, as all the stitching is done on the GPU.

The GPU is not discovered as a compatible device in an OpenCL-only build because OpenCL stitching is not supported on Nvidia cards.

closed 2019-08-28 by w-m