stitchEm/stitchEm #27: Build Help

filmcafeSteve, 2019-08-21:
Hi all,
Sorry in advance for this but I can't build this. 
I was a VideoStitch studio customer and I'm just trying to pick up where I left off as a stitcher not a coder. I've installed the GUI version of CMAKE and tried to compile there but get errors building. 
```
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find PNG (missing: PNG_LIBRARY) (found version "1.4.12")
```
I appreciate there are numerous external dependencies but is there any prospect of this being packaged up for folk like me?

Many thanks and sorry for the low level skills issues!

Steve

w-m, 2019-08-21:
It looks like you're on a Mac. To have a full Mac build you would need to install MacPorts (https://www.macports.org/install.php), [Qt](https://www.qt.io/download) and then a couple of external libraries by running the commands documented [here](https://github.com/stitchEm/stitchEm/blob/master/doc/BUILD.md#using-macports) in a Terminal.

To build VideoStitch Studio, then run

```
cmake -DCREATE_BOX_PACKAGE=OFF \
      -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=ON \
      -DQt5_DIR=~/Qt/5.9.6/clang_64/lib/cmake/Qt5 \
      -DMACPORTS=ON \
      -G Ninja \
      -DCMAKE_BUILD_TYPE=Release \
      stitchEm
```

and finally `ninja` in a Terminal.

I've created a ticket to provide Mac binaries, #28. While progress on Windows has been stagnant unfortunately, I'll try to make progress on this in the coming weeks.

filmcafeSteve, 2019-08-21:
Thanks so much for your help. I've made progress! Installed MacProts, Qt, and all of the dependencies.

Running the cmake command I get this error:
`CMake Error: The source "/Users/steve/stitchEm/CMakeLists.txt" does not match the source "/Users/steve/Desktop/stitchEm/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.`

I've no doubt I'm just being dumb, but I can't make sense of what I'm not doing right to solve this. I've tried the stitchEm folder on Desktop only, in users/steve only and in both places. Everything throws up one error or another.

Any further help is greatly appreciated - especially since I feel like I've made progress!
Many thanks

filmcafeSteve, 2019-08-21:
Ok, a few more tweaks, now this error:

```
-- CMake Build Type -- Debug
CMake Error at CMakeLists.txt:216 (message):
  external_deps folder not found and no system package manager set
```

w-m, 2019-08-21:
The cacher error can be avoided by creating a build folder and doing all CMake operations in this build folder. If you do another build with another configuration, create a different build folder.

>   external_deps folder not found and no system package manager set

This error is only thrown if CMake detects the system to be Windows. Can you confirm you are on a Mac?

Please try using the command line (not the CMake GUI), with the exact CMake command given above, in a clean build folder.

filmcafeSteve, 2019-08-21:
Thanks again. I have been running the exact command
```
cmake -DCREATE_BOX_PACKAGE=OFF \
      -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=ON \
      -DQt5_DIR=~/Qt/5.9.6/clang_64/lib/cmake/Qt5 \
      -DMACPORTS=ON \
      -G Ninja \
      -DCMAKE_BUILD_TYPE=Release \
      stitchEm
```
copied and pasted, with a clean download of the source (0.0.9), and using terminal.

w-m, 2019-08-21:
Please paste the output from the command here.

filmcafeSteve, 2019-08-21:
```
Stevens-MacBook-Pro:~ steve$ cmake -DCREATE_BOX_PACKAGE=OFF       -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=ON       -DQt5_DIR=~/Qt/5.9.6/clang_64/lib/cmake/Qt5       -DMACPORTS=ON       -G Ninja       stitchEm
-- GPU_BACKEND_CUDA='ON'
-- GPU_BACKEND_OPENCL='ON'
-- Compiler caching enabled (ccache)
-- System: APPLE
-- CMake Build Type -- Debug
CMake Error at CMakeLists.txt:216 (message):
  external_deps folder not found and no system package manager set

-- Configuring incomplete, errors occurred!

```

w-m, 2019-08-21:
Please use the latest master version, which has a few more build fixes than 0.0.9: https://github.com/stitchEm/stitchEm/archive/master.zip

Also give `-DCMAKE_BUILD_TYPE=Release` to CMake, which I posted above (but it's not in the dev documentation, as you wouldn't use that to develop).

filmcafeSteve, 2019-08-21:
Ok, so more progress but no success, I'm afraid

```
CMake Error at lib/apple.cmake:9 (find_package):
  By not providing "FindCeres.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Ceres", but
  CMake did not find one.

  Could not find a package configuration file provided by "Ceres" with any of
  the following names:

    CeresConfig.cmake
    ceres-config.cmake

  Add the installation prefix of "Ceres" to CMAKE_PREFIX_PATH or set
  "Ceres_DIR" to a directory containing one of the above files.  If "Ceres"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  lib/CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
```

filmcafeSteve, 2019-08-21:
I installed Ceres Solver - http://ceres-solver.org/installation.html#mac-os-x using: sudo port install ceres-solver

That's moved me to the next error:

```
CMake Error at apps/CMakeLists.txt:21 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" (requested
  version 5.6) with any of the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GLFW
    linked by target "videostitch-cmd" in directory /Users/steve/stitchEm/lib/samples/sample-02-fullCmd
libfaac
    linked by target "rtmpPlugin" in directory /Users/steve/stitchEm/IO/src/rtmp
libfaad
    linked by target "rtmpPlugin" in directory /Users/steve/stitchEm/IO/src/rtmp
librtmp
    linked by target "rtmpPlugin" in directory /Users/steve/stitchEm/IO/src/rtmp

-- Configuring incomplete, errors occurred!
```

Qt was installed today

w-m, 2019-08-21:
Alright, getting closer.

Please note that there are two commands for things to install with MacPorts: https://github.com/stitchEm/stitchEm/blob/master/doc/BUILD.md#using-macports

If ceres-solver was missing, you may have missed the second one.

Please replace the Qt path to the Qt version in the CMake command ("~/Qt/5.9.6") with the one you installed today.

filmcafeSteve, 2019-08-21:
I ran the second dependencies PORT install. I've no idea why Ceres wasn't installed. ffmpeg was, which is later in the sequence. I've since re-installed qt5 with MacPorts and have hopped that error.  I think it boiled down to 'openal' not installing and there being some issue with it in its latest version.

I've now managed to install it and have reached this point in Terminal:
```
-- --- VideoStitch-lib dependencies ---
-- Found OpenEXR: /opt/local/include;/opt/local/include/openexr  
-- OpenEXR libs: /opt/local/lib/libHalf.dylib;/opt/local/lib/libIex.dylib;/opt/local/lib/libImath.dylib;/opt/local/lib/libIlmImf.dylib;/opt/local/lib/libIlmThread.dylib
-- OpenEXR include: /opt/local/include;/opt/local/include/openexr
-- PNG: /opt/local/lib/libpng.dylib
-- R8B: r8b
-- OpenCV libs: opencv_core;opencv_calib3d;opencv_features2d;opencv_imgproc;opencv_flann;opencv_video
-- CERES LIBS: general;ceres
-- 
-- Found Doxygen: /opt/local/bin/doxygen (found version "1.8.15") found components:  doxygen dot 
-- Found CUDA: /usr/local/cuda (found version "8.0") 
-- VideoStitch I/O libs configured to build: avPlugin;jpgPlugin;pamPlugin;pngPlugin;rawPlugin;tiffPlugin;exrPlugin;rtmpPlugin
-- Qt minimum supported version: 5.6
-- Qt FOUND: 5.12.4
-- Studio version: Studio-v0.0.9-24-g9000ec8-master.2019-05-21
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CREATE_BOX_PACKAGE

-- Build files have been written to: /Users/steve
```

I have no .app or anything that I can seem to run. What should I have to show for myself?!

w-m, 2019-08-21:
This is just the configuration for the build. Now that it's complete run `ninja` in the directory to actually perform building the app.

If this finishes without error, you'll then have a binary that you can run: bin/x64/release/Studio/videostitch-studio

It's missing some of the niceties of an app package (icons, proper name, portability to other machines), but otherwise it should run just the same. Watch progress on #28 for app package.

filmcafeSteve, 2019-08-21:
```
144 warnings generated.
ninja: build stopped: subcommand failed.
```

:( Thanks so much for your help and patience. I will indeed wait for #28 

Thanks for what you're doing. I bought Studio and I can't re-license it on a new machine.

Steve

w-m, 2019-08-21:
Run ninja again, paste the output. Need the line that contains an error message (the warnings can be ignored, although it's strange, there shouldn't be any).

filmcafeSteve, 2019-08-21:
```
1 error generated.
CMake Error at cuda_compile_1_generated_cutransform.cu.o.Release.cmake:219 (message):
  Error generating
  /Users/steve/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/core1/./cuda_compile_1_generated_cutransform.cu.o
```

w-m, 2019-08-21:
Unfortunately this is not enough information or any known issue.

What kind of GPU are you using? If you're *not* using Nvidia graphics, disable CUDA support by deleting everything in the build folder, then running CMake again, but replace `-DGPU_BACKEND_CUDA=ON` with `-DGPU_BACKEND_CUDA=OFF`.

filmcafeSteve, 2019-08-21:
Ok, I've rerun everything with cmake with CUDA OFF, and built again with ninja and we have a winner! 

So much good stuff here!! Sadly it's crashing at 99% when I try to stitch. That's going to be a whole other thing though!

Thank you so much for your help and amazing patience! I'll see what I can achieve with a feww workflow tweaks

w-m, 2019-08-21:
I tried on a current MacBook and was able to reproduce a crash. I fixed it in a git branch, you can download the updated source here: https://github.com/stitchEm/stitchEm/archive/mask-merger-destructor.zip

While running Studio I noticed a different issue with OpenCL on this machine. I was still able to stitch but it could cause problems with other features, better fix it too. Please do the following:

- delete the folder "~/Library/Application Support/VideoStitch/cache" (Cmd+Shift+G in Finder to navigate there, maybe hidden)
- add "-DDISABLE_OPENCL_SPIR=ON" to the CMake command

filmcafeSteve, 2019-08-27:
Wonderful! I can stitch! Sadly though, I cannot export!

Application Process window export log shows:

```
Process is starting.
Process isn't running.
Process failed to start.
```

Application log shows

```
[2019/08/27 21:20:34] Missing field 'rig' for BaseProjectDefinition.
[2019/08/27 21:20:34] [Status] [audio pipeline configuration] Invalid configuration: No audio processor definition found
[2019/08/27 21:20:34] [ambisonic] Decoder bad coefficients given for stereo
[2019/08/27 21:20:35] [AudioPlayer] Audio data is not being fed to the audio device at a fast enough rate, cannot play audio.
[2019/08/27 21:21:25] [Status] [audio pipeline configuration] Invalid configuration: No audio input found
```

The same result if I use either or no audio in the stitch, and AAC or MP3.

Sorry! And thanks again for your patience and help

w-m, 2019-08-27:
Go to the `bin/x64/release` folder and copy the file called `videostitch-cmd` to the `Studio` folder. Then try the export again.

filmcafeSteve, 2019-08-28:
Absolutely amazing! Thank you so much for all of your fantastic support. I have a stitch!! 

Thank you for all you're doing to keep this project going. It's a great piece of software and I'm thrilled that it's not been lost.

w-m, 2019-08-28:
Hooray! I'm glad you made it completely through this arduous process and got something working in the end.

And thank you for the encouragement!

filmcafeSteve, 2019-08-29:
Honestly, it's you that's due the thanks! Hopefully the progress in #28 will mean I never need to darken your digital door again!

w-m, 2019-08-29:
I'm closing this issue as the build is working now. Feel free to come knocking to our digital door again if you have other questions!

closed 2019-08-29 by w-m

pharapeti, 2020-04-09:
> Go to the `bin/x64/release` folder and copy the file called `videostitch-cmd` to the `Studio` folder. Then try the export again.

Just ran into the same issue as described. This was the solution for me.

jepolo, 2020-04-13:
Thanks for this great thread, I helped save time getting the build going on Mac.

I did get everything working, but at the end the stitched output file is not being written to disk. I get the following Messages from the Ouput Log:

```
[Reader Controller] read a frame at timestamp 300300
stitched frame at 266933 useconds : 45 ms
stitched frame at -1 useconds : 57 ms
[libavoutput] LibavWriter : no implementation is instanciated
[libavoutput] failed to re-instanciate LibavWriter implementation.
[Reader Controller] read group 0 3 audio blocks starting at timestamp 533300
[audiopipeline] push audio mix /Users/jep/temp (siempre se borra)/VR/Gallinas/A0010016.MP4 send nb samples 512 at timestamp 533300
[audiopipeline] push audio mix /Users/jep/temp (siempre se borra)/VR/Gallinas/A0010016.MP4 send nb samples 512 at timestamp 543966
[audiopipeline] push audio mix /Users/jep/temp (siempre se borra)/VR/Gallinas/A0010016.MP4 send nb samples 512 at timestamp 554632
```
 This gets repeated for every frame processed. I am on macOS Catalina (10.15.4)

Thanks, and thanks for getting this useful tool out on Open Source.

pharapeti, 2020-04-14:
> This gets repeated for every frame processed. I am on macOS Catalina (10.15.4)

same here, on Mojave

w-m, 2020-04-15:
How did you install ffmpeg, through MacPorts? Did you install it including the non-free codecs?

Can you export to a JPEG sequence successfully, or is there a similar error?

jepolo, 2020-04-18:
Originally, I had installed via HomeBrew, but just to be on the safe side I:

1. Removed all the Homebrew packages. Everything, even Qt (which I installed via Homebrew).
2. Installed everything using MacPorts, and installed QT from its website.
3. Deleted the old build folder and Built everything again, using:

```
cmake -DCREATE_BOX_PACKAGE=OFF -DGPU_BACKEND_CUDA=OFF \
 -DGPU_BACKEND_OPENCL=ON -DQt5_DIR=~/Qt/5.14.2/clang_64/lib/cmake/Qt5 \
 -DMACPORTS=ON -DCMAKE_BUILD_TYPE=Release -G Ninja stitchEm
```

My mac has AMD Radeon Pro 5500M GPU. (Which VideoStitch recognized)
I use 6 GoPro Hero Black4 videos (2704x2028 each).

I ran the new VideoStitch  executable and I started a new project (just in case) with new source videos.

These are my findings:

- After Automatic Calibration, VideoStitch produced a 2048x1024 equirectangular projection that looks good.
- On the process Tab I click on "Process Now" with all defaults and I get a 2048x1024 mp4 stitch. (Success at this resolution)
- Then I return to the process tab and choose Size = 7680x3840  (pretty standard for this rig)
- Use all default settings and click "Process Now"

The Log Looks like:
```
[libavoutput] LibavWriter failed to instanciate implementation during creation.
[audiopipeline] Add output /Users/jep/trampolin
[ambisonic] Decoder bad coefficients given for stereo
[videoinput] Unknown av stream with index 3, ignoring.
[videoinput] Unknown av stream with index 2, ignoring.
[videoinput] Unknown av stream with index 3, ignoring.
[videoinput] Unknown av stream with index 3, ignoring.
[videoinput] Unknown av stream with index 2, ignoring.
[videoinput] Unknown av stream with index 3, ignoring.
Stitcher setup: 1138 ms
[Reader Controller] read a frame at timestamp 0
[Reader Controller] read group 0 25 audio blocks starting at timestamp 0
Initialize vumeter with nb channels 2 layout stereo
Initialize vumeter with nb channels 2 layout stereo
Initialize vumeter with nb channels 2 layout stereo
Initialize vumeter with nb channels 2 layout stereo
Initialize vumeter with nb channels 2 layout stereo
Initialize vumeter with nb channels 2 layout stereo
[audiopipeline] push audio mix /Users/jep/Videos/VR/Trampolin/A0010017.MP4 send nb samples 512 at timestamp 0
[libavoutput] LibavWriter : no implementation is instanciated
[libavoutput] AvMuxer : cannot create implementation without context
[libavoutput] failed to re-instanciate LibavWriter implementation.
... (last 4 lines repeat for further timecodes)
```

I was wondering if this would be different If I could apply the rig presets. I found the following file:
`stitchEm/apps/src/videostitch-studio-gui/presets/rigs/360Heroes - H3PRO6 (1440p).preset`
I tried installing it in my build folder (using same path), but it did not recognize it for calibration (maybe this is not yet implemented?)

Thanks for your help.

jepolo, 2020-04-18:
ffmpeg was installed as `ffmpeg +gpl2 +librtmp +nonfree` , just like in the Docs:

```
# Build tools
sudo port install CMake bison doxygen yasm ninja ccache

# Libraries
sudo port install opencv glew gsed jpeg libpng \
          tiff faac faad2 ceres-solver glfw glm OpenEXR \
          ffmpeg +gpl2 +librtmp +nonfree
```

I have not tried the JPEG sequence export yet. But thought being able to export at a lower resolution might uncover the problem.

Thanks.

w-m, 2020-04-19:
If exporting a resolution <4K works, and it stops working when exporting > 4K, than the installation itself seems fine.

Encoding a resolution beyond 4K with h264 can be tricky. If I recall correctly, the solution is to change the container format from MP4 to MOV.

w-m, 2020-04-19:
I just build and tried that, H264 output does not work with resolution > 4K with either container format on Mac.

MPEG and Motion JPEG outputs should work with these resolutions, ProRes too (in MOV containers).

You could export with a high bitrate or ProRes and then use the ffmpeg already installed to convert the output file to a codec of your choosing afterwards.