stitchEm/stitchEm #36: Linux Fedora 30  does not start wants CUDA  but only Opencl is enable OpenCL 1.1 Mesa 19.1.5

arpu, 2019-08-30:
```cmake -DCREATE_BOX_PACKAGE=OFF \                                                                                                                                                                                                   ✔  10214  23:10:05 
      -DGPU_BACKEND_CUDA=OFF -DGPU_BACKEND_OPENCL=ON \
      -G Ninja \
      -DCMAKE_BUILD_TYPE=Release
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

-- GPU_BACKEND_CUDA='OFF'
-- GPU_BACKEND_OPENCL='ON'
-- System: LINUX
-- CMake Build Type -- Release
-- USE_DELAY_LOAD = OFF
-- CUDA VERSION: 8.0
-- CUDA PATH: /usr/local/cuda-8.0
-- Could NOT find CUDA (missing: CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) (found suitable exact version "8.0")
No Alternative OpenCL SPIR compiler for Linux
-- --- OpenCL GPU backend ---
-- OpenCL version: 2.2
-- OpenCL includes: /usr/include
-- OpenCL libraries: /usr/lib64/libOpenCL.so
-- 
-- Main OpenCL SPIR compiler: /usr/bin/clang
-- Alternative OpenCL SPIR compiler: 
-- System: LINUX
-- --- VideoStitch-lib dependencies ---
-- OpenEXR libs: /usr/lib64/libHalf.so;/usr/lib64/libIex.so;/usr/lib64/libImath.so;/usr/lib64/libIlmImf.so;/usr/lib64/libIlmThread.so
-- OpenEXR include: /usr/include;/usr/include/openexr
-- PNG: /usr/lib64/libpng.so
-- R8B: r8b
-- OpenCV libs: /usr/lib64/libopencv_calib3d.so;/usr/lib64/libopencv_features2d.so;/usr/lib64/libopencv_flann.so;/usr/lib64/libopencv_video.so;/usr/lib64/libopencv_imgproc.so;/usr/lib64/libopencv_core.so
-- CERES LIBS: general;/usr/lib64/libceres.so
-- 
-- VideoStitch I/O libs configured to build: avPlugin;jpgPlugin;pamPlugin;pngPlugin;rawPlugin;tiffPlugin;exrPlugin;rtmpPlugin;v4l2Plugin;portaudioPlugin
-- Qt minimum supported version: 5.6
-- Qt FOUND: 5.12.4
-- Studio version: Studio-v0.0.9-31-g20693a5-master.2019-08-29
-- Configuring done
-- Generating done
-- Build files have been written to: /home/arpu/Work/githubsources/stitchEm_master
 arpu@testnet  ~/Work/githubsources/stitchEm_master   master ?  ninja                                                                                                                                                                                                                              ✔  10215  23:10:35 
[1/46] cd /home/arpu/Work/githubsources/stitchEm_master/lib && ls /home/arpu/Work/githubsources/stitchEm_master/generated_headers/version.hpp
/home/arpu/Work/githubsources/stitchEm_master/generated_headers/version.hpp
[16/18] Automatic MOC for target videostitch-studio
 arpu@testnet  ~/Work/githubsources/stitchEm_master   master ?  bin/x64/release/videostitch-studio                                                                                                                                                                                                 ✔  10216  23:10:40 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Host OpenCL allocated memory:
Device OpenCL allocated memory:
```

``` clinfo                                                                                                                                                                                                                             ✔  10220  23:12:33 
Number of platforms                               1
  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 19.1.5
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Clover
Number of devices                                 1
  Device Name                                     AMD Radeon (TM) RX 460 Graphics (POLARIS11, DRM 3.32.0, 5.2.9-200.fc30.x86_64, LLVM 8.0.0)
  Device Vendor                                   AMD
  Device Vendor ID                                0x1002
  Device Version                                  OpenCL 1.1 Mesa 19.1.5
  Driver Version                                  19.1.5
  Device OpenCL C Version                         OpenCL C 1.1 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
[... 62 lines cut ...]
  Device Extensions                               cl_khr_byte_addressable_store cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_fp64 cl_khr_fp16

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Clover
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [MESA]
  clCreateContext(NULL, ...) [default]            Success [MESA]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 Clover
    Device Name                                   AMD Radeon (TM) RX 460 Graphics (POLARIS11, DRM 3.32.0, 5.2.9-200.fc30.x86_64, LLVM 8.0.0)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 Clover
    Device Name                                   AMD Radeon (TM) RX 460 Graphics (POLARIS11, DRM 3.32.0, 5.2.9-200.fc30.x86_64, LLVM 8.0.0)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 Clover
    Device Name                                   AMD Radeon (TM) RX 460 Graphics (POLARIS11, DRM 3.32.0, 5.2.9-200.fc30.x86_64, LLVM 8.0.0)

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.12
  ICD loader Profile                              OpenCL 2.2
```

w-m, 2019-08-31:
Please try the build in a clean build folder, outside of the stitchEm source folder. The last argument to CMake configure call should be the location of the stitchEm source folder (e.g. "../stitchEm_master").

arpu, 2019-08-31:
ok done

```
/build_stitchem  cmake -DGPU_BACKEND_CUDA=OFF -DGPU_BACKEND_OPENCL=ON \                                                                                                                                                                                           ✔  10247  14:02:07 
      -G Ninja \
      -DCMAKE_BUILD_TYPE=Release ../stitchEm_master
-- GPU_BACKEND_CUDA='OFF'
-- GPU_BACKEND_OPENCL='ON'
-- System: LINUX
-- CMake Build Type -- Release
-- USE_DELAY_LOAD = OFF
-- CUDA VERSION: 8.0
-- CUDA PATH: /usr/local/cuda-8.0
-- Could NOT find CUDA (missing: CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) (found suitable exact version "8.0")
No Alternative OpenCL SPIR compiler for Linux
-- --- OpenCL GPU backend ---
-- OpenCL version: 2.2
-- OpenCL includes: /usr/include
-- OpenCL libraries: /usr/lib64/libOpenCL.so
-- 
-- Main OpenCL SPIR compiler: /usr/bin/clang
-- Alternative OpenCL SPIR compiler: 
-- System: LINUX
-- --- VideoStitch-lib dependencies ---
-- OpenEXR libs: /usr/lib64/libHalf.so;/usr/lib64/libIex.so;/usr/lib64/libImath.so;/usr/lib64/libIlmImf.so;/usr/lib64/libIlmThread.so
-- OpenEXR include: /usr/include;/usr/include/openexr
-- PNG: /usr/lib64/libpng.so
-- R8B: r8b
-- OpenCV libs: /usr/lib64/libopencv_calib3d.so;/usr/lib64/libopencv_features2d.so;/usr/lib64/libopencv_flann.so;/usr/lib64/libopencv_video.so;/usr/lib64/libopencv_imgproc.so;/usr/lib64/libopencv_core.so
-- CERES LIBS: general;/usr/lib64/libceres.so
-- 
-- VideoStitch I/O libs configured to build: avPlugin;jpgPlugin;pamPlugin;pngPlugin;rawPlugin;tiffPlugin;exrPlugin;rtmpPlugin;v4l2Plugin;portaudioPlugin
-- Qt minimum supported version: 5.6
-- Qt FOUND: 5.12.4
-- Studio version: Studio-v0.0.9-31-g20693a5-master.2019-08-29
-- Configuring done
-- Generating done
-- Build files have been written to: /home/arpu/Work/githubsources/build_stitchem
```

builds success with ninja 

but on start 

```
bin/x64/release/VahanaVR                                                                                                                                                                                                                         ✔  10250  14:11:14 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Host OpenCL allocated memory:
Device OpenCL allocated memory:
``
![Bildschirmfoto von 2019-08-31 14-11-46](https://user-images.githubusercontent.com/14173/64063804-9a727b80-cbf9-11e9-920f-4bfc7ab294a6.png)

w-m, 2019-08-31:
I believe the error stems from your OpenCL device not being found (and issue #37 leading to Vahana VR erroneously writing about CUDA devices).

Please confirm there are no OpenCL devices found by running `bin/x64/release/videostitch-cmd -d`, which lists the available OpenCL devices.

Your OpenCL printout says OpenCL 1.1 in a couple of places, is it actually OpenCL 1.2 capable?

Another thing to try: configure with `-DDISABLE_OPENCL_SPIR=ON`, build and run again.

arpu, 2019-08-31:
yes the opencl device is not found

```
Vbin/x64/release/videostitch-cmd -d                                                                                                                                                                                                               ✔  10251  14:11:50 
-d option used without parameter, list compatible GPU devices:
Could not find any compatible GPU device on this computer.
Host OpenCL allocated memory:
Device OpenCL allocated memory:

```

i use the build in mesa opencl build this only supports OpenCL 1.1
clinfo
```
Number of platforms                               1
  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 19.1.5
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Clover
Number of devices                                 1
  Device Name                                     AMD Radeon (TM) RX 460 Graphics (POLARIS11, DRM 3.32.0, 5.2.9-200.fc30.x86_64, LLVM 8.0.0)
  Device Vendor                                   AMD
  Device Vendor ID                                0x1002
  Device Version                                  OpenCL 1.1 Mesa 19.1.5
  Driver Version                                  19.1.5
  Device OpenCL C Version                         OpenCL C 1.1 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Max compute units                               14
  Max clock frequency                             1210MHz
  Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
  Preferred work group size multiple              64
  Preferred / native vector sizes                 
    char                                                16 / 16      
    short                                                8 / 8       
    int                                                  4 / 4       
    long                                                 2 / 2       
    half                                                 8 / 8        (cl_khr_fp16)
    float                                                4 / 4       
    double                                               2 / 2        (cl_khr_fp64)
  Half-precision Floating-point support           (cl_khr_fp16)
    Denormals                                     No
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 No
    Round to infinity                             No
    IEEE754-2008 fused multiply-add               No
    Support is emulated in software               No
  Single-precision Floating-point support         (core)
    Denormals                                     No
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 No
    Round to infinity                             No
    IEEE754-2008 fused multiply-add               No
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  No
  Double-precision Floating-point support         (cl_khr_fp64)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
[... 24 lines cut ...]
  Device Extensions                               cl_khr_byte_addressable_store cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_fp64 cl_khr_fp16

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Clover
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [MESA]
  clCreateContext(NULL, ...) [default]            Success [MESA]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 Clover
    Device Name                                   AMD Radeon (TM) RX 460 Graphics (POLARIS11, DRM 3.32.0, 5.2.9-200.fc30.x86_64, LLVM 8.0.0)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 Clover
    Device Name                                   AMD Radeon (TM) RX 460 Graphics (POLARIS11, DRM 3.32.0, 5.2.9-200.fc30.x86_64, LLVM 8.0.0)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 Clover
    Device Name                                   AMD Radeon (TM) RX 460 Graphics (POLARIS11, DRM 3.32.0, 5.2.9-200.fc30.x86_64, LLVM 8.0.0)

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.12
  ICD loader Profile                              OpenCL 2.2
```

arpu, 2019-08-31:
same with -DDISABLE_OPENCL_SPIR=ON

arpu, 2019-08-31:
looks like the mesa opencl is not found next would be try amd gpu pro opencl driver or the opensource rocm opencl driver from amd

w-m, 2019-08-31:
Yeah trying different drivers is a good idea!

arpu, 2019-09-06:
any plans to support mesa opencl OpenCL 1.1 Mesa 19.1.5  (OpenCL 1.1 Mesa 19.1.5)
if not i think this bug report can be closed

w-m, 2019-09-06:
stitchEm requires an OpenCL 1.2-capable device to run. Your AMD Radeon RX 460 is capable of OpenCL 2.0, so this should in general be fine.

I'm not familiar with Mesa OpenCL, but as your installation says OpenCL 1.1 in a couple of places, this may not work and you will have to use a different graphics driver.

Looking at the Mesa OpenCL package for Fedora 30 (https://fedora.pkgs.org/30/fedora-x86_64/mesa-libOpenCL-19.0.2-3.fc30.x86_64.rpm.html), it seems the driver provides a library called /usr/lib64/libMesaOpenCL.so.1

But the CMake configuration above shows that OpenCL is used from /usr/lib64/libOpenCL.so, also says OpenCL version 2.2.

The OpenCL version you use for building stitchEm should be the one that is installed alongside your currently active graphics driver.

w-m, 2019-09-11:
@arpu did you get it to work?

Please note down your findings, that would be helpful to people who might have a similar setup and stumble upon this issue in the future.

arpu, 2019-09-30:
@w-m  yes get some time to test this with the  amdgpu-pro opencl 
installed like this: https://www.one-button.org/posts/2019/08/amdgpu-pro-fedora-debian.html

clinfo:
```
Number of platforms                               1
  Platform Name                                   AMD Accelerated Parallel Processing
  Platform Vendor                                 Advanced Micro Devices, Inc.
  Platform Version                                OpenCL 2.1 AMD-APP (2906.7)
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd cl_amd_event_callback cl_amd_offline_devices 
  Platform Host timer resolution                  1ns
  Platform Extensions function suffix             AMD

  Platform Name                                   AMD Accelerated Parallel Processing
Number of devices                                 1
  Device Name                                     Baffin
  Device Vendor                                   Advanced Micro Devices, Inc.
  Device Vendor ID                                0x1002
  Device Version                                  OpenCL 1.2 AMD-APP (2906.7)
  Driver Version                                  2906.7
  Device OpenCL C Version                         OpenCL C 1.2 
  Device Type                                     GPU
  Device Board Name (AMD)                         AMD Radeon (TM) RX 460 Graphics
  Device Topology (AMD)                           PCI-E, 01:00.0
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Max compute units                               14
  SIMD per compute unit (AMD)                     4
  SIMD width (AMD)                                16
  SIMD instruction width (AMD)                    1
  Max clock frequency                             1210MHz
  Graphics IP (AMD)                               8.0
  Device Partition                                (core)
    Max number of sub-devices                     14
    Supported partition types                     None
    Supported affinity domains                    (n/a)
  Max work item dimensions                        3
  Max work item sizes                             1024x1024x1024
  Max work group size                             256
  Preferred work group size (AMD)                 256
  Max work group size (AMD)                       1024
  Preferred work group size multiple              64
  Wavefront width (AMD)                           64
  Preferred / native vector sizes                 
    char                                                 4 / 4       
    short                                                2 / 2       
    int                                                  1 / 1       
    long                                                 1 / 1       
    half                                                 1 / 1        (cl_khr_fp16)
    float                                                1 / 1       
    double                                               1 / 1        (cl_khr_fp64)
  Half-precision Floating-point support           (cl_khr_fp16)
    Denormals                                     No
    Infinity and NANs                             No
    Round to nearest                              No
    Round to zero                                 No
    Round to infinity                             No
    IEEE754-2008 fused multiply-add               No
    Support is emulated in software               No
  Single-precision Floating-point support         (core)
    Denormals                                     No
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
[... 54 lines cut ...]
    SPIR versions                                 1.2
  printf() buffer size                            4194304 (4MiB)
  Built-in kernels                                (n/a)
  Device Extensions                               cl_khr_fp64 cl_amd_fp64 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_fp16 cl_khr_gl_sharing cl_amd_device_attribute_query cl_amd_vec3 cl_amd_printf cl_amd_media_ops cl_amd_media_ops2 cl_amd_popcnt cl_khr_image2d_from_buffer cl_khr_spir cl_khr_gl_event 

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  No platform
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   No platform
  clCreateContext(NULL, ...) [default]            No platform
  clCreateContext(NULL, ...) [other]              Success [AMD]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 AMD Accelerated Parallel Processing
    Device Name                                   Baffin
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 AMD Accelerated Parallel Processing
    Device Name                                   Baffin
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 AMD Accelerated Parallel Processing
    Device Name                                   Baffin
```

now Vahana start successful

arpu, 2019-09-30:
still some problems to get the GPU working in the gui 
```
bin/x64/release/VahanaVR                                                                                                                                                                                                                         ✔  10928  16:29:03 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
QOpenGLTexture::setData() tried to set a null image
ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2564:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2564:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2564:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
ALSA lib pcm_dmix.c:1108:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
.[videoinput] Couldn't open file  for probing. Aborting.
[videoinput] Couldn't open file  for probing. Aborting.
[videoinput] Couldn't open file  for probing. Aborting.
[videoinput] Couldn't open file  for probing. Aborting.
QAbstractOpenGLTexture::setMipLevels(): This texture target does not support mipmaps
[videoinput] Couldn't open file  for probing. Aborting.
[videoinput] Couldn't open file  for probing. Aborting.
[videoinput] Couldn't open file  for probing. Aborting.
[videoinput] Couldn't open file  for probing. Aborting.
QOpenGLTexturePrivate::destroy() called without a current context.
Texture has not been destroyed
Host OpenCL allocated memory:
Device OpenCL allocated memory:
```

[image]
[image]

arpu, 2019-10-02:
ok looks like i get it working with disable SPIR

```cmake -DGPU_BACKEND_CUDA=OFF -DGPU_BACKEND_OPENCL=ON  -DDISABLE_OPENCL_SPIR=ON -G Ninja -DCMAKE_BUILD_TYPE=Release ..```

w-m, 2019-10-02:
When first running Vahana VR, did the process that optimises the program for the current GPU finish without an error? You could still try with the CMake flag -DDISABLE_OPENCL_SPIR=ON if there were issues with this.

Is the error from opening a new project or an old one? Can you stitch a new project with 4 sample inputs?

w-m, 2019-10-02:
Oh wow you fixed it while I was typing the reply? :)

arpu, 2019-10-02:
haha yes good timing :> but now i need to test how this will work

arpu, 2019-10-02:
i cannot see any process that optimises the programm :/

arpu, 2019-10-02:
@w-m  now i testing with bin/x64/release/VahanaVR 

after i want test the test inputs i get this

[image]

arpu, 2019-10-02:
```bin/x64/release/videostitch-cmd -d                                                                                                                                                                                                               ✔  10117  23:27:30 
-d option used without parameter, list compatible GPU devices:
Device 0: GPU device: AMD Baffin (64 Bit GPU), global memory size: 3227 MB
Host OpenCL allocated memory:
Device OpenCL allocated memory:
```

arpu, 2019-10-02:
testing with the videostitch-studio i get this

```
 arpu@testnet  ~/Work/githubsources/build_stitchem  bin/x64/release/videostitch-studio                                                                                                                                                                                                               ✔  10119  23:43:39 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
QObject::connect: No such slot PostProdStitcherController::setDisplayGrid(bool)
QObject::connect: No such signal PostProdProjectDefinition::reqReextract(SignalCompressionCaps*)
.Missing field 'rig' for BaseProjectDefinition.
[Status] [GPU] Invalid configuration: Selected GPU device can not display. Connect it to the monitor.
[Status] [GPU] Internal error: Uninitialized OpenCL context
Host OpenCL allocated memory:
Device OpenCL allocated memory:
 arpu@testnet  ~/Work/githubsources/build_stitchem  bin/x64/release/SimpleKernelTest                                                                                                                                                                                                                 ✔  10119  23:45:18 
[OpenCL] Creating context without OpenGL context sharing!
GPU device: AMD Baffin (64 Bit GPU), global memory size: 3183 MB
OpenCL version: OpenCL 1.2 AMD-APP (2906.7)
OpenCL MAX_WORK_GROUP_SIZE: 256
OpenCL MAX_WORK_ITEM_SIZES: (1024, 1024, 1024)
Host OpenCL allocated memory:
Device OpenCL allocated memory:
 dummyKernelTest: 0 used (3.84 KB max)

```

w-m, 2019-10-03:
The software needs a working OpenGL context that is shared when creating the OpenCL context. So that the stitched images, computed with OpenCL, can directly displayed with OpenGL.

It seems to me that the current OpenGL context can be found (https://github.com/stitchEm/stitchEm/blob/master/lib/src/backend/cl/context.cpp#L230), otherwise the line "[OpenCL] Creating context without OpenGL context sharing!" would be printed when running videostitch-studio. (It's okay to have the line in SimpleKernelTest, as no OpenGL context is created for running this simple computation test.)

The creation of the OpenCL context with sharing to the OpenGL context (https://github.com/stitchEm/stitchEm/blob/master/lib/src/backend/cl/context.cpp#L240) is probably what's failing with the invalid configuration error.

This can happen when the monitor is not connected to the GPU doing the OpenCL computations, but rather to an integrated GPU directly on the mainboard, for example.

But if your monitor is already connected to the correct GPU, then this unfortunately points to some kind of driver problem again, now involving the interchange between OpenGL and OpenCL.

closed 2020-05-03 by w-m