stitchEm/stitchEm #38: Original VideoStitch Software

KaptainKCG, 2019-10-12:
Hello! I know this is a long shot, but thought I'd ask since it seems the Windows version of this isn't ready yet.
A few years back I purchased a license for VideoStitch Studio V2.  Earlier this year I had to reformat my harddrive, and when I went to reinstall the software the license no longer validated and that's when I learned the company had gone under.
Does anyone know of a way to get my validation code working again? (I still have the original purchase email if needed for verification purposes) 
I just shot a 360 interview a few days ago, and if I can get this working again, it will save me a TON of time.

w-m, 2019-10-12:
The server that is validating the licenses is no longer running, so software can no longer be registered. You would need to find a backup of your home folder from a time where you had VideoStitch Studio installed and registered, and copy the "videostitch.lic" file over to your current installation to restore the software.

(cc @clairevdv please correct me if I'm wrong)

clairevdv, 2019-10-12:
That’s correct 😊

KaptainKCG, 2019-10-12:
Dang... Wish I would have known that a year ago. lol 
I'm searching my backups, but I'm pretty sure I don't have that file.

KaptainKCG, 2019-10-12:
Good news and Bad news:
Good:  I FOUND THE FILE!! 
Bad:  VideoStitch says "Invalid License Found.  Do you want to activate your product?"
I opened the .Lic file and everything looks accurate.  Version is set to "2.*", my email and name are both correct.  My only guess is the Machine ID is wrong (since I did a full re-install of windows?)
 **"machine_id" : "H+Ab8E+YnpycogDjW1tbW1s=\n",**

clairevdv, 2019-10-13:
Hi,

Indeed the license file will not work on another PC and a new Windows build is considered as a new PC.

Unfortunately, we haven’t yet compiled the source code without the license system. We plan to do so but have no deadline set.

KaptainKCG, 2019-10-14:
Damn... Well there's $500+ down the drain.  
Not sure what I'm gonna use for this project instead...

w-m, 2019-10-15:
Dang that sucks! :(

I don't have any suggestions for your current project. We've made progress over the weekend with the Windows build, please have a look at #16 for announcements on the Windows version from time to time. Hopefully it'll be ready for your next project!

closed 2019-10-16 by w-m

w-m, 2020-05-03:
The windows version can now be downloaded here: https://github.com/stitchEm/stitchEm/releases/latest