stitchEm/stitchEm #40: Ubuntu LTS  18.04.3 no cmake 3.11 has 3.10.2-1ubuntu2

arpu, 2019-10-25:
Hello
ubuntu lts 18.04.3  has only cmake 3.10 stitchEm need cmake 3.11 

i only can us ethe LTS for the amd pro gpu driver finding on 
https://www.amd.com/de/support/graphics/radeon-400-series/radeon-rx-400-series/radeon-rx-460

with opencl support

w-m, 2019-10-25:
To get a current CMake on Ubuntu LTS, please use the official Kitware APT Repository. Instructions are here: https://apt.kitware.com/

arpu, 2019-10-25:
after i get StitchEm build successfully on Ubuntu 
i get this OpenGL problem 
```
./VahanaVR 
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
mesa: for the   --simplifycfg-sink-common option: may only occur zero or one times!
mesa: for the   --global-isel-abort option: may only occur zero or one times!
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: radeonsi
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
free(): invalid next size (fast)
Aborted (core dumped)
```

glxinfo:
https://gist.github.com/arpu/7ebc937c68c2d3a216917d15be7350f0

arpu, 2019-10-26:
@w-m any idea what the Problem is?

w-m, 2019-10-27:
Unfortunately not. It's some kind of driver error that I'm not familiar with.

arpu, 2019-11-08:
ok now with latest amd-pro driver update this works
version 5.11.2019 19.30

closed 2019-11-08 by arpu