stitchEm/stitchEm #47: Mac cmake errors

aspentreemedia, 2019-12-02:
I bypassed the openAl in the install script and have everything else installed. However, when I attempt to do the cmake command I'm getting an error.

cmake -DGPU_BACKEND_CUDA=OFF -DGPU_BACKEND_OPENCL=ON  -DQt5_DIR=~/Qt/qt_/5.9.8/clang_64/lib/cmake/Qt5 -DCMAKE_BUILD_TYPE=Release -G Ninja ../../stitchEm

-- Studio version: Studio-v0.0.9-31-g20693a5-master.2019-08-29
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
librtmp
    linked by target "rtmpPlugin" in directory /Users/michaelweiland/Documents/Development/stitchEm/IO/src/rtmp

-- Configuring incomplete, errors occurred!

I've attached the error log.

Macbook Pro running 10.14.6 (Mojave), Xcode 11.2.1

Thanks,

Mike

[CMakeError.log](https://github.com/stitchEm/stitchEm/files/3911298/CMakeError.log)

w-m, 2019-12-02:
I assume you want to build VideoStitch Studio, so you will not need the RTMP plugin. Try adding `-DDISABLE_RTMP=ON` to the CMake invocation to disable building this plugin for which the library is missing.

aspentreemedia, 2019-12-03:
Thanks, that's at least gotten me closer... I think. I now it says it's generated the build files however looking at the build folder Studio and VahanaVR folders are empty.

I then run $ninja in the build folder but get this error:
ninja: error: '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/Accelerate.framework', needed by 'bin/x64/release/libvideostitch.dylib', missing and no known rule to make it

I'm a little unclear if I have to run ninja at that point but it appears to be the next step according to ticket #27 but it's not spelled out in the build.md and looking at the ninja help I don't see much of anything I should be passing if anything. 

Thanks,

Mike

w-m, 2019-12-03:
Yes, you need to run Ninja to build the project. Did you move, update or delete Xcode after running CMake? The system framework it's not finding at build time (when invoking ninja) must have still been there when configuring (running CMake).

After applying any changes to the system configuration, like changing something around Xcode, it's always best to clear out the whole build folder (all the files the CMake call created) and start fresh.

aspentreemedia, 2019-12-03:
No Xcode changes on my end as far as I know. This version of Xcode is pretty default as I've yet to bring over any old iOS or Mac projects to this machine. I cleared everything out of the build folder ran cmake again and then ninja and same error.

jeremad, 2019-12-03:
Can you try opening/creating a first project dummy with XCode UI? I think it installs a few stuffs when starting for the first time

aspentreemedia, 2019-12-03:
Opened Xcode and created a project but still no go. I started over with a new git stitchEm instance, created a build folder did a cmake into it and at this point it's using MacOSX10.15.sdk which is installed. Once the cmake is complete I ninja but it's at this point it's looking for MacOSX10.14.sdk which is NOT installed. I'll poke around to see if I can somehow download the 10.14.sdk 

Mike

aspentreemedia, 2019-12-03:
I updated build.ninja manually replacing -mmacosx-version-min=10.14 with 10.15 and it has gotten me a step closer. Ninja runs but I'm getting:
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

w-m, 2019-12-03:
What's the exact error message? You can run `ninja` again to get just the failing command.

aspentreemedia, 2019-12-03:
[1/491] cd /Users/michaelweiland/Documents/Devel...ent/stitchEm/build/generated_headers/version.hpp
/Users/michaelweiland/Documents/Development/stitchEm/build/generated_headers/version.hpp
[3/489] Linking CXX shared library bin/x64/release/libvideostitch.dylib
FAILED: bin/x64/release/libvideostitch.dylib 
: && ccache /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -Wall -Wextra -Wconversion -Wrange-loop-analysis -fcolor-diagnostics -Wno-sign-conversion -Wunreachable-code -Wno-unused-parameter -Wno-unused-function -Wno-missing-field-initializers -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -mmacosx-version-min=10.15 -dynamiclib -Wl,-headerpad_max_install_names  -o bin/x64/release/libvideostitch.dylib -install_name @rpath/libvideostitch.dylib lib/CMakeFiles/vslib-objects-opencl.dir/json/parser-generated/jsonParser.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/json/parser-generated/jsonLexer.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/ambisonic.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/asrc.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/audio.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/audioBlock.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/audiogen.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/audioPipeFactory.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/audioPipeline.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/converter.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/delayCalculator.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/envelopeDetector.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/filter.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/gain.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/resampler.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/sampleDelay.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/sigGen.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/wavReader.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/wavWriter.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/orah/orahAudioSync.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/audio/orah/orahProcessor.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/backend/common/glAllocator.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/backend/common/core1/voronoi.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/common/rwmutex.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/common/semaphore.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/common/thread.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/bufferedReader.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/controller.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/controllerInputFrames.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/ambDecoderDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/audioPipeDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/cameraDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/controlPointListDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/controlPointListUpdater.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/curves.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/delayedAction.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/depthDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/geometryDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/readerInputDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/inputDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/inputDefinitionUpdater.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/mergerMaskDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/mergerMaskUpdater.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/overlayInputDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/overlayInputDefinitionUpdater.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/panoDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/panoramaDefinitionUpdater.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/ptoParse.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/rigCameraDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/rigDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/defs/stereoRigDef.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/emor.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/geoProps.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/geoTransform.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/panoPipeline.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/panoStitcherBase.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/photoTransform.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/postprocessor.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/projections.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/pyramid.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/radial.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/readerController.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/rect.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/stereoPipeline.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/src/core/stitchOutput/algoOutput.cpp.o lib/CMakeFiles/vslib-objects-opencl.dir/ [... cut ...] stem/Library/Frameworks/Accelerate.framework /opt/local/lib/libsuitesparseconfig.dylib /opt/local/lib/libmetis.dylib /opt/local/lib/libcxsparse.dylib -framework OpenCL && :
ld: can't map file, errno=22 file '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/Accelerate.framework' for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

w-m, 2019-12-03:
Editing `build.ninja` is not advised, as you may end up with a broken configuration, and the changes can be overwritten by a cmake re-configuration (which may be called from ninja itself).

The problem is still the SDK version (`ld: can't map file, errno=22 file '/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/Accelerate.framework' for architecture x86_64` - it's trying to link against a framework, which is a directory, not a file, something is very broken in this linker invocation).

I'm not sure what the root of the problem is, maybe you're accidentally using a MacPorts compiler and not the Xcode compiler, and the MacPorts compiler has the 10.14 SDK hardcoded in. Is there anything custom in your build environment? Feel free to post the output of a clean CMake configuration run, I can have a look for things that look fishy.

You could try downgrading to Xcode 10.3, which ships with the 10.14 SDK.

aspentreemedia, 2019-12-04:
I've went to an older MacBook Pro running 10.14 and have gotten stitchEm to build and I've executed Studio! I'll see if I can figure out the difference and where my SDKs have gotten out of sync and continue with seeing what's up with my main system. I'll set up another clean stitchEm and report back the CMake.

I do have some general questions on StitchEm and whether or not it will suit my needs on the devices I intend to run it on, is there a forum or a better place to ask general questions?

jeremad, 2019-12-04:
You're in the right place :)

jeremad, 2019-12-04:
I'm glad you made it work

PR #48 (2019-12-03): MacPorts: ensure ffmpeg is installed w/ rtmpdump & non-free codecs
Fixes #47.
closed 2019-12-04 by jeremad