stitchEm/stitchEm #51: Video import Crash

aspentreemedia, 2019-12-05:
When creating a new project or dragging video files and dropping them into the drop pane causes the app to crash. I'm using 2 video files in an attempt to create a 180 equilatriangular video. If I select only 1 of the 2 videos in the New Project flow it imports than I can use the File -> Open Media to add the other video.

However I have a 2nd set of videos that I have tried this same workaround but it doesn't work, I have to import the same video from my first set and then I can Open Media to add the videos from the 2nd set. The issue with this is now I have 3 videos and there doesn't appear to be a way to remove the video from the 1st set.

I've attached a screenshot to show what I mean, the 1st video is from the 1st set and the only way I'm able to import is the other 2 is if I 1st import the 1st one when a new project is created and then I use the Open Media menu to add the other 2 but I don't see a way to remove the 1st one.

[image]

Mac OS 10.14.6 (Mojave)

Terminal output:
/Users/michaelweiland/Documents/Development/stitchEm/build/bin/x64/release/Studio/videostitch-studio ; exit;
Michaels-MacBook-Pro:~ michaelweiland$ /Users/michaelweiland/Documents/Development/stitchEm/build/bin/x64/release/Studio/videostitch-studio ; exit;
objc[48976]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fffa34513d8) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x1222b3f50). One of the two will be used. Which one is undefined.
QObject::connect: No such slot PostProdStitcherController::setDisplayGrid(bool)
QObject::connect: No such signal PostProdProjectDefinition::reqReextract(SignalCompressionCaps*)
.Missing field 'rig' for BaseProjectDefinition.
[ambisonic] Decoder bad coefficients given for stereo
Missing mandatory field 'video_codec' for object Ptv.
QAbstractOpenGLTexture::setMipLevels(): This texture target does not support mipmaps
[AudioPlayer] Audio data is not being fed to the audio device at a fast enough rate, cannot play audio.
Missing field 'rig' for BaseProjectDefinition.
[Status] [audio pipeline configuration] Invalid configuration: No audio processor definition found
[ambisonic] Decoder bad coefficients given for stereo
QOpenGLTexture::setData() tried to set a null image
QGLShader::compile(Vertex): ERROR: 0:1: '' :  version '330' is not supported

QObject::connect: No such slot PostProdStitcherController::setDisplayGrid(bool)
QObject::connect: No such signal PostProdProjectDefinition::reqReextract(SignalCompressionCaps*)
Missing field 'rig' for BaseProjectDefinition.
[ambisonic] Decoder bad coefficients given for stereo
Missing mandatory field 'video_codec' for object Ptv.
Segmentation fault: 11
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

w-m, 2019-12-05:
Could you please locate one of the crash reports in `~/Library/Logs/DiagnosticReports/` and attach it here?

Does the crash seem related to these particular inputs? Can you just drag in a couple of jpeg files, without it crashing? Then it would be very helpful to get a sample of the offending input files.

w-m, 2020-01-19:
@aspentreemedia can you provide the information (see last comment) on the crash? Otherwise I'll have to close this issue.

aspentreemedia, 2020-01-20:
Sorry I got sidetracked as my smallest videos so far are 3.3gb and I meant to put a little 10 sec sample together. Here's the crash report
[videostitch-studio_2020-01-20-135414_Michaels-MacBook-Pro.txt](https://github.com/stitchEm/stitchEm/files/4087659/videostitch-studio_2020-01-20-135414_Michaels-MacBook-Pro.txt)

I'm able to drop in jpegs and no crash. As stated in my initial report my workaround is to open the 1st mp4 file in Premiere and export it out. I then drag the newly exported video to the GUI and add the 2nd video file after creating the project (open media). I'll create a short 10 second clip this afternoon and upload the files.

w-m, 2020-01-22:
The crash is related to resampling of the audio tracks, so it makes sense that you don't see it with jpg-only inputs. When you upload a short sample, I can start reproducing and fixing the issue.

aspentreemedia, 2020-01-23:
That explains the next thing I noticed. When I got my first video in that was modified in Premiere and then the 2nd video that was straight from the camera I was getting an error that the audio settings were not the same so it couldn't sync the media using audio.

Here is a little clip straight from the camera and it causes a crash when I try to import it in.

http://videos.aspentreemedia.com.s3-website-us-east-1.amazonaws.com/2019_1126_230053_001.MP4.zip

aspentreemedia, 2020-02-16:
Did a little investigating. The mp4 files that my cameras are outputting are in Mono, when I open them in Premiere and export the default settings was outputting stereo. Instead of taking my files into Premiere to process which takes about 50 minutes per camera I now run this ffmpeg command to convert the mono to stereo and the resulting files import fine in StitchEm. It takes about 3 minutes per file with ffmpeg!

ffmpeg -i CamA.mp4 -ac 2 -c:v copy  CamA_Export.mp4