stitchEm/stitchEm #52: [Question]  port cuda to hip for multiplatform

arpu, 2019-12-27:
Hello,

any plans to use hip?  looks awesome for only  use cuda for all plattforms 
https://github.com/ROCm-Developer-Tools/HIP

w-m, 2020-01-06:
We don't have any plans on porting the GPU code to additional frameworks/platforms.

On the organizational side, the GPU stitching code is separated from the rest of the code, in https://github.com/stitchEm/stitchEm/tree/master/lib/src/backend. If you, or somebody else, wish to experiment with additional GPU backends, this can be done mostly in isolation from the rest of the code. I'd be happy to provide technical support for any such undertakings.

On the technical side, it seems HIP can't currently be used as a drop-in replacement, as it [doesn't support](https://github.com/ROCm-Developer-Tools/HIP/blob/master/docs/markdown/hip_faq.md#what-is-not-supported) OpenGL interop. Without a direct graphics interop, you would need to copy the stitched result roundtrip, from GPU to RAM and back, to display it, which kills the performance. Or write additional glue code to go HIP-CUDA-OpenGL on Nvidia devices and HIP-???-OpenGL on AMD devices, which doesn't really improve over the current situation of having a CUDA and an OpenCL backend.

w-m, 2020-01-19:
Feel free to reopen if there are new developments, or with information on HIP-OpenGL interop.

closed 2020-01-19 by w-m