stitchEm/stitchEm #53: More a Question: Stitching 180 Degree movies

aspentreemedia, 2020-01-18:
I have a question on the best settings when stitching 2 camera videos together for 180 degree videos. I have 2 cameras one pointing to the left and another over the top of the nodal point pointing right. I stitch them together but the resulting video there is a lot of black around it. I then take it into Premiere to crop it down. Ideally I'd like to know how to set up my panorama so I don't have to take this step.

My source footage is 3840 × 2160. The stitchEm output is 4096 × 2048. Here's an example of what a frame looks like, I'd like to make it so there is no black.

[image]

Thanks,

Mike

w-m, 2020-01-19:
A reduced output field-of-view (fov) is not directly possible through the UI, as there are too many combinations of settings that break the stitching in subtle ways.

If you are not afraid to tinker a little, you can create an equirectangular output with a reduced fov by editing the project file manually. It should be done as the last action before exporting, as some functionality (e.g. interactive view) will not behave correctly with a reduced fov. First, ensure that the output is perfectly centered. Use "Adjust on current frame" in the Orientation window until the output is centered vertically and horizontally. It's important to do this first, as you won't be able to do adjust orientation once cropped.

Second, save the project with the .ptv file ending, not .ptvb. Then edit the project file (now in JSON format) with a text editor. Change the "hfov" value (under "pano") from 360 to the intended value, e.g. 180. Reload the project, check the result in the "Output" window, and export.

aspentreemedia, 2020-02-04:
I've been able to produce videos doing this! However I've found an issue, not sure if it warrants another ticket. But when I attempt to set the dimensions to 8K or 6K resolutions with MOV and ProRel codec when I click process it tells me my settings have changed when I click Process. When I save and it then starts processing it goes through the motion and after several hours it's done, no errors on the screen but no file was generated. To get around this I selected Use Preferred size (sorry not at my workstation with SitchEm installed). I then use that dimensions and it renders. What I noticed when it fails, when it saves the file it changes the codec from ProRel to h264 and it's dimensions is greater than the max that h264 can handle. Instead of alerting me to this it tries to go through the motion. I've now noticed that when it works the file is immediately created and stays at 36k till the end when it's all done. 

Let me know if you want this written up separately as I can, just not sure if you do since I'm going around the GUI.

w-m, 2020-05-03:
#81 created for bug report in last comment, 180 degreee stitching was already resolved.

w-m, 2021-01-08:
Question was asked in the Facebook user group as well. We could add a paragraph describing the hack (see https://github.com/stitchEm/stitchEm/issues/53#issuecomment-576002637) to the user guide.

MajorTom101, 2023-07-23:
Hi, is there any intention to modify the software to allow the user to select for the output a particular angle of view? I would be interested too to stitch the videos of two cameras "only" covering about 180° hFOV.