stitchEm/stitchEm #54: Vahana and Orah 4i

peterridzon, 2020-01-22:
how can I connect to Vahana directly by Orah 4i camera, and its four H264 streams grab directly through RJ45? it's the same ex company, isn't it in the source code?

w-m, 2020-02-01:
It's possible to stitch the 4 streams from the camera with Vahana VR. The tricky part is getting the camera to start streaming. In hardware you would need a PoE injector to power the camera. In software you would need to setup a DHCP server, an RTMP nginx server for the camera to register with and then send some proprietary commands to the camera to start the streams at each power up. And maybe something else I'm forgetting. It's not my area of expertise, but a colleague has promised to have a look at how we can provide all the necessary information to get an Orah 4i running without the box.

Why do you ask, is your stitching box broken or do you wish to do a custom setup?

peterridzon, 2020-04-27:
sorry for the delay, I have a 20x orah cam, I do live multicam streaming on it, it's not very comfortable to have a 20x stitching box, I use NGINX it wouldn't be a problem, I could handle DHCP for the cameras, I have PoE solved, I connect the camera via optics or radio to the box, I know that the camera will release 4x rtmp H264 stream, the only thing I don't know are those commands :-)

peterridzon, 2020-04-27:
this is complete ORAH files from ex support webpage, if
would someone need https://www.dropbox.com/sh/tmw0871htsulmfr/AABBG90CImxooHLC7msPhn1La?dl=0

klem23, 2020-04-30:
Hi, 
sorry for the delay, but this camera is not in the scope of the open source project.
Anyway, we should quickly provide something to allow to start it and setup a conf to retrieve the streams.

klem23, 2020-05-02:
Please have a look here :
https://github.com/stitchEm/camorah

w-m, 2020-05-03:
Issue should be resolved, feedback and discussion can continue in the camorah repository.

closed 2020-05-03 by w-m

peterridzon, 2020-05-17:
Oh, this is spectacular, thank you very much I'm going to try :-)

johschmitz, 2021-06-16:
Anyone here knows how to extract the camera firmware or get the box Linux for camera firmware flashing?

johschmitz, 2021-06-17:
Just for the purpose of completeness I will post the link to the firmware here as well:
https://github.com/natecarlson/orah-4i-recovery