stitchEm/stitchEm #55: CUDA API Error

kiokju, 2020-02-13:
I try to show input Sources in Vahana VR.
Error occur and screen is black.

Here is error code. I had already install CUDA10.0.

C:\Users\buildbot\Works\buildbot-slave\windows_all\VideoStitch-all\lib\src\cuda/error.hpp(39) : CUDA Runtime API error 8:invalid device function

src/core1/cupanoRemmaper.cu(127) : CUDA Runtime API error 18:invalid texture reference

w-m, 2020-02-13:
What GPU are you using?

Please check the log window for more errors. If you see many other CUDA errors in the log, starting with "illegal address", then is is probably a duplicate of #49 - CUDA throwing many errors on Windows release builds.

If it's just the texture error, please check that your monitor is connected to the GPU you have selected to use for stitching.

kiokju, 2020-02-19:
I removed GTX 1080 Graphic card and replaced GTX 650.

Then error not occur.

Thanks.

closed 2020-02-19 by w-m