stitchEm/stitchEm #59: Jetson Nano cmake Error

aspentreemedia, 2020-02-25:
I'm attempting to get StitchEm Studio running on a Nvidia Jetson Nano. I'm running into it complaining that several variables are used in the project but they are not set:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
FFMPEG_avcodec
    linked by target "avPlugin" in directory /home/aspen/Documents/Development/stitchEm/IO/src/av
FFMPEG_avformat
    linked by target "avPlugin" in directory /home/aspen/Documents/Development/stitchEm/IO/src/av
FFMPEG_avutil
    linked by target "avPlugin" in directory /home/aspen/Documents/Development/stitchEm/IO/src/av
TIFF
    linked by target "tiffPlugin" in directory /home/aspen/Documents/Development/stitchEm/IO/src/tiff

I've tried a couple different cmake with -DISABLE_AV=ON and with -DISABLE_TIFF=ON
cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DISABLE_AV=ON -DDISABLE_RTMP=ON DISABLE_TIFF=ON -G Ninja stitchEm

cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF  -DDISABLE_RTMP=ON -G Ninja stitchEm

Any suggestions on how to get past this?
[CMakeError.log](https://github.com/stitchEm/stitchEm/files/4252419/CMakeError.log)

aspentreemedia, 2020-02-25:
System details if needed:
ffmpeg -version
ffmpeg version 3.4.6-0ubuntu0.18.04.1 Copyright (c) 2000-2019 the FFmpeg developers
built with gcc 7 (Ubuntu/Linaro 7.3.0-16ubuntu3)
configuration: --prefix=/usr --extra-version=0ubuntu0.18.04.1 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
libavutil      55. 78.100 / 55. 78.100
libavcodec     57.107.100 / 57.107.100
libavformat    57. 83.100 / 57. 83.100
libavdevice    57. 10.100 / 57. 10.100
libavfilter     6.107.100 /  6.107.100
libavresample   3.  7.  0 /  3.  7.  0
libswscale      4.  8.100 /  4.  8.100
libswresample   2.  9.100 /  2.  9.100
libpostproc    54.  7.100 / 54.  7.100

w-m, 2020-02-26:
There's a leading `D` missing in your invocation, it needs to be `-DDISABLE_AV=ON` to disable this plugin, `-D` to add the definition + `DISABLE_AV=ON`.

Without that plugin you won't be able to read or write any video, so that's probably not a useful build. You can add `PATHS path...` to these invocations to make sure the libraries are found.

aspentreemedia, 2020-02-26:
Good thing I omitted the extra D as that's not what I want to have happen! Is the issue my version of ffmpeg? It's using 3.4.6 as that's the latest verified release for the device, if I compile a 4.x version will that fix the issue? I didn't seem to run into this particular issue when I got StitchEm working on my Mac.

Finally how do I specify the paths exactly? Is that an argument I pass in in the cmake command or do I edit the CMakeLists.txt to point to the location where ffmpeg is installed (/usr/bin/ffmpeg). I'll probably need a good example to comprehend what needs to happen.

w-m, 2020-02-26:
I believe the ffmpeg version is not the issue, 3.4.6 should work. Check that you have installed not just the ffmpeg binaries (to run from the command line), but also the development libraries: avcodec, avformat, avutil.

I'm not familiar with the Jeston Nano environment, on a normal Ubuntu you could run `dpkg -l | grep avcodec` to see if the library is installed and `dpkg -L libavcodec-dev` to check the path it was installed to. If they are installed through the package manager, they should be automatically found in CMake.

In our installations on Linux these libraries are normally installed automatically, as dependencies of OpenCV.

aspentreemedia, 2020-02-27:
I installed 2 missing packages:
sudo apt-get install libavformat-dev
sudo apt-get install libavformat-dev

Then disabled TIFF as it was still an issue.

cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DDISABLE_RTMP=ON -DDISABLE_TIFF=ON -G Ninja stitchEm

I'm able to build successfully, at least it doesn't show any errors and says the build files have been written. I don't see a build folder nor anywhere videostich-studio was created.

Do I need to run anything else after the cmake?

jeremad, 2020-02-27:
you need to run `ninja`

jeremad, 2020-02-27:
There part is missing in the documentation, thanks for the catch, I'll update it asap

w-m, 2020-02-27:
I suggest you add `-DCUDA_LOCAL_ARCH_ONLY=ON` to you CMake invocation, if you want to run the app only on the device you build it on, to get a much faster build. And `-DCMAKE_BUILD_TYPE=Release` for an optimized build without debug symbols.

aspentreemedia, 2020-02-27:
Ok, I've added those to my cmake. It builds and then when I run ninja I'm getting an error:

aspen@nano-desktop:~/Documents/Development$ ninja
[3/959] Generating ../generated_headers/version.hpp
lib v0.0.9-55-g23f1206 @ master
Writing version.hpp
[6/959] cd /home/aspen/Documents/Devel...elopment/generated_headers/version.hpp
/home/aspen/Documents/Development/generated_headers/version.hpp
[177/959] Building CXX object lib/CMak...da.dir/src/util/compressionUtils.cpp.o
FAILED: lib/CMakeFiles/vslib-objects-cuda.dir/src/util/compressionUtils.cpp.o 
/usr/bin/c++  -DVS_CUDA -I/usr/local/cuda-10.0/include -IstitchEm/lib/src -IstitchEm/discovery/include -IstitchEm/lib/include -Igenerated_headers -Ilib/json -I/usr/include/eigen3 -isystem stitchEm/external_deps/include -isystem stitchEm/external_deps/include/libpng -isystem stitchEm/external_deps/include/glm -isystem _deps -isystem /usr/include/opencv4 -isystem /usr/include/openexr -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-function -Wno-missing-field-initializers -O3 -DNDEBUG -fPIC   -std=gnu++14 -MD -MT lib/CMakeFiles/vslib-objects-cuda.dir/src/util/compressionUtils.cpp.o -MF lib/CMakeFiles/vslib-objects-cuda.dir/src/util/compressionUtils.cpp.o.d -o lib/CMakeFiles/vslib-objects-cuda.dir/src/util/compressionUtils.cpp.o -c stitchEm/lib/src/util/compressionUtils.cpp
In file included from stitchEm/lib/src/util/compressionUtils.cpp:5:0:
stitchEm/lib/src/util/geometryProcessingUtils.hpp:69:99: error: ‘CV_CHAIN_APPROX_NONE’ was not declared in this scope
                                   std::vector<std::vector<cv::Point>>& points, const int method = CV_CHAIN_APPROX_NONE);
                                                                                                   ^~~~~~~~~~~~~~~~~~~~
In file included from stitchEm/lib/src/util/compressionUtils.hpp:6:0,
                 from stitchEm/lib/src/util/compressionUtils.cpp:4:
stitchEm/lib/src/util/compressionUtils.cpp: In static member function ‘static VideoStitch::Status VideoStitch::Util::Compression::polylineEncodeBinaryMask(int, int, const std::vector<unsigned char>&, std::__cxx11::string&)’:
stitchEm/lib/src/util/compressionUtils.cpp:33:91: error: ‘CV_CHAIN_APPROX_TC89_KCOS’ was not declared in this scope
       Util::GeometryProcessing::findImageContours<unsigned char>(size, data, 1, contours, CV_CHAIN_APPROX_TC89_KCOS));
                                                                                           ^
stitchEm/lib/include/libvideostitch/status.hpp:479:41: note: in definition of macro ‘PROPAGATE_FAILURE_STATUS’
     const VideoStitch::Status status = (call); \
                                         ^~~~
stitchEm/lib/src/util/compressionUtils.cpp:32:3: note: in expansion of macro ‘FAIL_RETURN’
   FAIL_RETURN(
   ^~~~~~~~~~~
[182/959] Building CXX object lib/CMak.../src/stabilization/ukfQuaternion.cpp.o
ninja: build stopped: subcommand failed.

w-m, 2020-02-27:
Hm, might be a problem of incompatible OpenCV versions. From the log it seems like you are using OpenCV 4 - what version exactly, how did you install it?

For a quick fix, try replacing `CV_CHAIN_APPROX_NONE` with `cv::CHAIN_APPROX_NONE` and `CV_CHAIN_APPROX_TC89_KCOS` with `cv::CHAIN_APPROX_TC89_KCOS`.

aspentreemedia, 2020-02-28:
The Nano came with CV 4.1.1 pre-installed. I verified it with a little python to print the version. Made the recommended changes and got a little further but still fails:

[3/959] Generating ../generated_headers/version.hpp
lib v0.0.9-55-g23f1206 @ master
Writing version.hpp
[7/959] cd /home/aspen/Documents/Devel...elopment/generated_headers/version.hpp
/home/aspen/Documents/Development/generated_headers/version.hpp
[185/959] Building CXX object lib/CMak...src/util/geometryProcessingUtils.cpp.o
FAILED: lib/CMakeFiles/vslib-objects-cuda.dir/src/util/geometryProcessingUtils.cpp.o 
/usr/bin/c++  -DVS_CUDA -I/usr/local/cuda-10.0/include -IstitchEm/lib/src -IstitchEm/discovery/include -IstitchEm/lib/include -Igenerated_headers -Ilib/json -I/usr/include/eigen3 -isystem stitchEm/external_deps/include -isystem stitchEm/external_deps/include/libpng -isystem stitchEm/external_deps/include/glm -isystem _deps -isystem /usr/include/opencv4 -isystem /usr/include/openexr -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-function -Wno-missing-field-initializers -O3 -DNDEBUG -fPIC   -std=gnu++14 -MD -MT lib/CMakeFiles/vslib-objects-cuda.dir/src/util/geometryProcessingUtils.cpp.o -MF lib/CMakeFiles/vslib-objects-cuda.dir/src/util/geometryProcessingUtils.cpp.o.d -o lib/CMakeFiles/vslib-objects-cuda.dir/src/util/geometryProcessingUtils.cpp.o -c stitchEm/lib/src/util/geometryProcessingUtils.cpp
stitchEm/lib/src/util/geometryProcessingUtils.cpp: In static member function ‘static VideoStitch::Status VideoStitch::Util::GeometryProcessing::findImageContours(cv::Size, const std::vector<_Tp>&, T, std::vector<std::vector<cv::Point_<int> > >&, int)’:
stitchEm/lib/src/util/geometryProcessingUtils.cpp:398:51: error: ‘CV_RETR_EXTERNAL’ was not declared in this scope
   cv::findContours(src_gray, contours, hierarchy, CV_RETR_EXTERNAL, method, cv::Point(0, 0));
                                                   ^~~~~~~~~~~~~~~~
[188/959] Building CXX object lib/CMak...ir/src/util/imageProcessingUtils.cpp.o
FAILED: lib/CMakeFiles/vslib-objects-cuda.dir/src/util/imageProcessingUtils.cpp.o 
/usr/bin/c++  -DVS_CUDA -I/usr/local/cuda-10.0/include -IstitchEm/lib/src -IstitchEm/discovery/include -IstitchEm/lib/include -Igenerated_headers -Ilib/json -I/usr/include/eigen3 -isystem stitchEm/external_deps/include -isystem stitchEm/external_deps/include/libpng -isystem stitchEm/external_deps/include/glm -isystem _deps -isystem /usr/include/opencv4 -isystem /usr/include/openexr -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-function -Wno-missing-field-initializers -O3 -DNDEBUG -fPIC   -std=gnu++14 -MD -MT lib/CMakeFiles/vslib-objects-cuda.dir/src/util/imageProcessingUtils.cpp.o -MF lib/CMakeFiles/vslib-objects-cuda.dir/src/util/imageProcessingUtils.cpp.o.d -o lib/CMakeFiles/vslib-objects-cuda.dir/src/util/imageProcessingUtils.cpp.o -c stitchEm/lib/src/util/imageProcessingUtils.cpp
stitchEm/lib/src/util/imageProcessingUtils.cpp: In static member function ‘static VideoStitch::Status VideoStitch::Util::ImageProcessing::findCropCircle(const int&, const int&, void*, int&, int&, int&, const VideoStitch::Ptv::Value*, const string*)’:
stitchEm/lib/src/util/imageProcessingUtils.cpp:26:43: error: ‘CV_RGBA2BGR’ was not declared in this scope
   cv::cvtColor(originalImage, inputImage, CV_RGBA2BGR);
                                           ^~~~~~~~~~~
[190/959] Building CXX object lib/CMak.../src/stabilization/ukfQuaternion.cpp.o
ninja: build stopped: subcommand failed.

w-m, 2020-02-28:
I've created a ticket for the code to work with OpenCV 4, but I won't be able to work on this in the coming days.

If you want to make progress on your build, proceed in the same fashion with the new errors. Look at the build output, you will find lines with "error: " in them. If it's something about `CV_XYZ` not being found, just replace these names with `cv::XYZ`. The filename and line number where to change it are also written in the build log.

aspentreemedia, 2020-02-28:
Ok, I'll go through and make changes as I run into issues and report back. Hopefully I can get Vahana working, I need to see what's missing or why I get stuck with RTMP. 

Let me run this use case by you and ultimately how I might use the nano in my workflow. For my camera rig I'll be adding a nano or a raspberry pi 4 to control the 2 cameras, right now I hit the record buttons of my 2 cameras at the same time. It works but in daylight it's hard to see if the red light is flashing and on 2 occasions now 1 of the cameras was not recording. If I can get Vahana VR running on either of these devices will it produce a 180 video pano much like what I would get from StitchEm Studio? 

My first thought is to connect over wifi or bluetooth to the device and send  commands to start/stop recording to a python app. However if I can utilize Vahana then I don't need a separate python script I just need a CLI or a way to interact with Vahana. I do not see a CLI so there doesn't appear to be a way to start things up through the command line. Am I missing something?

How feasible is a CLI? A CLI would also open the doors to stitching in the cloud. I could record the 2 videos and then upload to AWS and then have a GPU instance running StitchEm Studio start up, stitch and when the stitching is done then cycle the instance back off. That way I'm only paying for the time it takes to stitch the files. 

Thoughts on this use case?

w-m, 2020-02-28:
These are a couple of different scenarios, with different solutions.

First, for the offline (i.e. non-live) use case of stitching in the cloud: this one is the easiest to implement. The CLI is called videostitch-cmd, it should be built by default. You can run with `--help` to discover the options.

Basically this is what the "batch stitcher" from VideoStitch Studio is using, processing a Studio .ptv/.ptvb from the command line, without any UI. I believe this should work fine on a headless GPU server.

Other than a beginning and end frame and the output file name, you can not specify any project settings with this tool. So you need to setup your project with Studio. To change minor things like file names of the input files, I suggest you do this with Python. Read the Studio-configured .ptv file as json, and do the necessary changes, then feed it into videostitch-cmd.

(If you're familiar with C++, you can also of course adapt videostitch-cmd directly to your needs, it's open source, after all).

The live-stitching case is always more difficult. I guess you could replace the file names in your .ptv with the rtmp:// links (once you got them at least working in Vahana VR) and try stitching that with videostitch-cmd, and see what happens - to be honest, I can't remember whether this works, runs into issues of blocking/synchronization or just straight refuses to stitch.

The live scenario that you are describing is pretty much the idea of the Orah 4i, which has 4 camera streams and a stitching box, a device that controls the camera and stitches the feed. The stitching part for the Orah 4i was done in Python.

There's a SWIG Python interface on top of the C++ stitching interface, which is what the Orah 4i stitching box uses, you can poke around it in https://github.com/stitchEm/stitchEm/tree/master/lib/bindings/samples - but be warned that the code is tailored completely to the Orah 4i.

w-m, 2020-05-03:
For the build issue on the Jetson due to OpenCV4, see the fix in the `opencv4` branch. This should resolve the original issue.

Feel free to reopen the ticket if there are more issues with the Jetson build. For questions about the command line interface please open new issues.

closed 2020-05-03 by w-m