stitchEm/stitchEm #61: Streaming capture video display error

kiokju, 2020-02-28:
Did you Khow why this error happend?
( I use two magwell pro capture board & 6 gopro cameras )

[image]

w-m, 2020-02-28:
Is this a Windows debug build?

What GPU are you using now?

Is this error appearing immediately when adding the inputs, or after calibration?

Please post the complete output of the logging window (button on bottom right corner).

kiokju, 2020-03-02:
1. release mode.
2. GeForce RTX 2080 Ti
3. When after select Input video source(mapwell pro * 6)
4. Error code is below.

stitcher controller created
StateManager::changeState, state = 0, sender = ProjectWorkWidget
StateManager::changeState, state = 1, sender = ProjectWorkWidget
[2020/03/02 14:43:35] Debug: could not load plug-in "D:\Workspace\stitchEm\bin\x64\release\vahana_plugins\portaudio.dll": � ü���� %1��(��) ������ �� �����ϴ�.
[2020/03/02 14:43:35] Debug: could not load plug-in "D:\Workspace\stitchEm\bin\x64\release\vahana_plugins\rtmp.dll": ������ ������ ã�� �� �����ϴ�.
[2020/03/02 14:43:35] Debug: could not load plug-in "D:\Workspace\stitchEm\bin\x64\release\vahana_plugins\ximea_64.dll": ������ ������ ã�� �� �����ϴ�.
[2020/03/02 14:43:35] Debug: could not load plug-in "D:\Workspace\stitchEm\bin\x64\release\core_plugins\tiff.dll": ������ ���ν����� ã�� �� �����ϴ�.
[2020/03/02 14:43:35] Debug: could not load plug-in "D:\Workspace\stitchEm\bin\x64\release\core_plugins_CUDA\av_cuda.dll": ������ ������ ã�� �� �����ϴ�.
Missing field 'exposure' for BaseProjectDefinition.
Failed to apply the configuration to the inputs

- Thanks!!

w-m, 2020-03-02:
If it's a Windows release build, then this is probably the same issue listed here: #49 - none of the Windows release builds are working correctly unfortunately, due to a CUDA issue we are unable to pinpoint.

w-m, 2020-05-03:
Closing as duplicate of (now fixed) #49.

closed 2020-05-03 by w-m