stitchEm/stitchEm #70: Mac Studio has NVENC encoding options, which can't work

w-m, 2020-04-19:
NVENC is not avaiable on Mac, Studio crashes when selecting these. They shouldn't show up.

w-m, 2020-05-05:
On Linux, where the encoding could potentially work, it also crashes, see report in https://github.com/stitchEm/stitchEm/issues/86#issuecomment-623941296

tlalexander, 2020-05-06:
I should note that in Linux the crash occurred as soon as I clicked on the NVENC option in the dropdown. Before even trying to export.

w-m, 2020-06-28:
The NVENC encoding feature was being developed for Vahana VR and doesn't have a UI in Studio yet. It shows up in Studio due to code sharing between Vahana VR and Studio, but it was never working. It will be disabled in the next Studio release (#95).

closed 2020-06-28 by w-m