stitchEm/stitchEm #71: Provide automatic builds

amenk, 2020-04-26:
It would be great to provide automatic build, for example using GitHub actions.

amenk, 2020-04-26:
There seem to be automatic builds and docker images, but they are not really mentioned in the readme.

jeremad, 2020-04-26:
We could provide those on Linux. I'll have to take a look for MacOS. On windows this is a lost cause.

amenk, 2020-04-26:
I am mainly interested in linux, while there are also GitHub actions runners for Windows and macos

jeremad, 2020-04-26:
Sure. However the dependencies take forever to build on windows (at least two hours last time I did it), and it's really tricky to install all the GPU utils in a virtualized environment

amenk, 2020-04-26:
alright :-)

For Linux it might be enough to use upload-artifacts? https://help.github.com/en/actions/configuring-and-managing-workflows/persisting-workflow-data-using-artifacts

jeremad, 2020-04-26:
sure thing :). I am working on building a windows installer as we speak. But, with the lock down, I'll probably have some time this week and next week-end to work on that

w-m, 2020-05-03:
macOS build requires the build directory to be called `cmake_build`, next to (not inside) the source dir.

Additional CMake option is just `-DBUILD_MACOSX_BUNDLE=ON` (well and `-DCMAKE_BUILD_TYPE=Release`).

Then call `apps/dmg/make_studio_dmg.sh`, which creates a "VideoStitch Studio.dmg" file in the apps/dmg directory.

I moved the .app package from the .dmg to a zip manually, as GitHub didn't allow me to attach a .dmg to the release.

w-m, 2020-05-03:
https://github.com/stitchEm/stitchEm/blob/master/apps/dmg/make_dmg_functions.sh#L247-L271 this stuff requires a real window-managed user session, which I guess won't be there on the build machine, and this would fail. 

If we get to this point but these lines fail, I would rewrite this part to just zip up the app package.

(The .dmg also used to contain a link to the website, which I removed, and a nice background image, which I think we can live without, so the zip is equivalent).

jeremad, 2020-05-04:
Indeed it does not seem doable

PR #83 (2020-05-04): Provide automated builds on linux for master branch
Drop automated builds for older cuda version

Closes: #71
closed 2020-05-04 by jeremad