stitchEm/stitchEm #72: Linker errors crashing build

tlalexander, 2020-05-02:
Hello! I wonder if I've done something wrong. Builds are failing with a bunch of linker errors.

I get a repeated dump of linker errors that look like this:

```
[876/1395] Linking CXX executable bin/x64/debug/OptimalPanoSizeTest
FAILED: bin/x64/debug/OptimalPanoSizeTest 
: && /usr/bin/g++-6  -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-function -Wno-missing-field-initializers -g   lib/src/test/CMakeFiles/OptimalPanoSizeTest.dir/optimalPanoSizeTest.cpp.o  -o bin/x64/debug/OptimalPanoSizeTest  -Wl,-rpath,/media/taylor/external/Software/stitchEm/bin/x64/debug  bin/x64/debug/libvideostitch_cuda.so  bin/x64/debug/libvideostitch-gpudiscovery.so  /usr/lib/x86_64-linux-gnu/libOpenCL.so  /usr/lib/x86_64-linux-gnu/libnvidia-ml.so  /usr/local/cuda-10.0/lib64/libcudart_static.a  -ldl  /usr/lib/x86_64-linux-gnu/librt.so  /usr/lib/x86_64-linux-gnu/libcuda.so  -lpthread  /usr/lib/x86_64-linux-gnu/libHalf.so  /usr/lib/x86_64-linux-gnu/libIex.so  /usr/lib/x86_64-linux-gnu/libImath.so  /usr/lib/x86_64-linux-gnu/libIlmImf.so  /usr/lib/x86_64-linux-gnu/libIlmThread.so  /usr/lib/x86_64-linux-gnu/libdrm.so && :
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int)'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::base::CheckOpMessageBuilder::CheckOpMessageBuilder(char const*)'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_post'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_finish'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_etree'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::base::CheckOpMessageBuilder::~CheckOpMessageBuilder()'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_free_factor'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_pinv'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_counts'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_print_common'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_compress'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_solve'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_sfree'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int, google::CheckOpString const&)'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_free_sparse'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_spfree'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_free_dense'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_transpose'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `dgels_'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `dsyrk_'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_factorize'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_analyze_p'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `dpotrf_'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_nfree'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_amd'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_cumsum'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_symperm'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_lsolve'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_analyze'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::base::CheckOpMessageBuilder::ForVar2()'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_ltsolve'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_start'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_schol'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_calloc'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_multiply'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_zeros'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `dpotrs_'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_pvec'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_malloc'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_free'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_ipvec'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_amd'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_camd'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::LogMessageFatal::~LogMessageFatal()'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_chol'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::base::CheckOpMessageBuilder::NewString[abi:cxx11]()'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_triplet_to_sparse'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::LogMessage::stream()'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_cholsol'
collect2: error: ld returned 1 exit status

```
Any ideas what it could be? Thank you!

jeremad, 2020-05-02:
Hello! Can you provide a little bit more details:
* Which OS
* The cmake command you run
Thanks!

tlalexander, 2020-05-02:
Sure.

Ubuntu 18.04.4 LTS

I've tried the recommended cmake command (but I moved in to the stitchEm directory):

cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=ON  -G Ninja .

I've tried:
cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF  -G Ninja .

And I've tried it without cuda:
cmake -DGPU_BACKEND_CUDA=OFF -DGPU_BACKEND_OPENCL=ON -G Ninja .

In the last case the name of the binary changes to libvideostitch_opencl.so but I get the same linker errors:

```
(just some of the error message)
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `google::LogMessageFatal::~LogMessageFatal()'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_chol'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `google::base::CheckOpMessageBuilder::NewString[abi:cxx11]()'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_triplet_to_sparse'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `google::LogMessage::stream()'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_cholsol'
collect2: error: ld returned 1 exit status
[453/922] Linking CXX executable bin/x64/debug/videostitch-cmd
FAILED: bin/x64/debug/videostitch-cmd 
: && /usr/bin/g++-6  -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-function -Wno-missing-field-initializers -g   lib/samples/sample-02-fullCmd/CMakeFiles/videostitch-cmd.dir/videostitch-cmd.cpp.o lib/samples/sample-02-fullCmd/CMakeFiles/videostitch-cmd.dir/__/common/cmdUtils.cpp.o  -o bin/x64/debug/videostitch-cmd  -Wl,-rpath,/media/taylor/external/Software/stitchEm/bin/x64/debug  bin/x64/debug/libvideostitch_opencl.so  bin/x64/debug/libvideostitch-gpudiscovery.so  /usr/lib/x86_64-linux-gnu/libglfw.so  /usr/lib/x86_64-linux-gnu/libOpenCL.so  -lpthread  /usr/lib/x86_64-linux-gnu/libHalf.so  /usr/lib/x86_64-linux-gnu/libIex.so  /usr/lib/x86_64-linux-gnu/libImath.so  /usr/lib/x86_64-linux-gnu/libIlmImf.so  /usr/lib/x86_64-linux-gnu/libIlmThread.so  /usr/lib/x86_64-linux-gnu/libdrm.so && :
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int)'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `google::base::CheckOpMessageBuilder::CheckOpMessageBuilder(char const*)'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_post'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_finish'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_etree'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `google::base::CheckOpMessageBuilder::~CheckOpMessageBuilder()'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_free_factor'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_pinv'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_counts'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_print_common'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_compress'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_solve'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_sfree'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int, google::CheckOpString const&)'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_free_sparse'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_spfree'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_free_dense'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_transpose'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `dgels_'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `dsyrk_'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_factorize'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_analyze_p'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `dpotrf_'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_nfree'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_amd'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_cumsum'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_symperm'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_lsolve'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_analyze'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `google::base::CheckOpMessageBuilder::ForVar2()'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_ltsolve'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_start'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_schol'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_calloc'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_multiply'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_zeros'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `dpotrs_'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_pvec'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_malloc'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_free'
[... cut ...]
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_chol'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `google::base::CheckOpMessageBuilder::NewString[abi:cxx11]()'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cholmod_triplet_to_sparse'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `google::LogMessage::stream()'
bin/x64/debug/libvideostitch_opencl.so: undefined reference to `cs_di_cholsol'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
Thanks!

jeremad, 2020-05-03:
can you run the command `ls -l /usr/lib/x86_64-linux-gnu/libglog*` and give me the output?

It seems we have trouble linking to libglog

jeremad, 2020-05-03:
And did you install the dependencies listed in: https://github.com/stitchEm/stitchEm/blob/master/doc/BUILD.md#building-with-linux

tlalexander, 2020-05-04:
Sure I will install from scratch again. I've cloned the repo fresh and rerun the dependency installation commands. There were "0 changes" after running the install command, so yes everything was installed. (Note also a typo in the line "sudo apt install" in the docs - it needs a trailing slash.)

I've run:
```
export CC=gcc-6
export CXX=g++-6
```
but not the lines using "update-alternatives".

Then I've run:
cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -G Ninja stitchEm

(from outside the directory as recommended by the docs, in case that matters).

then "ninja".

Same error. Here is the output of the command:
```
$ ls -l /usr/lib/x86_64-linux-gnu/libglog*
-rw-r--r-- 1 root root 240452 Jun 23  2017 /usr/lib/x86_64-linux-gnu/libglog.a
lrwxrwxrwx 1 root root     16 Jun 23  2017 /usr/lib/x86_64-linux-gnu/libglog.so -> libglog.so.0.0.0
lrwxrwxrwx 1 root root     16 Jun 23  2017 /usr/lib/x86_64-linux-gnu/libglog.so.0 -> libglog.so.0.0.0
-rw-r--r-- 1 root root 134232 Jun 23  2017 /usr/lib/x86_64-linux-gnu/libglog.so.0.0.0

```

Thanks again!

jeremad, 2020-05-04:
I'm really scratching my head on this one. I tried to reproduce yesterday, without any success. You can try the binaires build automatically: https://github.com/stitchEm/stitchEm/actions/runs/94727170

this archive in whatever path and add this path to `LD_LIBRARY_PATH`

jeremad, 2020-05-04:
Oh, The build doc is probably also outdated, can you try with the default gcc? Should be gcc 7 IIRC

jeremad, 2020-05-04:
Yeah... The GCC thing was useful for CUDA 9, If you're using cuda 10, don't bother doing that

tlalexander, 2020-05-04:
Wow the binaries seem to work! I can run videostitch-studio after adding that folder to LD_LIBRARY_PATH. I will test it shortly.

I did a build with the default gcc, which is:
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

The build failed with the same image I get when I try to build this in a docker container:

The full command output for ninja was:

```
[1/539] cd /media/taylor/external/Software/lib && ls /media/taylor/external/Software/generated_headers/version.hpp
/media/taylor/external/Software/generated_headers/version.hpp
[14/539] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/cuvid.cpp.o
FAILED: IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/cuvid.cpp.o 
/usr/bin/g++-6  -DSUP_NVDEC -DSUP_NVENC -DUSE_AVFORMAT -DrtmpPlugin_EXPORTS -I_deps/nvenc_api-src/include/ffnvcodec -IstitchEm/lib/include -Igenerated_headers -IstitchEm/external_deps/include/nvenc -I/usr/local/cuda-10.2/include -IstitchEm/IO/src/rtmp/Nvidia/include -IstitchEm/IO/src/rtmp/include -IstitchEm/IO/src/rtmp/../common/include -IstitchEm/IO/src/rtmp//include -IstitchEm/IO/src/rtmp/../common/format/include -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -g -fPIC   -std=gnu++14 -MD -MT IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/cuvid.cpp.o -MF IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/cuvid.cpp.o.d -o IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/cuvid.cpp.o -c stitchEm/IO/src/rtmp/src/cuvid.cpp
In file included from stitchEm/IO/src/rtmp/src/cuvid.cpp:6:0:
stitchEm/IO/src/rtmp/Nvidia/include/frameQueue.hpp:15:21: fatal error: nvcuvid.h: No such file or directory
 #include <nvcuvid.h>
                     ^
compilation terminated.
[31/539] Linking CXX shared library bin/x64/debug/libvideostitch.so
ninja: build stopped: subcommand failed.

```

jeremad, 2020-05-04:
OK nice, this the same error in #84 , use the cmake options I gave you and you should be good to go!

jeremad, 2020-05-04:
I'll improve the document to avoid the roadblocks you got in #83

tlalexander, 2020-05-04:
Thank you! I am really happy for your help and for this software. Please let me know if there's anything I could do to help.

Also I could open a new issue but I'm missing some UI elements I think? Based on a youtube video of this software I think I should have "source/output/interactive/process" tabs on the top right, but I've got nothing. I do not seem to have any way to save output. Thoughts?

[image]

tlalexander, 2020-05-04:
Well I think that was with the binary from the automatic builds. Trying a build with the updated cmake command to see if something built on my system still has UI issues...

tlalexander, 2020-05-04:
For the regular build on my system I am still getting errors. I cloned the repo from scratch again, made a build directory, and ran cmake from there:

`cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -G Ninja ..`

I did not run any commands to modify which gcc it would use, and I just confirmed it used gcc 7.5.0.

I ran ninja and got a bunch of errors that look like this:

```
[497/983] Linking CXX executable bin/x64/debug/MetadataToneCurveTest
FAILED: bin/x64/debug/MetadataToneCurveTest 
: && /usr/bin/c++  -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-function -Wno-missing-field-initializers -g   lib/src/test/CMakeFiles/MetadataToneCurveTest.dir/metadataToneCurveTest.cpp.o  -o bin/x64/debug/MetadataToneCurveTest  -Wl,-rpath,/media/taylor/external/Software/stitchEm/build/bin/x64/debug  bin/x64/debug/libvideostitch_cuda.so  bin/x64/debug/libvideostitch-gpudiscovery.so  /usr/lib/x86_64-linux-gnu/libnvidia-ml.so  /usr/local/cuda-10.2/lib64/libcudart_static.a  -ldl  /usr/lib/x86_64-linux-gnu/librt.so  /usr/lib/x86_64-linux-gnu/libcuda.so  -lpthread  /usr/lib/x86_64-linux-gnu/libHalf.so  /usr/lib/x86_64-linux-gnu/libIex.so  /usr/lib/x86_64-linux-gnu/libImath.so  /usr/lib/x86_64-linux-gnu/libIlmImf.so  /usr/lib/x86_64-linux-gnu/libIlmThread.so  /usr/lib/x86_64-linux-gnu/libdrm.so && :
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int)'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::base::CheckOpMessageBuilder::CheckOpMessageBuilder(char const*)'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_post'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_finish'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_etree'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::base::CheckOpMessageBuilder::~CheckOpMessageBuilder()'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_free_factor'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_pinv'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_counts'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_print_common'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_compress'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_solve'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_sfree'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int, google::CheckOpString const&)'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_free_sparse'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_spfree'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_free_dense'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_transpose'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `dgels_'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `dsyrk_'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_factorize'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_analyze_p'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `dpotrf_'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_nfree'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_amd'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_cumsum'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_symperm'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_lsolve'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_analyze'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::base::CheckOpMessageBuilder::ForVar2()'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_ltsolve'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_start'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_schol'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_calloc'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_multiply'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_zeros'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `dpotrs_'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_pvec'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_malloc'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_free'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_ipvec'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_amd'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_camd'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::LogMessageFatal::~LogMessageFatal()'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_chol'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::base::CheckOpMessageBuilder::NewString[abi:cxx11]()'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cholmod_triplet_to_sparse'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `google::LogMessage::stream()'
bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cs_di_cholsol'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

My docker build also has issues, so I currently do not have any fully working build assuming my screenshot above does in fact show a bad UI. I'm getting closer though!

tlalexander, 2020-05-04:
Weird that this issue is back, I thought I fixed it by compiling with gcc7.5... Maybe I'm too tired and made a mistake. I can double check tomorrow.

w-m, 2020-05-04:
> Also I could open a new issue but I'm missing some UI elements I think? Based on a youtube video of this software I think I should have "source/output/interactive/process" tabs on the top right, but I've got nothing. I do not seem to have any way to save output. Thoughts?

Yes, please file an issue, with information on the Qt version you are using.

It's probably just a visibility issue, you should be able to work around it by using the keybindings to switch between the panes, Ctrl+1 through Ctrl+4.