stitchEm/stitchEm #84: Errors during an attempt to build with docker.

tlalexander, 2020-05-04:
Hello,

I've been trying to build with docker. I couldn't get the GPU or nvidia-smi to work using the dockerfiles in this repo, so I started from nvidia's cuda 10.2 ubuntu 18.04 docker image. See my dockerfile here:
https://github.com/tlalexander/stitchEm/blob/master/docker/user_image.dockerfile

All goes well until this error:

```
[454/986] Building CXX object IO/src/exr/CMakeFiles/exrPlugin.dir/src/exrOutput.cpp.o
[455/986] Building CXX object IO/src/portaudio/CMakeFiles/portaudioPlugin.dir/src/portAudioReader.cpp.o
[456/986] Building CXX object IO/src/portaudio/CMakeFiles/portaudioPlugin.dir/src/export.cpp.o
[457/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/aacDecoder.cpp.o
[458/986] Building CXX object IO/src/v4l2/CMakeFiles/v4l2Plugin.dir/src/v4l2.cpp.o
[459/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/audioEncoder.cpp.o
[460/986] Building CXX object IO/src/portaudio/CMakeFiles/portaudioPlugin.dir/src/portAudioWriter.cpp.o
[461/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/lameDecoder.cpp.o
[462/986] Building CXX object IO/src/rtmp/CMakeFiles/MetadataParsingTest.dir/src/metadataParser.cpp.o
[463/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/aacEncoder.cpp.o
[464/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/lameEncoder.cpp.o
[465/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/Nvidia/src/frameQueue.cpp.o
FAILED: IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/Nvidia/src/frameQueue.cpp.o 
/usr/bin/c++  -DSUP_NVDEC -DSUP_NVENC -DUSE_AVFORMAT -DrtmpPlugin_EXPORTS -I_deps/nvenc_api-src/include/ffnvcodec -Ilib/include -Igenerated_headers -Iexternal_deps/include/nvenc -I/usr/local/cuda-10.2/include -IIO/src/rtmp/Nvidia/include -IIO/src/rtmp/include -IIO/src/rtmp/../common/include -IIO/src/rtmp//include -IIO/src/rtmp/../common/format/include -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -g -fPIC   -std=gnu++14 -MD -MT IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/Nvidia/src/frameQueue.cpp.o -MF IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/Nvidia/src/frameQueue.cpp.o.d -o IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/Nvidia/src/frameQueue.cpp.o -c IO/src/rtmp/Nvidia/src/frameQueue.cpp
In file included from IO/src/rtmp/Nvidia/src/frameQueue.cpp:12:0:
IO/src/rtmp/Nvidia/include/frameQueue.hpp:15:21: fatal error: nvcuvid.h: No such file or directory
 #include <nvcuvid.h>
                     ^
compilation terminated.
[466/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/mockDecoder.cpp.o
[467/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/audioDecoder.cpp.o
[468/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/cuvid.cpp.o
FAILED: IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/cuvid.cpp.o 
/usr/bin/c++  -DSUP_NVDEC -DSUP_NVENC -DUSE_AVFORMAT -DrtmpPlugin_EXPORTS -I_deps/nvenc_api-src/include/ffnvcodec -Ilib/include -Igenerated_headers -Iexternal_deps/include/nvenc -I/usr/local/cuda-10.2/include -IIO/src/rtmp/Nvidia/include -IIO/src/rtmp/include -IIO/src/rtmp/../common/include -IIO/src/rtmp//include -IIO/src/rtmp/../common/format/include -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -g -fPIC   -std=gnu++14 -MD -MT IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/cuvid.cpp.o -MF IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/cuvid.cpp.o.d -o IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/cuvid.cpp.o -c IO/src/rtmp/src/cuvid.cpp
In file included from IO/src/rtmp/src/cuvid.cpp:6:0:
IO/src/rtmp/Nvidia/include/frameQueue.hpp:15:21: fatal error: nvcuvid.h: No such file or directory
 #include <nvcuvid.h>
                     ^
compilation terminated.
[469/986] Building CXX object IO/src/portaudio/CMakeFiles/portaudioPlugin.dir/src/portAudioDiscovery.cpp.o
[470/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/mockEncoder.cpp.o
[471/986] Building CXX object IO/src/rtmp/CMakeFiles/MetadataParsingTest.dir/test/orahMetadataParsingTest.cpp.o
[472/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/metadataParser.cpp.o
[473/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/pcmDecoder.cpp.o
[474/986] Linking CXX shared library bin/x64/debug/libvideostitch_cuda.so
[475/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/videoDecoder.cpp.o
[476/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/rtmpDiscovery.cpp.o
[477/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/export.cpp.o
[478/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/x264Encoder.cpp.o
[479/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/videoEncoder.cpp.o
[480/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/nvenc.cpp.o
[481/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/rtmpPublisher.cpp.o
[482/986] Building CXX object IO/src/rtmp/CMakeFiles/rtmpPlugin.dir/src/rtmpClient.cpp.o
ninja: build stopped: subcommand failed.
The command '/bin/sh -c cd /home/developer && cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -G Ninja  . &&     ninja' returned a non-zero code: 1
```

From some google searching it seems nvcuvid.h has gone away in CUDA 10+. See this issue for some discussion.
https://github.com/opencv/opencv_contrib/issues/1786

It is possible to run the GUI programs from docker, so this may be a way to ease installation in some cases, if we can get it to run!

jeremad, 2020-05-04:
Try again with `-DRTMP_NVENC=OFF` in the cmake command, it should fix it.

There a lot of fixes to make to have a real CUDA 10 support...

tlalexander, 2020-05-04:
Hey that did it! I had to remove a line from the dockerfile too. I'll push the updated one after another build finishes. I cleaned it up and want to test it.

Would you like a pull request with the working dockerfile?

jeremad, 2020-05-04:
sure!

tlalexander, 2020-05-04:
Well it looks like I can run VahanaVR but not videostitch-studio.

When I try to run videostitch-studio I get this output:
```
Found 1 device

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-developer'
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
using null output device, none available
ASSERT: "interopCtx->shareContext()" in file apps/src/libvideostitch-gui/videostitcher/stitchercontroller.cpp, line 149
Aborted (core dumped)

```

I ran the container with:
`docker run --gpus all -i -t -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix  stitchem/stitchem-cuda-user:latest bash`

My updated dockerfile is here:
https://github.com/tlalexander/stitchEm/blob/master/docker/user_image.dockerfile

Also I realized that still uses gcc6.

Anyway any thoughts are appreciated!

abhinavkshitij, 2021-04-23:
@tlalexander Could you check the GL libraries in your system? Ideally it should have nvidia drivers or mesa GL drivers, although glvnd should be able to resolve multiple drivers while loading GL libraries. 

https://askubuntu.com/questions/834254/steam-libgl-error-no-matching-fbconfigs-or-visuals-found-libgl-error-failed-t