stitchEm/stitchEm #85: Create user documentation page

w-m, 2020-05-04:
Documentation on how to use the software is currently missing.

Some links we could share:

Vahana VR introduction video:
https://www.youtube.com/watch?v=nR0qD2ELInU

3rd party Vahana tutorials:
https://www.youtube.com/watch?v=rDXbYDr4Pm8
https://www.youtube.com/watch?v=suQvrlTp-iM

VideoStitch Studio PDF user guide:
https://docplayer.net/64441394-Videostitch-studio-v2-3...
--> how do we get the original PDF?

VideoStich Studio video tutorials: https://www.youtube.com/watch?v=mu39B27lLMc...

Maybe we should backup the videos from the Orah/VideoStitch YouTube account, in case this is closed down by Google.

HePTaset, 2020-06-11:
Currently on it. Completely rewrite and rework a PDF documentation and might start with videos soon.

w-m, 2020-06-13:
In the context of the open source project, it would especially useful to have the documentation in the Markdown format. Markdown is edited like raw text, but can have elements like tables, headers, inline images and so forth, and is directly rendered by GitHub when shown. Like the landing page of the project or the https://github.com/stitchEm/stitchEm/blob/master/doc/BUILD.md page.

Having the documentation in Markdown would have the advantage of being able to collaboratively work on it, fix smaller details, update parts of it when features are updated, and so on, all in the same space as the code itself.

HePTaset, 2020-07-10:
New VideoStitch Tutorials here: youtube.com/playlist?list=PLIpVAdb_hI1qR3hpylczvlyITPhreN5rw