stitchEm/stitchEm #86: Issues with prebuilt linux binary

tlalexander, 2020-05-04:
Since I still cannot build this properly on my system I am trying to use the prebuilt binaries from here:
https://github.com/stitchEm/stitchEm/actions/runs/95225592

I am able to run the program, but there are graphical issues and I am also unable to process aka export files. The rest of the flow seems to work.

My system is running Ubuntu 18.04.

My package manager suggests I have qt5 version 5.9.5:
I am not sure which specific qt packages to list but here are some:
```
qttools5-dev/bionic,now 5.9.5-0ubuntu1 amd64 [installed]
qtbase5-dev/bionic-updates,bionic-security,now 5.9.5+dfsg-0ubuntu2.5 amd64 [installed]
```

qtchooser -l gives the following output:
```
4
5
default
qt4-x86_64-linux-gnu
qt4
qt5-x86_64-linux-gnu
qt5
```

Here's what my GUI looks like:

[image]

I was able to switch to the process tab using Ctrl+4, but I cannot get processing to work.
The process log shows:
```
Process is starting.
Process isn't running.
Process failed to start.
```

The terminal output when running the program and attempting to start processing is:
```
Found 1 device

using null output device, none available
QObject::connect: No such slot PostProdStitcherController::setDisplayGrid(bool)
QObject::connect: No such signal PostProdProjectDefinition::reqReextract(SignalCompressionCaps*)
.Missing field 'rig' for BaseProjectDefinition.
[Status] [audio pipeline configuration] Invalid configuration: No audio processor definition found
[ambisonic] Decoder bad coefficients given for unknown

```
I do not know if there are more details log files, or where they would be stored on linux.

Currently I'm finding it quite difficult to run this software on my ubuntu 18.04 machine! Once I get this working, perhaps I could help get an appimage build working or something.

w-m, 2020-05-05:
The processing will hand over the current project file to the `videostitch-cmd` tool that is in the same folder. If you run the `videostitch-cmd -d` contained in the package manually from the command line, does it print your GPU config, or are there any errors running it?

tlalexander, 2020-05-05:
Oh, good to know! videostitch-cmd was not executable. Once I set the executable bit I was able to process video. It works using CPU output. Unfortunately if I select either NVENC codec the program immediately crashes with a segmentation fault.

Given my compilation issues in #72, I have to ask: is it possible to use NVENC with Cuda 10?

And any insight on the broken UI?

Thank you! Happy I can process something now.

w-m, 2020-05-05:
> Given my compilation issues in #72, I have to ask: is it possible to use NVENC with Cuda 10?

Possibly, but consider it broken for the moment.

Our RTMP implementation uses the deprecated framework, so it had to be disabled. But when writing to video files, ffmpeg is used, which could potentially still work.

A lot might have been broken, as we replaced compilers, ffmpeg versions, CUDA versions and more, between the last working VideoStitch build and this open-source build.

I've added information about the crash happening on Linux as well to this ticket: https://github.com/stitchEm/stitchEm/issues/70