stitchEm/stitchEm #88: Crash on processing/export with codec HEVC (Nvidia NVENC)

elsewhat, 2020-05-06:
Related to the awesome 2.4.0 release 

Attempting to export to HEVC H.265 via the codec choice HEVC (Nvidia NVENC).
GPU is 1080TI with driver 445.87

VideoStitch studio crashes. 

The C:\ProgramData\NVIDIA Corporation\ShadowPlay\CaptureCore.log contains the following line which I assume is relevant 
`[videostitch-studio:15660:11500] CSPShareStateManager::Init: Not a game, not initialising!`

Full contents of C:\ProgramData\NVIDIA Corporation\ShadowPlay\CaptureCore.log
```
[06:20:33:26:254][E][videostitch-studio:15660:11500] QueryShadowPlayDdiShimStatusIN
[06:20:33:26:254][R][videostitch-studio:15660:11500] CreateSession_V2: Non profiled app, clear the flags(0x00000001)
[06:20:33:26:254][R][videostitch-studio:15660:11500] CAppSettings::ComputeAppSettings: m_uiIndex: -1 isGame: 0 isLauncher: 0 isAppBlackListed: 0 isEnableHook: 0  isFSDEnabled: 0 isMouseClipEnabled: 0 isDx9FOEnabled: 0 isCAEnabled: 0 m_bDX12ScreenshotWAR: 0
[06:20:33:26:254][R][videostitch-studio:15660:11500] CSPShareStateManager::Init: New MMF opened with id [0]
[06:20:33:26:254][E][videostitch-studio:15660:11500] CSPShareStateManager::Init: Not a game, not initialising!
[06:20:33:26:254][R][videostitch-studio:15660:11500] CShadowPlayProxyShim::ShimCreateSession: MsHybrid = 0
[06:20:33:26:266][R][videostitch-studio:15660:11500] CCursorCompose::CCursorCompose IsProcessDPIAware(1) IsWin8pt1OrGreater(0) IsWin10OrLaterVersionOS(1)
[06:20:33:26:266][R][videostitch-studio:15660:11500] CCursorCompose::RefreshDisplayMetrics IsDPIAware(1) Desktop(5120x1440) SystemMetric(5120x1440)
[06:20:33:26:268][R][videostitch-studio:15660:11500] CShadowPlayProxyShim::ShimCreateSession: OUT
[06:20:33:26:268][R][videostitch-studio:15660:11500] CreateSession_V2: Proxyshim instance (0xec9de8e0), pArgs->ver = 0x70340, pArgs->hDevice = 0x40000200, hLogoRenderer = 0xc0f34b30, hScreenshotCaptureDelegate = 0xc0f2b6e8, pArgs->d3dVersion = -2147483647, pArgs->rendererVersion = 0xc0258, pArgs->shadowPlayFlags = 0x0
[06:20:33:26:268][R][videostitch-studio:15660:11500] CreateSession_V2: Profilename = Base Profile
[06:20:33:29:695][E][videostitch-studio:15660:37852] CSPShareStateManager::Init: Not a game, not initialising!
[06:20:33:29:986][E][videostitch-studio:15660:37852] CSPShareStateManager::Init: Not a game, not initialising!
[06:20:33:29:987][E][videostitch-studio:15660:37852] CSPShareStateManager::Init: Not a game, not initialising!
[06:20:33:30:720][E][videostitch-studio:15660:37852] CSPShareStateManager::Init: Not a game, not initialising!

```
No relevant log lines being written to C:\ProgramData\VideoStitch Studio\VideoStitch Studio.log
```
[2020/05/06 20:39:06] ------------------------------------------------
[2020/05/06 20:39:06] Opening VideoStitch Studio
[2020/05/06 20:39:06] App version: Studio-v2.4.0-win-p1-master.2020-05-02
[2020/05/06 20:39:06] Lib version: v0.0.9-76-g4df2fc1
[2020/05/06 20:39:06] Found 0 translation files
[2020/05/06 20:39:06] Selected device 0
[2020/05/06 20:39:06] CUDA backend selected for stitching
[2020/05/06 20:39:06] Using CUDA backend for stitching
[2020/05/06 20:39:06] Preset file loaded: :/presets/audio/ambisonic-decoding.preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/blending/hard_egdes.preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/blending/linear.preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/blending/multiband.preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/blending/smooth_multiband.preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/calibration/default_studio.preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/mask/default_studio.preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/project/default_input.preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/project/default_project.preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/rigs/360Heroes - H3PRO10HD (1080p).preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/rigs/360Heroes - H3PRO10HD (2.7K).preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/rigs/360Heroes - H3PRO6 (1440p).preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/rigs/360Heroes - H3PRO6 (960p).preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/rigs/360Heroes - H3PRO7 (1440p).preset
[2020/05/06 20:39:06] Preset file loaded: :/presets/rigs/360Heroes - H3PRO7 (960p).preset
[2020/05/06 20:39:08] stitcher controller created
[2020/05/06 20:39:12] Open PTV: C:\dev\ElsewhatGamers\360 experiment\360 fortnite poc v2.ptvb
[2020/05/06 20:39:12] StateManager::changeState, state = 0, sender = MainWindow
[2020/05/06 20:39:13] Missing field 'rig' for BaseProjectDefinition.

[2020/05/06 20:39:13] [Status] [audio pipeline configuration] Invalid configuration: No audio processor definition found

[2020/05/06 20:39:14] [ambisonic] Decoder bad coefficients given for stereo

[2020/05/06 20:39:14] StateManager::changeState, state = 2, sender = MainWindow
[2020/05/06 20:39:14] [AudioPlayer] Audio data is not being fed to the audio device at a fast enough rate, cannot play audio.
```

PS same issue if I export to code H264 (NVIDIA NVENC)

elsewhat, 2020-05-06:
If I turn off Shadow Play recording in Geforce experience, the log entries in C:\ProgramData\NVIDIA Corporation\ShadowPlay\CaptureCore.log do not appear any more. However, the crash still happens.

Dump data in AppData\Local\CrashDumps shows the following in visual studio
[image]

w-m, 2020-05-06:
Thank you for the report. Issue related to #70.

Does it crash for you immediately when you select the output in Studio, or when you start exporting?

elsewhat, 2020-05-06:
It crashes immediately when the output is selected

PR #95 (2020-05-19): Disable NVENC output in Studio
NVENC output feature was in development, not ready to be shipped. No UI currently for Studio, the codec factory doesn't deal with the NVENC variants, a nullptr is created, and Studio crashes.

We can have a look at finishing the feature at some point, but for now, remove the Studio crash from the unfinished code.

Closes #88.
closed 2020-05-19 by w-m