stitchEm/stitchEm #89: Does it work with 2 cameras / couldn't find a forum

KArlkackt, 2020-05-07:
Sorry to use the issue system as a way to ask my question, but i havent found a forum where I could ask my question. I try to use the software to stitch footage from my gear 360, that has to lenses to capture 360 video and can't get it to work. Does the software work with only to lenses, or do I need 6 viewports to stitch a video.

So my main issue is, that there is no forum to ask this sort of questions / not enough info on the github page to answer my question.

w-m, 2020-05-07:
The number of inputs in a project is not fixed to 6, you can stitch together all kinds of different configurations.

To calibrate a two-lens fisheye-camera you would select "Circular fisheye" lens type in the calibration window and "Adjust inputs crop", manually cropping away the black bars. If you have both inputs next to each other in a single video stream, you would need to import this video twice, and crop the left part in the first input, and the right part in the second input.

The automatic calibration in VideoStitch Studio works with landmarks automatically detected in overlapping image regions. There is no the option to manually adjust these. I'm not sure how it goes in practice for the Gear 360, but I would assume there's a high likelihood of the automatic calibration failing, due to insufficient overlap.

In the case of unsuccessful automatic calibration with VideoStitch Studio, you can do the calibration with PTGui (a paid software, though), and import this calibration as a template into VideoStitch. The calibration can probably be reused for all your projects with the same camera.

Try to download a PTGui template for your camera make, or follow online tutorials on how to calibrate with PTGui.

HePTaset, 2020-06-21:
If you can provide footage (several scenes with completely different stitching conditions (close objects only, far objects only, mixed) I will create a preset, as I'm currently working on presets for most cams.

w-m, 2020-06-28:
As for the question about the forum, there is https://www.facebook.com/groups/stitchEm now to discuss and get help from the community!

closed 2020-06-28 by w-m