stitchEm/stitchEm #90: Best camera setup when creating 360 videos from games like fortnite

elsewhat, 2020-05-08:
So I am experimenting with creating 360 videos from games like fortnite. In fortnite (and many other games) you are able to enter a replay mode where you can control the camera. You can then play the replay from that camera angle and record with a software like OBS. Then rewind, position the camera differently and record again.

There are a 2-3 youtube channels producing such 360 game videos, such as [Vicinity360](https://www.youtube.com/channel/UCqOVnEltnqEein52CE9Zv3g/featured), but they are not sharing which software they are using and what camera setup they have. 

The question is what is the best approach to position the camera and how many recordings are ideal to perform the stitching in Video stitch.

In fortnite you are able to set focal length to 11mm, which I believes corresponds to 120 horizontal FOV. With a 16:9 aspect ration the vertical FOV is 90 degrees (ref https://themetalmuncher.github.io/fov-calc/). There is no way to exactly position a camera in the replay, so you need to find a reference point and afterwards rotate the camera through mouse movement.

I've done a couple attempts with a standard skybox 6 camera recording from front,left,back,right, top and bottom. Here Video stitch has issues finding enough common points between from the top /bottom cameras to the other ones. 
Here is one video example where all cameras except top were well connected. Here the camera is in a fixed location (called drone free in fortnite)
https://youtu.be/wIneaHvpRwg
(synchronization between cameras is slightly off, but that's solveable)

I believe the reason why this approach is not ideal, is because the vertical FOV is 90 degrees and then the overlap is minimal/non-existing.

Here is a screenshot from another 6 camera recording (this one is actually using 16mm focal length camera so it can be slightly better with 11mm). Here the camera is attached to the heli (called drone attached) using a common reference point but with different camera rotations.
[image]
The screenshot is one of the few cases where I could visually see a connection between the bottom camera (nr 2) and the left camera, but this was not detected by Video stitch (and I wouldn't expect it to either).

I believe setting up a camera calibration (with PTGui) would not work in practice. This because it is not feasible to "place" the camera at the exact location and with the exact rotation needed (since it's based on keyboard wsad and mouse movements)

What other alternatives are there to the camera setup?
 
Best idea I have is a 8 camera setup which I have not yet tested.
Four with y-rotation -45 degrees: Front, left, back, right
Four with y-rotation +45 degrees: Front, left, back, right

Any other ideas?

w-m, 2020-05-11:
> There are a 2-3 youtube channels producing such 360 game videos, such as Vicinity360, but they are not sharing which software they are using and what camera setup they have.

I'd try writing to them to ask for their setup. Maybe they'd be happy to share it with you.

> There is no way to exactly position a camera in the replay, so you need to find a reference point and afterwards rotate the camera through mouse movement.

That seems unnecessarily hard to do. If you're a little bit off with the position, you'll get stitching artefacts in the output. I haven't though through your use case much and I don't know about Fortnite in particular, but I'd look about ways to set these programatically. Either through the game's console (e.g. CS:GO has `spec_pos` and `spec_goto` and stuff like that) or through modding, or through just having some tool that repeats the same keystrokes, to end up in the exact same place.

> I believe the reason why this approach is not ideal, is because the vertical FOV is 90 degrees and then the overlap is minimal/non-existing.

Create larger overlap by adding more cameras, and position your cameras in a way that every camera has image features in, not just the sky. There's no requirement to have some cameras on the horizon level, you can pitch up and down, to cover the sky/the ground with multiple cameras, while still having enough image content to calibrate.

> The screenshot is one of the few cases where I could visually see a connection between the bottom camera (nr 2) and the left camera, but this was not detected by Video stitch (and I wouldn't expect it to either).

In the calibration pane, choose "manual frame selection" and add a couple of frames from different points in time where you can visually see the connections. Automatic frame selection for calibration could miss those.

> I believe setting up a camera calibration (with PTGui) would not work in practice. This because it is not feasible to "place" the camera at the exact location and with the exact rotation needed (since it's based on keyboard wsad and mouse movements)

If you can set the cameras programatically, you can also manually set their parameters in the .ptv JSON file.

elsewhat, 2020-05-12:
Thanks for the tips. 

In the meantime I have been able to get some more experience. 
https://www.youtube.com/watch?v=MBML3qfCsck

This was filmed with 16:10 aspect ratio, 10mm camera focal length and 5 video captures. 
The video captures where rotate along the horizontal axis with vertical rotation alternating between +30/-30. I had a good reference point for this video. 

There is one minor uncovered area, but this could also be because of inaccuracies in the setup (especially the rotation). Used PTGui for the calibration and got down to about 3.0 in average control point distance.

Tested one more video
https://www.youtube.com/watch?v=ZmDEI6GNmkM&feature=youtu.be

Here I had 4 videos along the horizontal axis (at 0, 90, 180 and 270 degrees), and 2 for the top and 2 for the bottom. In this case the alignment was not as successful(sound is also off, but that came from the OBS screencapture). It has 14.0 in average control point distance and visual "skewed" shadows in some stitches. This skewing is not caused by synchronization (as I had visual cues to synch exact to an exact frame for all cameras), but I believe it's an effect parallax. 

One of the main reasons for the parallax is that it appears the camera does not rotate around the lens, but instead of the main invisible body of it. This causes parallax issues ref PTGuis introduction video https://youtu.be/O2JaDKSOtwY?t=170

After your mail I investigate a bit around console, mods and hidden commands for placing the camera exactly. Console and mods are very difficult in fortnite, but[ found a "hidden" command ](https://www.epicgames.com/fortnite/en-US/news/fortnite-battle-royale-replay-system)for PC only.
Ctrl+Numpad# - Stores current camera type/position, and selected player
Ctrl+Alt+Numpad# - Stores current camera type/position, selected player AND position on the timeline

The last one would be ideal, but actually crashes the game in the current version. But the first one works and together with a visual cue (such as a build placement or shot) it should be sufficient.

My next experiment would be a 6 camera setup with  16:10 aspect ratio and 10mm camera focal length. Rotate along the horizontal axis(0,60,120,180,240,300 - Use minimap to quality ensure the right rotation) and  at +- 30 degrees for vertical rotation. 

If ok with you, I'll leave this issue open for a couple of weeks to share my experiences before closing as it may be interesting for others.

w-m, 2020-06-28:
Feel free to take the dicsussion to https://www.facebook.com/groups/stitchEm, where you could get more feedback from the community of users!

closed 2020-06-28 by w-m