stitchEm/stitchEm #93: the algorithm principle

15757170756, 2020-05-14:
Thanks for your source code with video stitch。The code library has lots of code，and I can't understand the algorithm principle in this code。So is there any paper or flow chart that can explain how exactly the code work。
Thank you very much！

w-m, 2020-05-14:
There are currently no technical overview documents. I thought about writing up some parts of it, is there anything you would be interested in, in particular?

The general idea of the stitching part is to set up blending masks for all the inputs in advance. Then for every pixel in the output image of every frame, map this pixel's coordinates to a sphere, and back into the input camera coordinates. Apply lens distortion, fetch the color value from the input image at the mapped point and apply photometric corrections. Blend inputs together with the weights from precomputed blending masks.

The main GPU kernel for that mapping (called warpKernbel) is here: https://github.com/stitchEm/stitchEm/blob/master/lib/src/backend/common/core1/warpKernel.gpu.incl - but it will be quite hard to follow around the different variants, there's lots of C macro fake-templating going on for all the different types of distortions, output and input projections.

Most of the transformation code from camera to sphere, from output to sphere, distortions, and so on, are found here: https://github.com/stitchEm/stitchEm/blob/master/lib/src/backend/common/core/transformStack.gpu

If you want to step through the C++ code, this is the main entry point for every frame stitched: https://github.com/stitchEm/stitchEm/blob/master/lib/src/core/controller.cpp#L734

15757170756, 2020-05-19:
thanks for your technical guidance！

closed 2020-06-28 by w-m