stitchEm/stitchEm #100: OpenCL 1.2

RoyWiggins, 2020-05-19:
I have an older GPU, and VideoStitch Studio is failing to build a GPU program, so it's not usable. Am I correct and it requires OpenCL >2.0 support on the GPU? Might be worth putting that in the docs somewhere in case someone with a several years old GPU comes along. 

```
[2020/05/19 19:02:05] Missing field 'rig' for BaseProjectDefinition.
[2020/05/19 19:02:16] OpenCL failed to build program downsampler
[2020/05/19 19:02:16]
[2020/05/19 19:02:16] "D:/dev/stitchEm/lib/src/backend/cl/image/downsampler.cl", line 6: warning: unrecognized #pragma #pragma clang diagnostic push ^ "D:/dev/stitchEm/lib/src/backend/cl/image/downsampler.cl", line 7: warning: unrecognized #pr
[2020/05/19 19:02:16] agma #pragma clang diagnostic ignored "-Wcast-align" ^ "D:/dev/stitchEm/lib/src/backend/cl/../..\backend/common/image/downsampler.gpu", line 22: error: a type qualifier is not allowed kernel void downsampleRGBASurfKernel(write_onl
[2020/05/19 19:02:16] y image2d_t dst, const read_only image2d_t src, unsigned dstWidth, ^ "D:/dev/stitchEm/lib/src/backend/cl/image/downsampler.cl", line 11: warning: unrecognized #pragma #pragma cl
[2020/05/19 19:02:16] ang diagnostic pop ^ 1 error detected in the compilation of "C:\Users\Roy\AppData\Local\Temp\OCL9996T1.cl". Frontend phase failed compilation.
[2020/05/19 19:02:16] [Status] [GPU] Runtime error: Build program failure
```

w-m, 2020-05-20:
VideoStitch Studio only requires OpenCL 1.2 with texturing support. Sometimes older GPU drivers will have problems though, mostly by not correctly implementing the 1.2 standard.

What GPU are you using, on what OS? Can you make sure that you have the latest available driver for this GPU?

Because it's a little hard to parse, here is the extracted error message again (the pragma warnings can be ignored):

```
"D:/dev/stitchEm/lib/src/backend/cl/../..\backend/common/image/downsampler.gpu", line 22: error: a type qualifier is not allowed kernel void downsampleRGBASurfKernel(write_only image2d_t dst, const read_only image2d_t src, unsigned dstWidth,
```

The OpenCL kernel compiler probably has an issue with the `write_only` or `read_only` qualifiers.

RoyWiggins, 2020-05-20:
It's a Radeon R9 280x (aka "Tahiti", apparently), on Windows 10. I only recently rebuilt the OS, so I think I have a pretty recent driver (I can see it listed as "Adrenalin 19.20", elsewhere "2841.19"). It does look like it's the read_only/write_only qualifier that's tripping it up, but those are both specified in the [opencl 1.2 spec](https://www.khronos.org/registry/OpenCL/specs/opencl-1.2.pdf) (page 228). 

GPUs are pretty well a black box to me, so I have limited experience with debugging these things. It's a weird one, "a type qualifier is not allowed" _barely_ exists when googled. GPU Caps Viewer reports my device running OpenCL 1.2 and "CL_DEVICE_IMAGE_SUPPORT: YES", so I should be fine there...

RoyWiggins, 2020-05-20:
Okay, so I don't have a *good* way to compile GPU code, but I realized that the GPU Caps Viewer is bundled with demos, and those demos have their code loose, so I can just try throwing some code in there and see if I can reproduce the error. Which I can. The read_only is not the problem, it's the `const image2d_t `. 

`void test(const read_only image2d_t src ) {}` 

```
CL kernel compil error: "C:\Users\Roy\AppData\Local\Temp\OCL12564T9.cl", line 231: error: a type
          qualifier is not allowed
  void test(const read_only image2d_t src ) {}
            ^
```

`void test(const image2d_t src ) {}`
```
CL kernel compil error: "C:\Users\Roy\AppData\Local\Temp\OCL12564T10.cl", line 231: error: a type
          qualifier is not allowed
  void test(const image2d_t src ) {}
            ^
```
Compiles without complaint:
`void test(image2d_t src ) {}`
`void test(read_only image2d_t src ) {}`
`void test(__read_only image2d_t src ) {}`
`void test(const float src ) {}`

RoyWiggins, 2020-05-20:
Oh, [here we are](https://www.khronos.org/registry/OpenCL/sdk/1.1/docs/man/xhtml/qualifiers.html):

>  The type qualifiers const, restrict and volatile as defined by the C99 specification are supported. These qualifiers cannot be used with image2d_t or image3d_t type. Types other than pointer types shall not use the restrict qualifier.

w-m, 2020-05-20:
Oh, good find, seems indeed to be the `const`! We can just remove the const there, also another one in `sphereSweep.gpu`, these two seem to be the only occurrences in the codebase.

RoyWiggins, 2020-05-20:
I tried removing the const by hand-patching `libvideostitch_opencl.dll` and it has stopped complaining and is able to open a file.

RoyWiggins, 2020-05-20:
This is off-topic, but: is it possible to use this software to stabilize video that's already been stitched into an equirectangular projection? The video I'm working with comes out of a Ricoh Theta, which I've already stitched using the tool it came with, but it doesn't have any image stabilization at all. 

The format out-of-the-camera is a sort of twin fisheye (one video, two fisheye images). I don't think this software supports stitching those together, does it?

w-m, 2020-05-21:
> I tried removing the const by hand-patching libvideostitch_opencl.dll and it has stopped complaining and is able to open a file.

Nice! Could you do a pull request where you remove the two occurrences in the code base?

> This is off-topic, but: is it possible to use this software to stabilize video that's already been stitched into an equirectangular projection?

No, that's not a feature of the software.

Adding it to the code would mean properly supporting equirectangular inputs (which might at the moment only be supported half-accidentally in the .ptv files, and not at all in the UI), and possibly adapting the stablization algo (https://github.com/stitchEm/stitchEm/blob/master/lib/src/stabilization/rotationStabilization.cpp) to deal with a single pre-stitched input.

> The format out-of-the-camera is a sort of twin fisheye (one video, two fisheye images). I don't think this software supports stitching those together, does it?

That's possible, you'll have to import the video twice and use the fisheye cropping tool. I've already written a bit about two-lens cameras here: https://github.com/stitchEm/stitchEm/issues/89

PR #115 (2021-03-02): Don't use const on image2d_t
The type qualifiers const, restrict and volatile as defined by the C99 specification are supported. These qualifiers cannot be used with image2d_t or image3d_t type. Types other than pointer types shall not use the restrict qualifier.

Fixes #100
closed 2021-03-02 by w-m