stitchEm/stitchEm #105: IDEA: Remove Output configuration

HePTaset, 2020-06-11:
The window is relatively useless. The two options are better of in other, related windows:

"Show input numbers" would be better of in the preferences or a small icon somewhere to quickly enable and disable.

"Blending" belongs in the Process panel.

w-m, 2020-06-13:
> "Show input numbers" would be better of in the preferences or a small icon somewhere to quickly enable and disable.

Yeah, it could also live in the calibration pane.

> "Blending" belongs in the Process panel.

This can't move to the process panel, the visual difference in output between multiband blending and linear blending, as well as the feathering strength need to be checked in the output. The process panel being full screen takes away the output preview.

HePTaset, 2020-06-13:
>Yeah, it could also live in the calibration pane.

Not just calibration - use it in sync all the time to manually fix the snc if I see ghosting resulting from badly synced footage. Maybe somewhere in the main UI to quickly turn on/off?

>This can't move to the process panel, the visual difference in output between multiband blending and linear blending, as well as the feathering strength need to be checked in the output. The process panel being full screen takes away the output preview.

Ah, you are right! "Blender" selection could be shown in the process panel, too though.