stitchEm/stitchEm #113: IDEA: Add Numerical Transform Option

HePTaset, 2020-06-25:
Editing the Orientation can still be a pain if your panorama is already perfect in one or two directions.
(e.g. the horizon is perfect already, but you want to yaw it 90 degrees)

Adding a Numerical Transform (similar to the one in PTGui) would certainly help a lot!

HePTaset, 2020-06-25:
[image]

w-m, 2020-06-28:
You could show the numerical values for yaw/pitch/roll in the Stabilization & Orientation pane in editable text fields, over the button to "Adjust on current frame". The current button would have to be renamed to reflect that it should be pressed to change the orientation through mouse-dragging, while you could change it numerically with the new text fields.