stitchEm/stitchEm #114: Progress bar when rendering a video overcounts and is generally off

HePTaset, 2020-06-27:
Export seems to work fine, but for a 500 frame video it went up to ~590Frames. (The % bar seems to be off, too.)
The log seems to show the correct values though!

Seems to be a minor GUI issue.

w-m, 2020-06-28:
Is the progress bar off for all projects/exports, or just for some peculiar inputs?

Sometimes the metadata in the input videos can be off, so ffmpeg/libav reports the input videos being shorter than they actually are. But when exporting, VideoStitch Studio will keep stitching frames as long as there are frames coming from the inputs, so it will go beyond the original estimate of number of frames.

If it happens consistently for all kinds of inputs, it has to be a bug in the progress bar logic / communication between batch stitcher and videostitch-cmd output.

HePTaset, 2020-06-28:
Okay I checked with several inputs - while it was still processing it was always **times two as the real output**. 
[image]

_Don't know how I saw the 590 Frames yesterday, couldn't reproduce it._

w-m, 2020-06-28:
The output progress bar counts the lines in the log that start with "stitched frame": https://github.com/stitchEm/stitchEm/blob/master/apps/src/libvideostitch-gui/mainwindow/vscommandprocess.cpp#L58-L63

They are emitted by the frameBuffer in libvideostitch here: https://github.com/stitchEm/stitchEm/blob/master/lib/src/core/stitchOutput/frameBuffer.hpp#L75-L85

They are indeed emitted twice per frame, if I run videostitch-cmd with a sample project (with -v 3), I get something like

```
stitched frame at 633967 useconds  : 9 ms
stitched frame at -1 useconds  : 11 ms
```

for every frame.

-> Investigate why `FrameBuffer::streamSynchronize` is called twice