stitchEm/stitchEm #118: "No CUDA capable GPU detected on your system"

frostworx, 2020-07-20:
I built the binary unter linux (Radeon VII, amdgpu) with
cmake -DGPU_BACKEND_CUDA=OFF -DGPU_BACKEND_OPENCL=ON
but when starting the binary ./VahanaVR with pwd in bin/x64/debug  I only get the requester
"No CUDA capable GPU detected on your system"
Is this supposed to work?
I've seen the closed issue https://github.com/stitchEm/stitchEm/issues/24
but
`./videostitch-cmd -d`
only shows 
```
-d option used without parameter, list compatible GPU devices:
Could not find any compatible GPU device on this computer.
Host OpenCL allocated memory:
Device OpenCL allocated memory:
```

w-m, 2020-07-27:
(General heads-up: Vahana VR was only tested with CUDA and mostly used on Windows. Vahana VR on Linux with OpenCL are untested waters.)

Is it possible that you have several OpenCL SDKs installed? The issue you are seeing could stem from linking to the OpenCL SDK of a different vendor (e.g. Intel), which then at runtime won't find your AMD GPU. Please have a look at the CMake configuration output and check the OpenCL path.

If you are sure that the correct OpenCL SDK is linked, you could step through the OpenCL discovery code (https://github.com/stitchEm/stitchEm/blob/master/discovery/src/device.cpp#L358) with a debugger, to find where it's going wrong.

frostworx, 2020-07-28:
thanks for trying to help!
I just rechecked this and I only have one opencl-sdk installed. maybe opencl-mesa is not feature complete for this?
the output of clinfo looks like this (nothing about spir, but setting DISABLE_OPENCL_SPIR doesn'T fix this as well, so this might be unrelated):

```
 clinfo
Number of platforms                               1
  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 20.1.4
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   Clover
Number of devices                                 1
  Device Name                                     AMD Radeon VII (VEGA20, DRM 3.37.0, 5.7.10-zen1-1-zen, LLVM 10.0.0)
  Device Vendor                                   AMD
  Device Vendor ID                                0x1002
  Device Version                                  OpenCL 1.1 Mesa 20.1.4
  Driver Version                                  20.1.4
  Device OpenCL C Version                         OpenCL C 1.1 
  Device Type                                     GPU
  Device Profile                                  FULL_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Max compute units                               60
  Max clock frequency                             1801MHz
  Max work item dimensions                        3
  Max work item sizes                             256x256x256
  Max work group size                             256
  Preferred work group size multiple              64
  Preferred / native vector sizes                 
    char                                                16 / 16      
    short                                                8 / 8       
    int                                                  4 / 4       
    long                                                 2 / 2       
    half                                                 0 / 0        (n/a)
    float                                                4 / 4       
    double                                               2 / 2        (cl_khr_fp64)
  Half-precision Floating-point support           (n/a)
  Single-precision Floating-point support         (core)
    Denormals                                     No
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 No
    Round to infinity                             No
    IEEE754-2008 fused multiply-add               No
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  No
  Double-precision Floating-point support         (cl_khr_fp64)
    Denormals                                     Yes
    Infinity and NANs                             Yes
    Round to nearest                              Yes
    Round to zero                                 Yes
    Round to infinity                             Yes
    IEEE754-2008 fused multiply-add               Yes
    Support is emulated in software               No
  Address bits                                    64, Little-Endian
  Global memory size                              17179869184 (16GiB)
  Error Correction support                        No
  Max memory allocation                           13743895347 (12.8GiB)
  Unified memory for Host and Device              No
  Minimum alignment for any data type             128 bytes
  Alignment of base address                       32768 bits (4096 bytes)
  Global Memory cache type                        None
  Image support                                   No
  Local memory type                               Local
  Local memory size                               32768 (32KiB)
  Max number of constant args                     16
[... 10 lines cut ...]

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  Clover
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [MESA]
  clCreateContext(NULL, ...) [default]            Success [MESA]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 Clover
    Device Name                                   AMD Radeon VII (VEGA20, DRM 3.37.0, 5.7.10-zen1-1-zen, LLVM 10.0.0)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 Clover
    Device Name                                   AMD Radeon VII (VEGA20, DRM 3.37.0, 5.7.10-zen1-1-zen, LLVM 10.0.0)
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 Clover
    Device Name                                   AMD Radeon VII (VEGA20, DRM 3.37.0, 5.7.10-zen1-1-zen, LLVM 10.0.0)

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.12
  ICD loader Profile                              OpenCL 2.2

```

w-m, 2020-07-28:
Oh, this reports 1.1 for the OpenCL device version for some reason, which is ancient. We check that at least OpenCL 1.2 is supported (https://github.com/stitchEm/stitchEm/blob/master/discovery/src/device.cpp#L327-L328).

frostworx, 2020-07-28:
why did I miss that (I only saw the last line with OpenCL 2.2 ...) ? thanks for taking a look and sorry for the noise...

frostworx, 2020-07-28:
me again.
I installed the proprietary opencl-amd drivers (with my Radeon VII the opencl-amdgpu-pro-pal version will be used) and without restarting X or even the whole system stitchEm starts correctly! maybe this is stable enough to keep it installed aside the opensource drivers.

w-m, 2020-07-28:
Great! Seems like we can close the issue then.

closed 2020-07-28 by w-m