stitchEm/stitchEm #121: Jetson Nano cmake Errors part deux

aspentreemedia, 2020-09-02:
Sorry I got sidetracked with the whole quarantine thing and I'm just now getting back to my nano experiment... I've gotten the latest stitchEm and for good measure I started over with a fresh nano install. However now I'm running into this cmake error:

mike@nano-desktop:~/Documents/Development/stitchEm$ cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DDISABLE_RTMP=ON -DCUDA_LOCAL_ARCH_ONLY=ON -DCMAKE_BUILD_TYPE=Release -DDISABLE_TIFF=ON -G Ninja stitchEm

-- GPU_BACKEND_CUDA='ON'
-- GPU_BACKEND_OPENCL='OFF'
-- USE_DELAY_LOAD=OFF
-- System: LINUX
-- CMake Build Type -- Release
-- CUDA VERSION: 10.2
-- CUDA PATH: /usr/local/cuda-10.2
-- CUDA: /usr/lib/aarch64-linux-gnu/libcuda.so
-- CUDART: /usr/local/cuda-10.2/lib64/libcudart.so
-- CUDA_LIBRARIES: /usr/local/cuda-10.2/lib64/libcudart_static.a;Threads::Threads;dl;/usr/lib/aarch64-linux-gnu/librt.so
-- Using /usr/local/cuda-10.2/bin/nvcc to query CUDA device properties
CMake Error at cmake/CUDAArch.cmake:24 (if):
  if given arguments:

    "(" "QUERY_CUDA_EXIT" "EQUAL" "1" ")" "OR"

  Unknown arguments specified
Call Stack (most recent call first):
  cmake/CUDA.cmake:119 (include)
  CMakeLists.txt:405 (include)

-- Configuring incomplete, errors occurred!
See also "/home/mike/Documents/Development/stitchEm/CMakeFiles/CMakeOutput.log".
See also "/home/mike/Documents/Development/stitchEm/CMakeFiles/CMakeError.log".

Thanks,

Mike

w-m, 2020-09-03:
What CMake version are you running?

I'm not sure why this doesn't work, I thought this to be correct CMake code, and the variable would always exist after executing the process. Please try whether it works when replacing line 24 in cmake/CUDAArch.cmake with

`if((${QUERY_CUDA_EXIT} EQUAL 1) OR ${QUERY_CUDA_ERR})`

(additionally resolving the `QUERY_CUDA_EXIT` variable).

aspentreemedia, 2020-09-03:
I had to upgrade cmake, it initially complained. I'm currently running:
cmake version 3.18.2

When I changed line 24 and tried again I get this:

-- GPU_BACKEND_CUDA='ON'
-- GPU_BACKEND_OPENCL='OFF'
-- USE_DELAY_LOAD=OFF
-- System: LINUX
-- CMake Build Type -- Release
-- CUDA VERSION: 10.2
-- CUDA PATH: /usr/local/cuda-10.2
-- CUDA: /usr/lib/aarch64-linux-gnu/libcuda.so
-- CUDART: /usr/local/cuda-10.2/lib64/libcudart.so
-- CUDA_LIBRARIES: /usr/local/cuda-10.2/lib64/libcudart_static.a;Threads::Threads;dl;/usr/lib/aarch64-linux-gnu/librt.so
-- Using /usr/local/cuda-10.2/bin/nvcc to query CUDA device properties
CMake Error at cmake/CUDAArch.cmake:25 (if):
  if given arguments:

    "(" "53" "EQUAL" "1" ")" "OR"

  Unknown arguments specified
Call Stack (most recent call first):
  cmake/CUDA.cmake:119 (include)
  CMakeLists.txt:405 (include)

aspentreemedia, 2020-09-03:
Here's the cmake error I received about my version not being compatible:

CMake Error at CMakeLists.txt:11 (cmake_minimum_required):
  CMake 3.11 or higher is required.  You are running version 3.10.2

So I updated it to 3.18.2

w-m, 2020-09-03:
Feel free to comment out lines 24-26 to make progress on your build, as the nvcc execution is working and you won't need to print the error message.

I'm still not sure why CMake complains about this `if` statement. If you can experiment a little with the code and provide a variant that is working for you of lines 24-26, a pull request would be appreciated.

aspentreemedia, 2020-09-04:
I'll take a stab at the if statement, I did comment out just so I could get going. It builds successfully but then when I do the ninja command I get the following:

mike@nano-desktop:~/Documents/Development/stitchEm$ ninja
ninja: error: 'stitchEm/apps/src/videostitch-studio-gui/presets/rigs/YI360.preset.preset', needed by 'apps/src/videostitch-studio-gui/videostitch-studio_autogen/EWIEGA46WW/qrc_videostitch-studio-gui.cpp', missing and no known rule to make it

To get around this I found in 
/stitchEm/apps/src/videostitch-studio-gui/videostitch-studio-gui.qrc

There was an extra .preset in:
<file>presets/rigs/YI360.preset</file>

Once I took out the extra .preset and ran ninja I got further in to the build process but then ran into this:
40/959] Building CXX object lib/CMake...dir/src/audio/orah/orahAudioSync.cpp.o
stitchEm/lib/src/audio/orah/orahAudioSync.cpp: In member function ‘void VideoStitch::Audio::Orah::OrahAudioSync::findOffset()’:
stitchEm/lib/src/audio/orah/orahAudioSync.cpp:219:9: warning: this statement may fall through [-Wimplicit-fallthrough=]
         }
         ^
stitchEm/lib/src/audio/orah/orahAudioSync.cpp:222:7: note: here
       case SyncState::SYNCHRONIZED:  // and UNSYNCHRONIZED...
       ^~~~
[61/959] Building CXX object lib/CMake...core/defs/inputDefinitionUpdater.cpp.o
FAILED: lib/CMakeFiles/vslib-objects-cuda.dir/src/core/defs/inputDefinitionUpdater.cpp.o 
/usr/bin/c++ -DVS_CUDA -I/usr/local/cuda-10.2/include -IstitchEm/lib/src -IstitchEm/discovery/include -IstitchEm/lib/include -Igenerated_headers -Ilib/json -I/usr/include/eigen3 -isystem stitchEm/external_deps/include/libpng -isystem stitchEm/external_deps/include/glm -isystem stitchEm/external_deps/include -isystem _deps -isystem /usr/include/opencv4 -isystem /usr/include/openexr -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-function -Wno-missing-field-initializers -O3 -DNDEBUG -fPIC -std=gnu++14 -MD -MT lib/CMakeFiles/vslib-objects-cuda.dir/src/core/defs/inputDefinitionUpdater.cpp.o -MF lib/CMakeFiles/vslib-objects-cuda.dir/src/core/defs/inputDefinitionUpdater.cpp.o.d -o lib/CMakeFiles/vslib-objects-cuda.dir/src/core/defs/inputDefinitionUpdater.cpp.o -c stitchEm/lib/src/core/defs/inputDefinitionUpdater.cpp
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
[66/959] Building CXX object lib/CMake...e/defs/panoramaDefinitionUpdater.cpp.o
ninja: build stopped: subcommand failed.

w-m, 2020-09-04:
The fall through is intended there (but we can't mark It to tell the compiler, using C++14 at the moment, `[[fallthrough]` annotation is supported from C++17).

You seem to be using gcc? Then add `-Wno-implicit-fallthrough` to this line:
https://github.com/stitchEm/stitchEm/blob/master/CMakeLists.txt#L339