stitchEm/stitchEm #125: Calibration and Stabilization controls greyed out in  VideoStich Studio

jumpjack, 2020-10-31:


fpga68k, 2021-01-21:
same here - makes the program useless

w-m, 2021-01-21:
Please add more details: what platform are you using, what inputs have you loaded in the project?

fpga68k, 2021-01-21:
I tried Studio-v2.4.0-p1-master on Win10 x64 with a single file from Xiaomi Mija Mi Sphere 360 (two fisheye back to back, side by side in input video)
In Calibration Window the Rig selection the Rig is Custom and the Lensetype is Fullframe fisheye (which is wrong). It is not possible to change these settings.
Hope that helps.

w-m, 2021-01-21:
The software doesn't directly support multiple views from a single input video. To make this kind of media work, add the same input video twice. Then use the crop tools to crop out the left view from the first input source, and the right view from the second one.

fpga68k, 2021-01-22:
Oh, OK so more a user error (as a result of a misleading GUI). When adding file twice I can crop left and right side. But I still can't stich them together. Output look still like distorted two fisheye sidebyside. But hat might be different user error.

w-m, 2021-01-22:
Try following along the user guide about calibration: https://github.com/stitchEm/stitchEm/blob/master/doc/user-guide/user-guide.md#calibration

You can also post questions to the Facebook users group: https://www.facebook.com/groups/stitchEm

closed 2021-02-01 by w-m