stitchEm/stitchEm #129: Some errors on Xavier

wentao0614-wang, 2020-12-15:
Hello, I try to install your stitchEm on NVIDIA AGX Xavier platform. My system is Ubuntu 18.04, CUDA version is 10.2, cudnn version is 8.0, opencv version is 3.3.1, QT version is 5.9.5. 
This is my cmake information:

nvidia@jetson-0423718018083:~/stitchEm$ cmake -DGPU_BACKEND_CUDA=ON -DRTMP_NVENC=OFF -DGPU_BACKEND_OPENCL=OFF -DDISABLE_TIFF=ON -G Ninja
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

-- GPU_BACKEND_CUDA='ON'
-- GPU_BACKEND_OPENCL='OFF'
-- USE_DELAY_LOAD=OFF
-- System: LINUX
-- CMake Build Type -- Debug
-- CUDA VERSION: 10.2
-- CUDA PATH: /usr/local/cuda-10.2
-- Found CUDA: /usr/local/cuda-10.2 (found suitable exact version "10.2") 
-- CUDA: /usr/lib/aarch64-linux-gnu/libcuda.so
-- CUDART: /usr/local/cuda-10.2/lib64/libcudart.so
-- CUDA_LIBRARIES: /usr/local/cuda-10.2/lib64/libcudart_static.a;Threads::Threads;dl;/usr/lib/aarch64-linux-gnu/librt.so
-- NVCC arch flags: -gencode=arch=compute_30,code=sm_30;-gencode=arch=compute_35,code=sm_35;-gencode=arch=compute_50,code=sm_50;-gencode=arch=compute_52,code=sm_52;-gencode=arch=compute_61,code=sm_61;-gencode=arch=compute_61,code=compute_61
-- System: LINUX
-- --- VideoStitch-lib dependencies ---
-- OpenEXR libs: /usr/lib/aarch64-linux-gnu/libHalf.so;/usr/lib/aarch64-linux-gnu/libIex.so;/usr/lib/aarch64-linux-gnu/libImath.so;/usr/lib/aarch64-linux-gnu/libIlmImf.so;/usr/lib/aarch64-linux-gnu/libIlmThread.so
-- OpenEXR include: /usr/include;/usr/include/openexr
-- PNG: /usr/lib/aarch64-linux-gnu/libpng.so
-- R8B: r8b
-- OpenCV libs: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab
-- CERES LIBS: general;/usr/lib/libceres.so
-- 
-- Found CUDA: /usr/local/cuda-10.2 (found version "10.2") 
-- VideoStitch I/O libs configured to build: avPlugin;jpgPlugin;pamPlugin;pngPlugin;rawPlugin;exrPlugin;v4l2Plugin;portaudioPlugin
-- Qt minimum supported version: 5.6
-- Qt FOUND: 5.9.5
-- Studio version: Studio-v2.4.0-win-p1-28-g67b7fbc-master.2020-08-05
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nvidia/stitchEm

**I face a problem when I type _Ninja_:**

nvidia@jetson-0423718018083:~/stitchEm$ ninja
[1/634] cd /home/nvidia/stitchEm/lib && ls /home/nvidia/stitchEm/generated_headers/version.hpp
/home/nvidia/stitchEm/generated_headers/version.hpp
[3/633] Building CXX object IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o
FAILED: IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o 
/usr/bin/c++   -Ilib/include -Igenerated_headers -IIO/src/common/format/include -IIO/src/common/format/../include -Wall -Wextra -Wno-missing-field-initializers -g -fPIC   -Wno-deprecated-declarations -std=gnu++14 -MD -MT IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o -MF IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o.d -o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o -c IO/src/common/format/src/filemuxer.cpp
IO/src/common/format/src/filemuxer.cpp: In member function ‘bool VideoStitch::Output::FileMuxer::MP4WebOptimizerInternal(const string&)’:
IO/src/common/format/src/filemuxer.cpp:85:32: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
     if (formatCtx->streams[i]->codecpar->codec_type == AVMEDIA_TYPE_AUDIO) {
                                ^~~~~~~~
IO/src/common/format/src/filemuxer.cpp:86:43: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
       nbChannels = formatCtx->streams[i]->codecpar->channels;
                                           ^~~~~~~~
[4/633] Building CXX object IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o
FAILED: IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o 
/usr/bin/c++   -Ilib/include -Igenerated_headers -IIO/src/common/format/include -IIO/src/common/format/../include -Wall -Wextra -Wno-missing-field-initializers -g -fPIC   -Wno-deprecated-declarations -std=gnu++14 -MD -MT IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o -MF IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o.d -o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o -c IO/src/common/format/src/muxer.cpp
IO/src/common/format/src/muxer.cpp: In constructor ‘VideoStitch::Output::Muxer::Muxer(size_t, const string&, std::vector<VideoStitch::AVEncoder>&, const AVDictionary*)’:
IO/src/common/format/src/muxer.cpp:98:51: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
       if (avcodec_parameters_from_context(stream->codecpar, encoder.codecContext) < 0) {
                                                   ^~~~~~~~
IO/src/common/format/src/muxer.cpp:98:81: error: ‘avcodec_parameters_from_context’ was not declared in this scope
       if (avcodec_parameters_from_context(stream->codecpar, encoder.codecContext) < 0) {
                                                                                 ^
IO/src/common/format/src/muxer.cpp: In member function ‘void VideoStitch::Output::Muxer::writeHeader()’:
IO/src/common/format/src/muxer.cpp:170:34: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
       if (formatCtx->streams[i]->codecpar) {
                                  ^~~~~~~~
IO/src/common/format/src/muxer.cpp:171:72: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
         auto codecDesc = avcodec_descriptor_get(formatCtx->streams[i]->codecpar->codec_id);
[... 99 lines cut ...]
apps/src/libvideostitch-base/videowidget.cpp:272:30: error: ‘glVertex2f’ was not declared in this scope
   glVertex2f(hMargin, vMargin);
                              ^
apps/src/libvideostitch-base/videowidget.cpp:325:9: error: ‘glEnd’ was not declared in this scope
   glEnd();
         ^
apps/src/libvideostitch-base/videowidget.cpp: In member function ‘virtual void VideoWidget::resizeGL(int, int)’:
apps/src/libvideostitch-base/videowidget.cpp:330:16: error: ‘GL_PROJECTION’ was not declared in this scope
   glMatrixMode(GL_PROJECTION);
                ^~~~~~~~~~~~~
apps/src/libvideostitch-base/videowidget.cpp:330:29: error: ‘glMatrixMode’ was not declared in this scope
   glMatrixMode(GL_PROJECTION);
                             ^
apps/src/libvideostitch-base/videowidget.cpp:331:18: error: ‘glLoadIdentity’ was not declared in this scope
   glLoadIdentity();
                  ^
apps/src/libvideostitch-base/videowidget.cpp:332:27: error: ‘glOrtho’ was not declared in this scope
   glOrtho(0, w, h, 0, 0, 1);
                           ^
apps/src/libvideostitch-base/videowidget.cpp:333:16: error: ‘GL_MODELVIEW’ was not declared in this scope
   glMatrixMode(GL_MODELVIEW);
                ^~~~~~~~~~~~

I wait for several hours and the program was still running under this situation.
Could you please tell me what cause this? And what extra information you need to check? Please tell me how to solve them. Thank you very much!

wentao0614-wang, 2020-12-15:
nvidia@jetson-0423718018083:~/stitchEm$ ninja
[1/634] cd /home/nvidia/stitchEm/lib && ls /home/nvidia/stitchEm/generated_headers/version.hpp
/home/nvidia/stitchEm/generated_headers/version.hpp
[3/633] Building CXX object IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o
FAILED: IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o 
/usr/bin/c++   -Ilib/include -Igenerated_headers -IIO/src/common/format/include -IIO/src/common/format/../include -Wall -Wextra -Wno-missing-field-initializers -g -fPIC   -Wno-deprecated-declarations -std=gnu++14 -MD -MT IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o -MF IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o.d -o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o -c IO/src/common/format/src/filemuxer.cpp
IO/src/common/format/src/filemuxer.cpp: In member function ‘bool VideoStitch::Output::FileMuxer::MP4WebOptimizerInternal(const string&)’:
IO/src/common/format/src/filemuxer.cpp:85:32: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
     if (formatCtx->streams[i]->codecpar->codec_type == AVMEDIA_TYPE_AUDIO) {
                                ^~~~~~~~
IO/src/common/format/src/filemuxer.cpp:86:43: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
       nbChannels = formatCtx->streams[i]->codecpar->channels;
                                           ^~~~~~~~
[4/633] Building CXX object IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o
FAILED: IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o 
/usr/bin/c++   -Ilib/include -Igenerated_headers -IIO/src/common/format/include -IIO/src/common/format/../include -Wall -Wextra -Wno-missing-field-initializers -g -fPIC   -Wno-deprecated-declarations -std=gnu++14 -MD -MT IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o -MF IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o.d -o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o -c IO/src/common/format/src/muxer.cpp
IO/src/common/format/src/muxer.cpp: In constructor ‘VideoStitch::Output::Muxer::Muxer(size_t, const string&, std::vector<VideoStitch::AVEncoder>&, const AVDictionary*)’:
IO/src/common/format/src/muxer.cpp:98:51: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
       if (avcodec_parameters_from_context(stream->codecpar, encoder.codecContext) < 0) {
                                                   ^~~~~~~~
IO/src/common/format/src/muxer.cpp:98:81: error: ‘avcodec_parameters_from_context’ was not declared in this scope
       if (avcodec_parameters_from_context(stream->codecpar, encoder.codecContext) < 0) {
                                                                                 ^
IO/src/common/format/src/muxer.cpp: In member function ‘void VideoStitch::Output::Muxer::writeHeader()’:
IO/src/common/format/src/muxer.cpp:170:34: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
       if (formatCtx->streams[i]->codecpar) {
                                  ^~~~~~~~
IO/src/common/format/src/muxer.cpp:171:72: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
         auto codecDesc = avcodec_descriptor_get(formatCtx->streams[i]->codecpar->codec_id);
                                                                        ^~~~~~~~
[5/633] Building CXX object apps/src/libvideostitch-base/CMakeFiles/videostitch-base.dir/videowidget.cpp.o
FAILED: apps/src/libvideostitch-base/CMakeFiles/videostitch-base.dir/videowidget.cpp.o 
/usr/bin/c++  -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_MULTIMEDIA_LIB -DQT_NETWORK_LIB -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DVS_LIB_COMMON_COMPILATION -Dvideostitch_base_EXPORTS -Iapps/src/libvideostitch-base -Iapps/src/libvideostitch-base/videostitch-base_autogen/include -Iapps/src -Idiscovery/include -Ilib/include -Igenerated_headers -isystem /usr/include/aarch64-linux-gnu/qt5 -isystem /usr/include/aarch64-linux-gnu/qt5/QtCore -isystem /usr/lib/aarch64-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/aarch64-linux-gnu/qt5/QtGui -isystem /usr/include/aarch64-linux-gnu/qt5/QtOpenGL -isystem /usr/include/aarch64-linux-gnu/qt5/QtWidgets -isystem /usr/include/aarch64-linux-gnu/qt5/QtConcurrent -isystem /usr/include/aarch64-linux-gnu/qt5/QtMultimedia -isystem /usr/include/aarch64-linux-gnu/qt5/QtNetwork -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-variable -Wno-unused-variable -g -fPIC   -fPIC -std=gnu++14 -MD -MT apps/src/libvideostitch-base/CMakeFiles/videostitch-base.dir/videowidget.cpp.o -MF apps/src/libvideostitch-base/CMakeFiles/videostitch-base.dir/videowidget.cpp.o.d -o apps/src/libvideostitch-base/CMakeFiles/videostitch-base.dir/videowidget.cpp.o -c apps/src/libvideostitch-base/videowidget.cpp
apps/src/libvideostitch-base/videowidget.cpp: In member function ‘virtual void VideoWidget::initializeGL()’:
apps/src/libvideostitch-base/videowidget.cpp:32:13: error: ‘GL_LIGHTING’ was not declared in this scope
   glDisable(GL_LIGHTING);
             ^~~~~~~~~~~
apps/src/libvideostitch-base/videowidget.cpp:33:16: error: ‘GL_FLAT’ was not declared in this scope
   glShadeModel(GL_FLAT);
                ^~~~~~~
apps/src/libvideostitch-base/videowidget.cpp:33:23: error: ‘glShadeModel’ was not declared in this scope
[... cut ...]
apps/src/libvideostitch-base/videowidget.cpp:332:27: error: ‘glOrtho’ was not declared in this scope
   glOrtho(0, w, h, 0, 0, 1);
                           ^
apps/src/libvideostitch-base/videowidget.cpp:333:16: error: ‘GL_MODELVIEW’ was not declared in this scope
   glMatrixMode(GL_MODELVIEW);
                ^~~~~~~~~~~~
[6/633] Building NVCC (Device) object lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/core1/cuda_compile_1_generated_cutransform.cu.o
**ninja: build stopped: subcommand failed.**

w-m, 2020-12-15:
Use `-DCUDA_LOCAL_ARCH_ONLY=ON` in the CMake invocation to drastically cut down the compilation time, building the binaries only for the CUDA architecture of the current device.

> IO/src/common/format/src/filemuxer.cpp:85:32: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?

This is caused by using an old ffmpeg version. What ffmpeg version do you have installed in /usr/local?

> apps/src/libvideostitch-base/videowidget.cpp: In member function ‘virtual void VideoWidget::initializeGL()’:
> apps/src/libvideostitch-base/videowidget.cpp:32:13: error: ‘GL_LIGHTING’ was not declared in this scope
> glDisable(GL_LIGHTING);
> ^~~~~~~~~~~

These OpenGL functions should be provided by the Qt OpenGL headers in this file. Maybe Qt+OpenGL needs a custom setup on Xavier?

wentao0614-wang, 2020-12-17:
Thank you for your reply. First of all, I changed the version of ffmpeg to 4.0.2, which seems to have solved this problem:

```
apps/src/libvideostitch-base/videowidget.cpp: In member function ‘virtual void VideoWidget::initializeGL()’:
apps/src/libvideostitch-base/videowidget.cpp:32:13: error: ‘GL_LIGHTING’ was not declared in this scope
glDisable(GL_LIGHTING);
^~~~~~~~~~~
```
_**However, when I type Ninja,I encountered a new problem：**_
[1/694] cd /home/nvidia/stitchEm/stitc...stitchEm/generated_headers/version.hpp
/home/nvidia/stitchEm/stitchEm/generated_headers/version.hpp
[2/694] Building NVCC (Device) object ...ors/cuda_compile_1_generated_tint.cu.o
FAILED: lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/processors/cuda_compile_1_generated_tint.cu.o 
cd /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/processors && /usr/local/bin/cmake -E make_directory /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/processors/. && /usr/local/bin/cmake -D verbose:BOOL=OFF -D build_configuration:STRING=Debug -D generated_file:STRING=/home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/processors/./cuda_compile_1_generated_tint.cu.o -D generated_cubin_file:STRING=/home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/processors/./cuda_compile_1_generated_tint.cu.o.cubin.txt -P /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/processors/cuda_compile_1_generated_tint.cu.o.Debug.cmake
nvcc fatal   : Unsupported gpu architecture 'compute_1'
CMake Error at cuda_compile_1_generated_tint.cu.o.Debug.cmake:216 (message):
  Error generating
  /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/processors/./cuda_compile_1_generated_tint.cu.o

[3/694] Building NVCC (Device) object ..._compile_1_generated_boundsKernel.cu.o
FAILED: lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/cuda_compile_1_generated_boundsKernel.cu.o 
cd /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda && /usr/local/bin/cmake -E make_directory /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/. && /usr/local/bin/cmake -D verbose:BOOL=OFF -D build_configuration:STRING=Debug -D generated_file:STRING=/home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/./cuda_compile_1_generated_boundsKernel.cu.o -D generated_cubin_file:STRING=/home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/./cuda_compile_1_generated_boundsKernel.cu.o.cubin.txt -P /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/cuda_compile_1_generated_boundsKernel.cu.o.Debug.cmake
nvcc fatal   : Unsupported gpu architecture 'compute_1'
CMake Error at cuda_compile_1_generated_boundsKernel.cu.o.Debug.cmake:216 (message):
  Error generating
  /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/./cuda_compile_1_generated_boundsKernel.cu.o

[4/694] Building NVCC (Device) object ..._compile_1_generated_mergerKernel.cu.o
FAILED: lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/core1/cuda_compile_1_generated_mergerKernel.cu.o 
cd /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/core1 && /usr/local/bin/cmake -E make_directory /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/core1/. && /usr/local/bin/cmake -D verbose:BOOL=OFF -D build_configuration:STRING=Debug -D generated_file:STRING=/home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/core1/./cuda_compile_1_generated_mergerKernel.cu.o -D generated_cubin_file:STRING=/home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/core1/./cuda_compile_1_generated_mergerKernel.cu.o.cubin.txt -P /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/core1/cuda_compile_1_generated_mergerKernel.cu.o.Debug.cmake
nvcc fatal   : Unsupported gpu architecture 'compute_1'
CMake Error at cuda_compile_1_generated_mergerKernel.cu.o.Debug.cmake:216 (message):
  Error generating
  /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/core1/./cuda_compile_1_generated_mergerKernel.cu.o

[5/694] Building NVCC (Device) object ..._compile_1_generated_fillRenderer.cu.o
FAILED: lib/CMakeFiles/cuda_compile_1.dir/src/render/cuda_compile_1_generated_fillRenderer.cu.o 
cd /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/render && /usr/local/bin/cmake -E make_directory /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/render/. && /usr/local/bin/cmake -D verbose:BOOL=OFF -D build_configuration:STRING=Debug -D generated_file:STRING=/home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/render/./cuda_compile_1_generated_fillRenderer.cu.o -D generated_cubin_file:STRING=/home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/render/./cuda_compile_1_generated_fillRenderer.cu.o.cubin.txt -P /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/render/cuda_compile_1_generated_fillRenderer.cu.o.Debug.cmake
nvcc fatal   : Unsupported gpu architecture 'compute_1'
CMake Error at cuda_compile_1_generated_fillRenderer.cu.o.Debug.cmake:216 (message):
  Error generating
[... cut ...] end/cuda/cuda_compile_1_generated_exampleKernel.cu.o.Debug.cmake
nvcc fatal   : Unsupported gpu architecture 'compute_1'
CMake Error at cuda_compile_1_generated_exampleKernel.cu.o.Debug.cmake:216 (message):
  Error generating
  /home/nvidia/stitchEm/stitchEm/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/./cuda_compile_1_generated_exampleKernel.cu.o

ninja: build stopped: subcommand failed.

Can you take a look at it for me? Then tell me what to do to avoid this mistake. Thank you very much.

w-m, 2020-12-17:
The new error comes from the `-DCUDA_LOCAL_ARCH_ONLY=ON` I told you to use. It seems the script that is supposed to query for the local architecture fails in some way and returns "1" instead of "72". The OpenGL problem is likely still there, just isn't reached as the compilation drops out much sooner.

In case you didn't do this before, please clear out your CMake build folder completely, and then try a clean build again with `-DCUDA_LOCAL_ARCH_ONLY=ON` in the first CMake invocation.

If that produces the same errors (`nvcc fatal : Unsupported gpu architecture 'compute_1'`), clean out the build folder again and try with `-DCUDA_TARGET_ARCH=72` instead.

natecarlson, 2021-02-24:
@w-m  Sounds like this can be closed; the work continues in #132.

closed 2021-02-24 by w-m