stitchEm/stitchEm #131: RTMP streaming not working

tripolskypetr, 2021-01-28:
The program just crash, is anyone tested it?

w-m, 2021-01-28:
Reproduce: create a new project with test inputs. Add an RTMP output, leave every value as the default value. Notice that "Encoder" and "Preser" in Encoder settings are completely empty (and that level is set to "4.2", which is probably also a problem, see #80), and the dropdown boxes don't contain any values. Click save. Application crashes.

CheckBackfocus, 2021-03-31:
Can confirm trying to save RTMP stream causes VR App to die. Are there any clever workaround that can be done in the config file?

w-m, 2021-03-31:
I can't test it myself at the moment. If you're feeling adventurous, you could experiment with adding something like [this RTMP output block](https://github.com/stitchEm/stitchEm/blob/master/tests/cli_tests/etc/assets_ptv/videoformat01/rtmp.ptv#L11-L25) manually to the project file.

CheckBackfocus, 2021-04-01:
Ok, so this does add the RTMP button to the GUI, however, it reports "Could not create an output for configuration 'rtmp'" when attempting to start the stream.

I did try manually calling codecs, such as;

```
	"type": "rtmp",
	"filename": "rtmp://127.0.0.1:1935/live/output",
    "downsampling_factor": 1,
    "pub_user" : "",
    "pub_enc_passwd" : "",

	// video encoder settings depend on the selected encoder
	"video_codec" : "x264",
	"tune" : "",
	"profile" : "baseline",
	"level" : "4.2",
	"filename" : "rtmp://localhost:1935/live/stream",
	"bitrate_mode" : "VBR",
	"quality_balance" : 5,
	"b_frames" : 2,
	"bitrate" : 8001,
	"cbr_padding" : false,
	"target_usage" : 4,
	"gop" : -1

```
..but no luck. Any initial ideas of where it might be worth checking the source? Do you know if RTMP works on, say, a Linux build?

(NB: This is using the pre-built Win 10 x64 release - so could try and manual build too)

w-m, 2021-04-01:
Have a look at the log window (second button from the bottom right) when trying to create the output, it may give more information on why the output can't be created.

For the crash in Vahana VR when creating the output I think it's best to do a manual debug build and just reproduce it inside a 
debugger. Visual Studio and Qt Creator IDEs are both supported on Windows. The crash can probably be reproduced on Linux (maybe even macOS) as well, where it would be much easier to set up such a build.

There is more documentation on the rtmp configurations in the readme in this folder: https://github.com/stitchEm/stitchEm/tree/master/IO/src/rtmp

So you could try at least three different encoders (libx264, QSV, NVENC). I'm not sure how up to date the readme is. If in doubt, check the code in the subfolders.

CheckBackfocus, 2021-04-01:
Ok, that's really interesting - thanks.

In terms of the program log, it reports it is unable to find rtmp.dll at the given directory - but the dll in where it should be. I'll play with the debugger and go from there.

For the sake of logging, here's the log output: 

```
StateManager::changeState, state = 0, sender = ProjectWorkWidget
[2021/04/01 17:45:34] Debug: could not load plug-in "C:\Program Files\VideoStitch\VahanaVR\vahana_plugins\rtmp.dll": The specified module could not be found.
[2021/04/01 17:45:34] [videoinput] Couldn't open file for probing. Aborting.
[2021/04/01 17:45:34] [videoinput] Couldn't open file for probing. Aborting.
[2021/04/01 17:45:34] [videoinput] Couldn't open file for probing. Aborting.
[2021/04/01 17:45:34] [videoinput] Couldn't open file for probing. Aborting.
[2021/04/01 17:45:34] [videoinput] Couldn't open file for probing. Aborting.
[2021/04/01 17:45:34] [videoinput] Couldn't open file for probing. Aborting.
[2021/04/01 17:45:34] Missing mandatory field 'preset' for object Ptv.
StateManager::changeState, state = 2, sender = ProjectWorkWidget
```

w-m, 2021-04-01:
If the file is there, but it complains that the plugin can't be loaded, it's probably due to the plugin dll missing dependencies ("The specified module could not be found" is just the last error message printed).

I've just booted up Windows now. I opened rtmp.dll in Dependency Walker and added the VahanaVR directory to the Module Search Order, gives this result:

[image]

I'm not sure if the Intel icls stuff missing on my machine would is a dealbreaker (are they delay-loaded only by librtmp, or what is that pink symbol saying?), but I believe libfaac.dll and libfaad2.dll missing is a problem. They don't seem to be there for me in a clean install.

Maybe there's a problem that they were built but not included in the installer?

w-m, 2021-04-01:
> Maybe there's a problem that they were built but not included in the installer?

The change where we went from using custom self-built dependencies to mostly vcpkg-stuff, there is a change in libfaac and libfaad, see https://github.com/stitchEm/stitchEm/pull/43/files (ctrl+f libfaac). Curiously, libfaac and libfaad are never mentioned in the installer listings, and never have been? I'm not sure why that is.

The Intel dlls seen above on the other hand are mentioned in the installer scripts, and are in fact installed to the Vahana VR directory. Is that good enough to enable rtmp.dll to find them (even if they are supposed to be at a different absolute path, taking from the screenshot above)? Sorry, I'm not very knowledgeable in Window dependency resolution and am stabbing in the dark a lot.

w-m, 2021-04-01:
Intel icls rose colored errors seem to be from different CPU archs or something. Moving the Vahana VR directory to the top of the list, if finds the ones in the Vahana VR directory, and is happy with then. libfaac and libfaad2 remain missing, and some API-MS-WIN-CRT stuff (but IIRC that's usually just from Dependency Walker having a missing directory).

Still unclear why faac and faad2 aren't in the installer script, and seemingly never have been (https://github.com/stitchEm/stitchEm/blob/master/apps/installer/common-dll.iss#L57-L63). @jeremad, ideas?

CheckBackfocus, 2021-04-07:
So, attempting to build from VS2019 is giving loads of  (1000+) errors after a successful CMake. Will attempt Ubuntu build and test RTMP streaming from that platform.

w-m, 2021-04-07:
VS2017 wasn't an option? What kinds of errors do you see?

w-m, 2021-04-08:
> Still unclear why faac and faad2 aren't in the installer script, and seemingly never have been (https://github.com/stitchEm/stitchEm/blob/master/apps/installer/common-dll.iss#L57-L63). @jeremad, ideas?

Thinking about it: maybe they were included as static libraries before, and now are dynamic libraries.

So two possible fixes. Either change the build https://github.com/stitchEm/stitchEm/blob/master/IO/src/rtmp/CMakeLists.txt#L236-L270 to put out static libraries, and keep the installer script as-is.

Or, add the two new dynamic dlls to the installer script: https://github.com/stitchEm/stitchEm/blob/master/apps/installer/common-dll.iss#L57-L63

CheckBackfocus, 2021-04-08:
> VS2017 wasn't an option? What kinds of errors do you see?

Will attempt a build with VS2017 shortly. VS19 highlights syntax errors everywhere, so could just be a weird complier thing.

CheckBackfocus, 2021-04-08:
As update, the compile is smoother under VS2017, but still fails. 

Most of the errors are syntax related from not being able to use "OVR_CAPI_GL.h". The file is present in the Oculus SDK - so I'm trying to find a way to help the complier find it.

CheckBackfocus, 2021-04-13:
Program runs after manual build and placement of Libfaac & faad! 

Doesn't seem very happy streaming to YouTube, but will to twitch, which may well be a case of this machine being made out of old calculators and cheese scrapings. 

Can only encoder via H264, can't see (or manually use via changing config files) NVIDA or Intel options. Investigating.