stitchEm/stitchEm #132: Still errors on Xavier

2018boyan, 2021-02-22:
I am the person who take over the work that install this wonderful software on xavier. I still meet quite same issues as predecessor.
This is my cmake:

nvidia@jetson-0423718018083:~$ cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -G Ninja stitchEm
-- The C compiler identification is GNU 6.5.0
-- The CXX compiler identification is GNU 6.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- GPU_BACKEND_CUDA='ON'
-- GPU_BACKEND_OPENCL='OFF'
-- USE_DELAY_LOAD=OFF
-- System: LINUX
-- CMake Build Type -- Debug
-- Performing Test HAVE_FLAG_ADDRESS_SANITIZER
-- Performing Test HAVE_FLAG_ADDRESS_SANITIZER - Failed
-- Performing Test HAVE_FLAG_SANITIZE_ADDRESS
-- Performing Test HAVE_FLAG_SANITIZE_ADDRESS - Success
-- Performing Test HAVE_FLAG_LEAK_SANITIZER
-- Performing Test HAVE_FLAG_LEAK_SANITIZER - Failed
-- Performing Test HAVE_FLAG_SANITIZE_THREAD
-- Performing Test HAVE_FLAG_SANITIZE_THREAD - Failed
-- Performing Test HAVE_FLAG_SANITIZE_UNDEFINED_BEHAVIOR
-- Performing Test HAVE_FLAG_SANITIZE_UNDEFINED_BEHAVIOR - Success
-- CUDA VERSION: 10.2
-- CUDA PATH: /usr/local/cuda-10.2
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-10.2 (found suitable exact version "10.2") 
**-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so**  
-- Found GLEW: /usr/include (found version "2.1.0") 
-- CUDA: /usr/lib/aarch64-linux-gnu/libcuda.so
-- CUDART: /usr/local/cuda-10.2/lib64/libcudart.so
-- CUDA_LIBRARIES: /usr/local/cuda-10.2/lib64/libcudart_static.a;Threads::Threads;dl;/usr/lib/aarch64-linux-gnu/librt.so
-- NVCC arch flags: -gencode=arch=compute_30,code=sm_30;-gencode=arch=compute_35,code=sm_35;-gencode=arch=compute_50,code=sm_50;-gencode=arch=compute_52,code=sm_52;-gencode=arch=compute_61,code=sm_61;-gencode=arch=compute_61,code=compute_61
-- Looking for CL_VERSION_2_2
-- Looking for CL_VERSION_2_2 - found
-- Found OpenCL: /usr/lib/aarch64-linux-gnu/libOpenCL.so (found version "2.2") 
-- Found BISON: /usr/bin/bison (found suitable version "3.0.4", minimum required is "3.0") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- System: LINUX
-- Found OpenCV: /usr (found version "3.3.1") 
-- --- VideoStitch-lib dependencies ---
-- Found OpenEXR: /usr/include;/usr/include/openexr  
-- OpenEXR libs: /usr/lib/aarch64-linux-gnu/libHalf.so;/usr/lib/aarch64-linux-gnu/libIex.so;/usr/lib/aarch64-linux-gnu/libImath.so;/usr/lib/aarch64-linux-gnu/libIlmImf.so;/usr/lib/aarch64-linux-gnu/libIlmThread.so
-- OpenEXR include: /usr/include;/usr/include/openexr
-- PNG: /usr/lib/aarch64-linux-gnu/libpng.so
-- R8B: r8b
-- OpenCV libs: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab
-- CERES LIBS: general;/usr/lib/libceres.so
-- 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components: doxygen missing components: dot
-- Found CUDA: /usr/local/cuda-10.2 (found version "10.2") 
-- VideoStitch I/O libs configured to build: avPlugin;jpgPlugin;pamPlugin;pngPlugin;rawPlugin;tiffPlugin;exrPlugin;v4l2Plugin;portaudioPlugin;rtmpPlugin
-- Qt minimum supported version: 5.6
-- Qt FOUND: 5.9.5
-- Studio version: Studio-v2.4.0-win-p1-53-g7bb49a3-master.2021-01-04
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nvidia

2018boyan, 2021-02-22:
The "#" is added to show openGL and a line with only "--" before Doxygen. It seems cmake has found openGL but still print errors when I type "ninja".

nvidia@jetson-0423718018083:~$ ninja
[0/1] Re-running CMake...
-- GPU_BACKEND_CUDA='ON'
-- GPU_BACKEND_OPENCL='OFF'
-- USE_DELAY_LOAD=OFF
-- System: LINUX
-- CMake Build Type -- Debug
-- CUDA VERSION: 10.2
-- CUDA PATH: /usr/local/cuda-10.2
-- Found CUDA: /usr/local/cuda-10.2 (found suitable exact version "10.2") 
################################################
no opengl and GLEW
################################################
-- CUDA: /usr/lib/aarch64-linux-gnu/libcuda.so
-- CUDART: /usr/local/cuda-10.2/lib64/libcudart.so
-- CUDA_LIBRARIES: /usr/local/cuda-10.2/lib64/libcudart_static.a;Threads::Threads;dl;/usr/lib/aarch64-linux-gnu/librt.so
-- NVCC arch flags: -gencode=arch=compute_30,code=sm_30;-gencode=arch=compute_35,code=sm_35;-gencode=arch=compute_50,code=sm_50;-gencode=arch=compute_52,code=sm_52;-gencode=arch=compute_61,code=sm_61;-gencode=arch=compute_61,code=compute_61
-- System: LINUX
-- --- VideoStitch-lib dependencies ---
-- OpenEXR libs: /usr/lib/aarch64-linux-gnu/libHalf.so;/usr/lib/aarch64-linux-gnu/libIex.so;/usr/lib/aarch64-linux-gnu/libImath.so;/usr/lib/aarch64-linux-gnu/libIlmImf.so;/usr/lib/aarch64-linux-gnu/libIlmThread.so
-- OpenEXR include: /usr/include;/usr/include/openexr
-- PNG: /usr/lib/aarch64-linux-gnu/libpng.so
-- R8B: r8b
-- OpenCV libs: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab
-- CERES LIBS: general;/usr/lib/libceres.so
-- 
-- Found CUDA: /usr/local/cuda-10.2 (found version "10.2") 
-- VideoStitch I/O libs configured to build: avPlugin;jpgPlugin;pamPlugin;pngPlugin;rawPlugin;tiffPlugin;exrPlugin;v4l2Plugin;portaudioPlugin;rtmpPlugin
-- Qt minimum supported version: 5.6
-- Qt FOUND: 5.9.5
-- Studio version: Studio-v2.4.0-win-p1-53-g7bb49a3-master.2021-01-04
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nvidia

It seems that ninja re-running CMake doesnot not find openGL.

2018boyan, 2021-02-22:
2. Another issue is still about ffmpeg.

[533/983] Linking CXX shared library b.../x64/debug/core_plugins/libavPlugin.so
FAILED: bin/x64/debug/core_plugins/libavPlugin.so 
: && /usr/bin/c++ -fPIC -Wall -Wextra -Wno-missing-field-initializers -g   -shared -Wl,-soname,libavPlugin.so -o bin/x64/debug/core_plugins/libavPlugin.so IO/src/common/CMakeFiles/common.dir/src/filesystem.cpp.o IO/src/common/CMakeFiles/common.dir/src/frameRateHelpers.cpp.o IO/src/common/CMakeFiles/common.dir/src/io.cpp.o IO/src/common/CMakeFiles/common.dir/src/multiFileInput.cpp.o IO/src/common/CMakeFiles/common.dir/src/numberedFilesOutput.cpp.o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/avMuxer.cpp.o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/filemuxer.cpp.o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/livemuxer.cpp.o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/qt-faststart.cpp.o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/util.cpp.o IO/src/av/CMakeFiles/avPlugin.dir/src/avWriter.cpp.o IO/src/av/CMakeFiles/avPlugin.dir/src/baseAllocator.cpp.o IO/src/av/CMakeFiles/avPlugin.dir/src/d3dAllocator.cpp.o IO/src/av/CMakeFiles/avPlugin.dir/src/d3dDevice.cpp.o IO/src/av/CMakeFiles/avPlugin.dir/src/export.cpp.o IO/src/av/CMakeFiles/avPlugin.dir/src/libavReader.cpp.o IO/src/av/CMakeFiles/avPlugin.dir/src/netStreamReader.cpp.o IO/src/av/CMakeFiles/avPlugin.dir/src/timeoutUtil.cpp.o IO/src/av/CMakeFiles/avPlugin.dir/src/videoReader.cpp.o  -Wl,-rpath,/usr/local/cuda-10.2/lib64:/home/nvidia/bin/x64/debug:/usr/local/cuda-10.2/lib64/stubs  /usr/local/cuda-10.2/lib64/libcudart.so  /usr/local/lib/libavcodec.a  /usr/local/lib/libavutil.a  /usr/local/lib/libavformat.a  bin/x64/debug/libvideostitch_cuda.so  /usr/local/cuda-10.2/lib64/stubs/libnvidia-ml.so  /usr/local/cuda-10.2/lib64/libcudart_static.a  -lpthread  -ldl  /usr/lib/aarch64-linux-gnu/librt.so  /usr/lib/aarch64-linux-gnu/libcuda.so  /usr/lib/aarch64-linux-gnu/libHalf.so  /usr/lib/aarch64-linux-gnu/libIex.so  /usr/lib/aarch64-linux-gnu/libImath.so  /usr/lib/aarch64-linux-gnu/libIlmImf.so  /usr/lib/aarch64-linux-gnu/libIlmThread.so  /usr/lib/aarch64-linux-gnu/libdrm.so  -Wl,-rpath-link,/home/nvidia/bin/x64/debug && :
/usr/bin/aarch64-linux-gnu-ld: /usr/local/lib/libavcodec.a(fft_neon.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `ff_cos_32' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libavcodec.a(fft_neon.o): In function `fft32_neon':
/home/nvidia/ffmpeg-4.0.2/libavcodec/aarch64/fft_neon.S:356:(.text+0x3e8): dangerous relocation: unsupported relocation
/usr/bin/aarch64-linux-gnu-ld: /usr/local/lib/libavcodec.a(fft_neon.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `ff_cos_64' which may bind externally can not be used when making a shared object; recompile with -fPIC

I install ffmpeg-4.0.2 and it seems to install successfully.
[image]
How to fix this issue. Could you please help me? Thanks a lot.

w-m, 2021-02-23:
As OpenGL is found, but there are specific OpenGL errors during compilation, this may rather be due to an OpenGL profile mismatch.

If you just need the stitching part (the C++/CUDA library), consider not building the app part (VideoStitch Studio and Vahana VR) at all, by using `-DBUILD_APPS=OFF` in the CMake invocation. This should get rid the OpenGL errors posted above.

If you want to run the apps on the device, you will have to do some coding to replace the OpenGL calls throwing the compilation errors with ones available on your platform. From the error messages posted I would guess that stitchEm Qt apps are using Desktop OpenGL commands that aren't available on the OpenGL ES profile on the Jetson or Xavier.

I'm not very proficient in OpenGL, so I can't give you much advice for that. If it were me, I'd comment out all the lines that
throw these OpenGL compilation errors and try to reach a working build (with then probably completely black outputs in the window), and then one-for-one work on reinstating the OpenGL lines with the appropriate code of the available profile.

w-m, 2021-02-23:
As for the issue with ffmpeg, follow the hint in the compilation output: it requires `-fPIC` to be added the compilation invocation.

To compile libavPlugin with `-fPIC` in CMake, add

`set_property(TARGET ${PLUGIN_NAME} PROPERTY POSITION_INDEPENDENT_CODE TRUE)`

below this line: 
https://github.com/stitchEm/stitchEm/blob/master/IO/src/av/CMakeLists.txt#L70

natecarlson, 2021-02-23:
It sounds like the Jetson should support full OpenGL:
https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/software_features_jetson_nano.html#wwpID0E0CB0HA

It appears it should also be possible to cross-compile on an x64 box:
https://forums.developer.nvidia.com/t/cross-compiling-c-c-code-for-jetson-nano-in-ubuntu-18-04-in-a-host-pc-x86-x64/73473

..however, that would really complicate the CUDA version auto-detection/etc.  :)

w-m, 2021-02-23:
Oh, so is it the other way round: the OpenGL functions throwing errors on the Jetson compilation are from an ancient profile, and have been deprecated? Work to provide a solution by replacing one-by-one with available counterparts should still apply though :)

natecarlson, 2021-02-23:
Just validating basic Linux build (on x64) with CUDA, to make sure the directions are (relatively) up to date.. I'm using the NVidia CUDA 10.2 docker image with Ubuntu 18.04, which should be relatively similar to the Jetson development environment (except x64 instead of ARM, obviously.)

I'd be interested to see if these same steps (pulling in all the GL dependencies, etc) work properly on a Jetson. I wouldn't be surprised if the fPIC error still occurs - but interesting to see.

* Prep a host with an Nvidia card of some sort (P620 in my case) with docker-ce + nvidia-docker2
* Start a Docker shell with Ubuntu 18.04 and CUDA 10.2 development environment: `docker run --gpus all -it nvidia/cuda:10.2-devel-ubuntu18.04 bash`
* Add repo for updated version of cmake: `apt-key adv --keyserver keyserver.ubuntu.com --recv-keys DE19EB17684BA42D ; apt update ; apt install software-properties-common ; apt-add-repository 'deb https://apt.kitware.com/ubuntu/ bionic main'`
* Install dependencies: `apt update ; apt -y install git bison clang-tools cmake doxygen flex libceres-dev libeigen3-dev libfaac-dev libfaad-dev libmp3lame-dev libopencv-dev libopenexr-dev libpng-dev libportaudio-ocaml-dev librtmp-dev libturbojpeg0-dev libx264-dev ninja-build ocl-icd-opencl-dev opencl-headers portaudio19-dev qt5-default qtmultimedia5-dev qttools5-dev swig wget xxd libglew-dev libglfw3-dev libglm-dev libdrm-dev`
* Clone the repo: `cd /var/tmp ; git clone https://github.com/stitchEm/stitchEm.git`
* Prep build: `mkdir /var/tmp/stitchEm-build ; cd /var/tmp/stitchEm-build ; cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -G Ninja ../stitchEm`
* Compile: `cd /var/tmp/stitchEm-build/ ; ninja`
* This results in a good compile: `[983/983] Linking CXX executable bin/x64/debug/VahanaVR`

natecarlson, 2021-02-23:
Looks like the following details setting up cross-compiling:
https://docs.nvidia.com/jetson/l4t-multimedia/cross_platform_support.html

Note: if you've got an aarch64 system available, looks like it would be much simpler to use this docker container:
https://ngc.nvidia.com/catalog/containers/nvidia:l4t-base
Could also theoretically use this via qemu aarch64 on an x64 host - but I'm betting that building the source within qemu will likely be much slower than cross compilation.

* Set up the docker container as above - but add --privileged to the docker run command.. this is required to support aarch64 emulation to install the required libraries within the Jetson root image.
* Download the aarch64 toolchain: `cd /var/tmp ; wget --content-disposition "https://developer.nvidia.com/embedded/dlc/l4t-gcc-7-3-1-toolchain-64-bit" ; mv gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu.tar.xz* gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu.tar.xz ; tar xivf gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu.tar.xz`
* Manually download the SD card image for Xavier NX on your Linux host
* Grab the root partition of the image: ```
kpartx -av sd-blob.img
add map loop2p1 (253:1): 0 28260352 linear 7:2 870400
add map loop2p2 (253:2): 0 131072 linear 7:2 2048
add map loop2p3 (253:3): 0 131072 linear 7:2 133120
add map loop2p4 (253:4): 0 896 linear 7:2 264192
add map loop2p5 (253:5): 0 896 linear 7:2 266240
add map loop2p6 (253:6): 0 129024 linear 7:2 268288
add map loop2p7 (253:7): 0 1024 linear 7:2 397312
add map loop2p8 (253:8): 0 512 linear 7:2 399360
add map loop2p9 (253:9): 0 512 linear 7:2 401408
add map loop2p10 (253:10): 0 204800 linear 7:2 403456i
add map loop2p11 (253:11): 0 262144 linear 7:2 608256
dd if=/dev/mapper/loop2p1 of=/var/tmp/jetson-nx-rootfs.img bs=1M
kpartx -d sd-blob.img
```
* Extract that partition: `mkdir /mnt/jetson ; mount -o loop,ro /var/tmp/jetson-nx-rootfs.img /mnt/jetson ; cd /mnt/jetson ; tar -cvf /var/tmp/jetson-nx-rootfs-extract.tar . ; cd ; umount /mnt/jetson`
* Copy the tar file over to the docker image
* Extract rootfs in docker image: `mkdir /var/tmp/jetson ; cd /var/tmp/jetson ; tar xvf /var/tmp/jetson-nx-rootfs-extract.tar`
* Install qemu for arm: `apt install qemu-system-arm qemu-user-static ; cp /usr/bin/qemu-aarch64-static /var/tmp/jetson/usr/bin/ ; mount --bind /dev /var/tmp/jetson/dev ; rm /var/tmp/jetson/etc/resolv.conf  ;  cp /etc/resolv.conf /var/tmp/jetson/etc ; chroot /var/tmp/jetson /usr/bin/qemu-aarch64-static /usr/bin/apt update ; chroot /var/tmp/jetson /usr/bin/qemu-aarch64-static /usr/bin/apt -y install libceres-dev libeigen3-dev libfaac-dev libfaad-dev libmp3lame-dev libopencv-dev libopenexr-dev libpng-dev libportaudio-ocaml-dev librtmp-dev libturbojpeg0-dev libx264-dev ocl-icd-opencl-dev opencl-headers portaudio19-dev qt5-default qtmultimedia5-dev qttools5-dev libglew-dev libglfw3-dev libglm-dev libdrm-dev`
* Set env variables for cross compilation: `export TARGET_ROOTFS=/var/tmp/jetson ; export PATH=/var/tmp/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/:$PATH ; export CROSS_COMPILE=aarch64-linux-gnu-`

Edit: I stopped shortly after this - cmake doesn't detect cuda when running within the aarch64 root. Ugh.  :/

natecarlson, 2021-02-23:
Trying via the aarch64 docker image.. on the host, install qemu-user-static first. I am guessing this won't work for CUDA detection, as I'm betting the device isn't actually going to work.

* `docker run --gpus all -it nvcr.io/nvidia/l4t-base:r32.5.0 bash` -- start a bash shell, run 'uname -m' and make sure it returns aarch64
* Run the same junk as above, modifying to work on arm when needed.
```
apt update ; apt install gnupg
# Install cmake from static binary, PPA doesn't do arm64
cd /var/tmp/ ; wget 'https://github.com/microsoft/CMake/releases/download/v3.19.4268486/cmake-3.19.4268486-MSVC_2-Linux-arm.sh' ; bash /var/tmp/cmake-3.19.4268486-MSVC_2-Linux-arm.sh --skip-license --prefix=/usr/local
# Add CUDA repo
apt-key adv --fetch-keys  http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/7fa2af80.pub
bash -c 'echo "deb http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/sbsa /" > /etc/apt/sources.list.d/cuda.list'
bash -c 'echo "deb http://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu1804/sbsa /" > /etc/apt/sources.list.d/cuda_learn.list'
apt update
# Make the nvidia-utils version match the driver version on the Docker host
apt install cuda-toolkit-11-2
# Install the rest of the junk
apt update ; apt -y install git bison clang-tools doxygen flex libceres-dev libeigen3-dev libfaac-dev libfaad-dev libmp3lame-dev libopencv-dev libopenexr-dev libpng-dev libportaudio-ocaml-dev librtmp-dev libturbojpeg0-dev libx264-dev ninja-build ocl-icd-opencl-dev opencl-headers portaudio19-dev qt5-default qtmultimedia5-dev qttools5-dev swig wget xxd libglew-dev libglfw3-dev libglm-dev libdrm-dev
# Clone and build stitchEm
cd /var/tmp ; git clone https://github.com/stitchEm/stitchEm.git
# Modify cuda version from 10.2 to 11.2
echo 11.2 > /var/tmp/stitchEm/cuda.version
# Build
mkdir /var/tmp/stitchEm-build ; cd /var/tmp/stitchEm-build ; cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -G Ninja ../stitchEm
cd /var/tmp/stitchEm-build/ ; ninja
```

It's compiling, very slowly, since it's emulated.. will update if it errors or not.  :)

natecarlson, 2021-02-23:
Trying with BUILD_APPS=OFF..

```
rm -rfv /var/tmp/stitchEm-build/ ; mkdir /var/tmp/stitchEm-build ; cd /var/tmp/stitchEm-build ; cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -DBUILD_APPS=OFF -G Ninja ../stitchEm
ninja -j24
```

This ends up failing with:
```
[273/557] Building NVCC (Device) object lib/CMakeFiles/cud...rc/backend/cuda/cuda_compile_1_generated_boundsKernel.cu.o
FAILED: lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/cuda_compile_1_generated_boundsKernel.cu.o
cd /var/tmp/stitchEm-build/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda && /usr/local/bin/cmake -E make_directory /var/tmp/stitchEm-build/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/. && /usr/local/bin/cmake -D verbose:BOOL=OFF -D build_configuration:STRING=Debug -D generated_file:STRING=/var/tmp/stitchEm-build/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/./cuda_compile_1_generated_boundsKernel.cu.o -D generated_cubin_file:STRING=/var/tmp/stitchEm-build/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/./cuda_compile_1_generated_boundsKernel.cu.o.cubin.txt -P /var/tmp/stitchEm-build/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/cuda_compile_1_generated_boundsKernel.cu.o.Debug.cmake
nvcc fatal   : Unsupported gpu architecture 'compute_30'
CMake Error at cuda_compile_1_generated_boundsKernel.cu.o.Debug.cmake:216 (message):
  Error generating
  /var/tmp/stitchEm-build/lib/CMakeFiles/cuda_compile_1.dir/src/backend/cuda/./cuda_compile_1_generated_boundsKernel.cu.o
```
..which is expected, as Cuda 11.2 doesn't support that compute version anymore. Note that this shouldn't be an issue on an actual Jetson device with Cuda 10.2.

Removed compute_30 line from cmake/CUDA.cmake, started build again.

This resulted in a successful build of videostitch-cmd.. if it works or not is a completely different question, of course.  :)

2018boyan, 2021-02-24:
Thank you very much. I will try these useful suggestions one by one.

natecarlson, 2021-02-24:
> Thank you very much. I will try these useful suggestions one by one.

What's your end goal here? Do you want a working UI, or do you prefer the command line version?

My comments were kind of a stream of consciousness thing; if CLI gives you what you need I can get you a single list of commands that should work. If you need the GUI I'll need to play around with dependencies some more; in theory since it's working on x64 it should be possible for it to work on arm64 also, as they are the same version of the same os. (Unless, of course, the old OpenGL profiles just aren't available in the same version of the library on arm - certainly possible.)

Regardless the OpenGL profile will need to be modernized at some point!

natecarlson, 2021-02-24:
@2018boyan to confirm, you're continuing the work started in #129, correct? If so, can that ticket be closed as work is continuing here?

2018boyan, 2021-02-24:
I want a working UI actually but getting a command line version is also better than now. Thank you a lot for helping.

2018boyan, 2021-02-24:
Yes, it was published by my predecessor and could not update. I will continue this work.

natecarlson, 2021-02-24:
> I want a working UI actually but getting a command line version is also better than now. Thank you a lot for helping.

OK - yeah, that seems like a reasonable place to start!  :)

I would start with a freshly-imaged SD card (a new card, or else you can create an image of the card before imaging it to save the work that has already been done, just in case!), as it looks like you've got a custom ffmpeg/etc install that should not be necessary. I believe that CUDA 10.2 with development libraries are included by default (I haven't yet worked on a real Jetson device!); if not, go ahead and install them.

Then, try the following.. I've commented out the stuff related to CUDA 11.2; if the 10.2 dev libraries aren't available feel free to give that a try too.

```
# Install GnuPG for apt signing, if not already there.
sudo apt update ; sudo apt install gnupg
# Install static version of cmake to /usr/local - version in repository is not new enough.
cd /var/tmp/ ; wget 'https://github.com/microsoft/CMake/releases/download/v3.19.4268486/cmake-3.19.4268486-MSVC_2-Linux-arm.sh' ; sudo bash /var/tmp/cmake-3.19.4268486-MSVC_2-Linux-arm.sh --skip-license --prefix=/usr/local
# Not installing CUDA 11.2, as 10.2 _should_ already be on the system.
# apt-key adv --fetch-keys  http://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64/7fa2af80.pub
# bash -c 'echo "deb http://developer.downloa-y d.nvidia.com/compute/cuda/repos/ubuntu1804/sbsa /" > /etc/apt/sources.list.d/cuda.list'
# bash -c 'echo "deb http://developer.download.nvidia.com/compute/machine-learning/repos/ubuntu1804/sbsa /" > /etc/apt/sources.list.d/cuda_learn.list'
# apt update
# apt -y install cuda-toolkit-11-2
# Install build dependencies, minus cmake (installed from static binary above)
sudo apt update ; sudo apt -y install git bison clang-tools doxygen flex libceres-dev libeigen3-dev libfaac-dev libfaad-dev libmp3lame-dev libopencv-dev libopenexr-dev libpng-dev libportaudio-ocaml-dev librtmp-dev libturbojpeg0-dev libx264-dev ninja-build ocl-icd-opencl-dev opencl-headers portaudio19-dev qt5-default qtmultimedia5-dev qttools5-dev swig wget xxd libglew-dev libglfw3-dev libglm-dev libdrm-dev
# Clone stitchEm repository
cd /var/tmp ; git clone https://github.com/stitchEm/stitchEm.git
# Do _not_ modify the CUDA version to 11.2 - only do this if 10.2 doesn't work and you install 11.2 instead..
# echo 11.2 > /var/tmp/stitchEm/cuda.version
# Create build directory; note any CMAKE failures!  This disables the GUI apps.
mkdir /var/tmp/stitchEm-build ; cd /var/tmp/stitchEm-build ; cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -DBUILD_APPS=OFF -G Ninja ../stitchEm
# Perform the build
cd /var/tmp/stitchEm-build/ ; ninja
```

2018boyan, 2021-02-24:
With 
cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -DBUILD_APPS=OFF -G Ninja ../stitchEm

It shows a new error:
[270/557] Building CXX object IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o
FAILED: IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o 
/usr/bin/c++   -I/home/nvidia/stitchEm/lib/include -Igenerated_headers -I/home/nvidia/stitchEm/IO/src/common/format/include -I/home/nvidia/stitchEm/IO/src/common/format/../include -Wall -Wextra -Wno-missing-field-initializers -g -fPIC   -Wno-deprecated-declarations -std=gnu++14 -MD -MT IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o -MF IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o.d -o IO/src/common/format/CMakeFiles/format_cuda_ON.dir/src/muxer.cpp.o -c /home/nvidia/stitchEm/IO/src/common/format/src/muxer.cpp
/home/nvidia/stitchEm/IO/src/common/format/src/muxer.cpp: In constructor ‘VideoStitch::Output::Muxer::Muxer(size_t, const string&, std::vector<VideoStitch::AVEncoder>&, const AVDictionary*)’:
/home/nvidia/stitchEm/IO/src/common/format/src/muxer.cpp:98:51: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
       if (avcodec_parameters_from_context(stream->codecpar, encoder.codecContext) < 0) {
                                                   ^~~~~~~~
                                                   codec
/home/nvidia/stitchEm/IO/src/common/format/src/muxer.cpp:98:11: error: ‘avcodec_parameters_from_context’ was not declared in this scope
       if (avcodec_parameters_from_context(stream->codecpar, encoder.codecContext) < 0) {
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nvidia/stitchEm/IO/src/common/format/src/muxer.cpp:98:11: note: suggested alternative: ‘avcodec_free_context’
       if (avcodec_parameters_from_context(stream->codecpar, encoder.codecContext) < 0) {
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           avcodec_free_context
/home/nvidia/stitchEm/IO/src/common/format/src/muxer.cpp: In member function ‘void VideoStitch::Output::Muxer::writeHeader()’:
/home/nvidia/stitchEm/IO/src/common/format/src/muxer.cpp:170:34: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
       if (formatCtx->streams[i]->codecpar) {
                                  ^~~~~~~~
                                  codec
/home/nvidia/stitchEm/IO/src/common/format/src/muxer.cpp:171:72: error: ‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
         auto codecDesc = avcodec_descriptor_get(formatCtx->streams[i]->codecpar->codec_id);
                                                                        ^~~~~~~~
                                                                        codec
[271/557] Building CXX object lib/src/epipolar/CMakeFiles/epipolar_CUDA.dir/epipolarCurvesAlgorithm.cpp.o
ninja: build stopped: subcommand failed.

It looks like caused by a old ffmpeg version.  But I updated my ffmpeg in yesterday.

nvidia@jetson-0423718018083:~/stitchEm-build$ ffmpeg
ffmpeg version 4.3.2 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
  configuration: --enable-shared --prefix=/usr/local/ffmpeg
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
Hyper fast Audio and Video encoder
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'

natecarlson, 2021-02-24:
As I noted above, please go back to a brand new image first -- the version of ffmpeg shipped with the OS should work fine.

2018boyan, 2021-02-24:
Ok, I get it and will flash xavier.

w-m, 2021-02-24:
As suggested in https://github.com/stitchEm/stitchEm/issues/129#issuecomment-747419066, you can set the target arch by something like `-DCUDA_TARGET_ARCH=72` in the CMake invocation. Replace 72 with whatever target arch is your target computer. Really no need to build all the CUDA archs, if you plan to use the build on a single device, as it takes most of the compilation time.

2018boyan, 2021-02-24:
[image]

Does that mean success? It is the first time that get [557/557] :) Thank you!

natecarlson, 2021-02-24:
Yup! The binaries should be under the bin directory.

2018boyan, 2021-02-24:
Thank you! I also do cmake about a version with app. I just add #inclue<GL/glu.h> on every error hpp and seems success.
[image]

But when I find VahanaVR and try to ./VahanaVR or just type VahanaVR:

nvidia@jetson-0423718018083:~/stitchEm-build/bin/x64/debug$ VahanaVR
bash: VahanaVR: command not found
nvidia@jetson-0423718018083:~/stitchEm-build/bin/x64/debug$ ./VahanaVR
./VahanaVR: error while loading shared libraries: libnvidia-ml.so.1: cannot open shared object file: No such file or directory

[image]

How to use it or fix this error? Thanks!

natecarlson, 2021-02-24:
Awesome!

It appears that NVML is not available on the Jetson:
https://forums.developer.nvidia.com/t/libnvidia-ml-so-and-nvml-h-support/83005/2
https://github.com/Syllo/nvtop/issues/54

..so, yeah, looks like the UI isn't going to work on these, at least not without some modifications.  :/

natecarlson, 2021-02-24:
It appears that NVML is used for more than just the UI; it's the library used to monitor the hardware/etc.

However, good news! It looks like the definitions to disable NVML for Tegra devices already exists; no idea how tested it is etc.. but worth a try.

Please start from the cmake point again, and add:
`-DTEGRA_DEMO`
..to your build options. IE:
`cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -DBUILD_APPS=OFF -G Ninja ../stitchEm`

..and then rebuild.

2018boyan, 2021-02-25:
Thanks. I just use the following command and meet a new type of error.

nvidia@jetson-0423718018083:~/stitchEm-build$ cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -DTEGRA_DEMO=ON -fPIC -G Ninja ../stitchEm

-- Qt minimum supported version: 5.6
-- Qt FOUND: 5.9.5
-- Studio version: Studio-v2.4.0-win-p1-53-g7bb49a3-master.2021-01-04
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GLOG
    linked by target "videostitch_cuda" in directory /home/nvidia/stitchEm/lib
libv4l2
    linked by target "rtmpPlugin" in directory /home/nvidia/stitchEm/IO/src/rtmp

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/stitchEm-build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/stitchEm-build/CMakeFiles/CMakeError.log".

natecarlson, 2021-02-25:
I make absolutely no guarantees that this will help - but try making the following changes..

```
nc@testgpu02:~/tmp/stitchem/stitchEm$ git diff
diff --git a/IO/src/rtmp/CMakeLists.txt b/IO/src/rtmp/CMakeLists.txt
index 6e375f4..6673401 100644
--- a/IO/src/rtmp/CMakeLists.txt
+++ b/IO/src/rtmp/CMakeLists.txt
@@ -15,7 +15,7 @@ if(APPLE OR (NOT GPU_BACKEND_CUDA))
   set(RTMP_NVENC OFF CACHE BOOL "" FORCE)
 elseif(TEGRA_DEMO)
   set(RTMP_NVENC OFF CACHE BOOL "" FORCE)
-  set(RTMP_NVENC_M ON CACHE BOOL "" FORCE)
+  #set(RTMP_NVENC_M ON CACHE BOOL "" FORCE)
 endif()

 if(LINUX AND (NOT TEGRA_DEMO))
diff --git a/lib/CMakeLists.txt b/lib/CMakeLists.txt
index 38c88db..35eb0cd 100644
--- a/lib/CMakeLists.txt
+++ b/lib/CMakeLists.txt
@@ -732,9 +732,9 @@ if(APPLE)
 elseif (NOT ANDROID)
   vs_lib_link_libraries("PUBLIC_COMMON" Threads::Threads)
 endif(APPLE)
-if(TEGRA_DEMO OR WINDOWS)
-    vs_lib_link_libraries("PUBLIC_COMMON" ${GLOG})
-endif()
+#if(TEGRA_DEMO OR WINDOWS)
+#    vs_lib_link_libraries("PUBLIC_COMMON" ${GLOG})
+#endif()
 vs_lib_link_libraries("COMMON" ${OpenCV_LIBRARIES})
 vs_lib_link_libraries("COMMON" ${PNG_LIBRARY} ${CERES_LIBS} ${HID} ${R8B_LIBRARY})
 vs_lib_link_libraries("COMMON" o2b)
```

I don't have a Xavier (or Jetson) on hand, so it's hard to make sure this stuff works, but hopefully we can keep stumbling through it!

natecarlson, 2021-02-25:
Ugh, doing a compile with those options on x64, looks like further issues:
```
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuGLMapBufferObject_v2'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuGLGetDevices_v2'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuMemcpy'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuGLUnregisterBufferObject'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuGLCtxCreate_v2'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuGLRegisterBufferObject'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuCtxPushCurrent_v2'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuCtxPopCurrent_v2'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuCtxGetDevice'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuGLUnmapBufferObject'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuPointerGetAttribute'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuCtxDestroy_v2'
/usr/bin/ld: bin/x64/debug/libvideostitch_cuda.so: undefined reference to `cuCtxGetCurrent'
```

There are probably references that need to be fixed in the codebase too.

natecarlson, 2021-02-25:
This hack may solve that issue; note that I only tested the build on x64 with BUILD_APPS=OFF, and I didn't test the resulting binaries at all.. but give it a try.
```
diff --git a/cmake/CUDA.cmake b/cmake/CUDA.cmake
index 3baa062..e5cb91a 100755
--- a/cmake/CUDA.cmake
+++ b/cmake/CUDA.cmake
@@ -33,14 +33,10 @@ endif(WINDOWS)

 if(LINUX)
   set(CUDA_PROPAGATE_HOST_FLAGS "FALSE")
-  if(TEGRA_DEMO)
-    set(LINUX_CUDA_PATH ${CUDA_TOOLKIT_TARGET_DIR}/lib)
-    find_library(GLEW_LIBRARIES GLEW NO_CMAKE_FIND_ROOT_PATH)
-    find_library(OPENGL_LIBRARIES GL NO_CMAKE_FIND_ROOT_PATH)
-  else()
-    find_package(GLEW)
-    find_library(CUDA cuda PATHS "${LINUX_CUDA_PATH}/stubs")
-    find_library(CUDART cudart PATHS ${LINUX_CUDA_PATH})
+  find_package(GLEW)
+  find_library(CUDA cuda PATHS "${LINUX_CUDA_PATH}/stubs")
+  find_library(CUDART cudart PATHS ${LINUX_CUDA_PATH})
+  if(NOT TEGRA_DEMO)
     find_library(NVML nvidia-ml PATHS "${LINUX_CUDA_PATH}/stubs")
   endif()
   find_library(CUVID nvcuvid)
```

2018boyan, 2021-02-26:
Thank you very much for your warm help! I am so sorry that did not do it yesterday for going out. Do you mean that change the CUDA.cmake and re-make again?

I do a change on CUDA.cmake as following lines:

if(LINUX)
  set(CUDA_PROPAGATE_HOST_FLAGS "FALSE")
  find_package(GLEW)
  find_library(CUDA cuda PATHS "${LINUX_CUDA_PATH}/stubs")
  find_library(CUDART cudart PATHS ${LINUX_CUDA_PATH})
  if(NOT TEGRA_DEMO)
    find_library(NVML nvidia-ml PATHS "${LINUX_CUDA_PATH}/stubs")
  endif()
  find_library(CUVID nvcuvid)
endif(LINUX)

and run:
nvidia@jetson-0423718018083:~/stitchEm-build$ cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -DTEGRA_DEMO=ON -fPIC -G Ninja ../stitchEm

still get same error:
-- Qt minimum supported version: 5.6
-- Qt FOUND: 5.9.5
-- Studio version: Studio-v2.4.0-win-p1-53-g7bb49a3-master.2021-01-04
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GLOG
    linked by target "videostitch_cuda" in directory /home/nvidia/stitchEm/lib
libv4l2
    linked by target "rtmpPlugin" in directory /home/nvidia/stitchEm/IO/src/rtmp

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/stitchEm-build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/stitchEm-build/CMakeFiles/CMakeError.log".

natecarlson, 2021-02-26:
Sorry that it took me awhile to get back to you here!

You also need to make the set of changes from the previous message:
https://github.com/stitchEm/stitchEm/issues/132#issuecomment-785548480

Once you make those changes too, if it still gives you the same error, try removing the contents of your build directory and try again.

2018boyan, 2021-02-27:
No worries. You really help me quite a lot. Thank you and It works!
New issue is outdated CUDA driver. My CUDA version is 10.2, should I update it? Which version is better?

[image]

w-m, 2021-02-27:
Congrats on getting the full build working!

I will suggest again to specify the exact CUDA version of Xavier (see https://github.com/stitchEm/stitchEm/issues/132#issuecomment-784986953) - or is that producing errors in CMake?

If you've done that and it still produces the same error on launch, try printing the results of this check that produces the error:
https://github.com/stitchEm/stitchEm/blob/master/discovery/src/device.cpp#L443-L447

Is `result == cudaErrorInsufficientDriver` triggered, or `driverVersion < frameworkVersion`?

2018boyan, 2021-03-01:
Thank you! Sorry for late reply.

I add -DCUDA_TARGET_ARCH=72 now, but still meet error.

nvidia@jetson-0423718018083:~$ cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -DTEGRA_DEMO=ON -DCUDA_TARGET_ARCH=72 -fPIC -G Ninja ../stitchEm

And then I print these information as below:
nvidia@jetson-0423718018083:~/stitchEm-build/bin/x64/debug$ ./VahanaVR 
cudaErrorInsufficientDriver
driverVersion 10000
frameworkVersion 2082101488
Host allocated memory:
Device allocated memory:
Host CUDA allocated memory:
Device CUDA allocated memory:

w-m, 2021-03-01:
Update the GPU driver on the device to support at least the CUDA SDK version that you're using to build the software.

As I understand, you're using CUDA 10.2 SDK to build stitchEm. But `cudaDriverGetVersion` reports support only for up to CUDA 10.0 (and `cudaRuntimeGetVersion` produces a garbage value). So the driver seems to be older than the installed SDK.

2018boyan, 2021-03-02:
Thank you. It is a little bit hard to upgrade Nvidia driver on Xavier, could I just use CUDA 10.0 to build stitchEm?

natecarlson, 2021-03-02:
If it's available on the system, sure.

You could also try just commenting out the return in the check and see if it actually still works..

w-m, 2021-03-02:
I confirm that you can use CUDA 10.0 to build stitchEm.

2018boyan, 2021-03-04:
Thanks for you help. Sorry for late response. Rebuild xavier environment costs quite a lot time. On CUDA-10, It seems success.
[image]

This time, I could open vahanaVR, but the new problem is that I could not open my camera.
[image]
[image]
[image]

This is my cmake
nvidia@jetson-0423718018083:~/stitchEm-build$ cmake -DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -DTEGRA_DEMO=ON -DCUDA_TARGET_ARCH=72 -fPIC -G Ninja ../stitchEm
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- GPU_BACKEND_CUDA='ON'
-- GPU_BACKEND_OPENCL='OFF'
-- USE_DELAY_LOAD=OFF
-- System: LINUX
-- CMake Build Type -- Debug
-- Performing Test HAVE_FLAG_ADDRESS_SANITIZER
-- Performing Test HAVE_FLAG_ADDRESS_SANITIZER - Failed
-- Performing Test HAVE_FLAG_SANITIZE_ADDRESS
-- Performing Test HAVE_FLAG_SANITIZE_ADDRESS - Success
-- Performing Test HAVE_FLAG_LEAK_SANITIZER
-- Performing Test HAVE_FLAG_LEAK_SANITIZER - Success
-- Performing Test HAVE_FLAG_SANITIZE_THREAD
-- Performing Test HAVE_FLAG_SANITIZE_THREAD - Success
-- Performing Test HAVE_FLAG_SANITIZE_UNDEFINED_BEHAVIOR
-- Performing Test HAVE_FLAG_SANITIZE_UNDEFINED_BEHAVIOR - Success
-- CUDA VERSION: 10.0
-- CUDA PATH: /usr/local/cuda-10.0
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-10.0 (found suitable exact version "10.0") 
-- Found OpenGL: /usr/lib/aarch64-linux-gnu/libOpenGL.so   
-- Found GLEW: /usr/include (found version "2.0.0") 
-- CUDA: /usr/lib/aarch64-linux-gnu/libcuda.so
-- CUDART: /usr/local/cuda-10.0/lib64/libcudart.so
-- CUDA_LIBRARIES: /usr/local/cuda-10.0/lib64/libcudart_static.a;-lpthread;dl;/usr/lib/aarch64-linux-gnu/librt.so
-- NVCC arch flags: -gencode=arch=compute_72,code="sm_72"
-- Looking for CL_VERSION_2_2
-- Looking for CL_VERSION_2_2 - found
-- Found OpenCL: /usr/lib/aarch64-linux-gnu/libOpenCL.so (found version "2.2") 
-- Found BISON: /usr/bin/bison (found suitable version "3.0.4", minimum required is "3.0") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- System: LINUX
-- Found OpenCV: /usr (found version "3.3.1") 
-- --- VideoStitch-lib dependencies ---
-- Found OpenEXR: /usr/include;/usr/include/openexr  
-- OpenEXR libs: /usr/lib/aarch64-linux-gnu/libHalf.so;/usr/lib/aarch64-linux-gnu/libIex.so;/usr/lib/aarch64-linux-gnu/libImath.so;/usr/lib/aarch64-linux-gnu/libIlmImf.so;/usr/lib/aarch64-linux-gnu/libIlmThread.so
-- OpenEXR include: /usr/include;/usr/include/openexr
-- PNG: /usr/lib/aarch64-linux-gnu/libpng.so
-- R8B: r8b
-- OpenCV libs: opencv_calib3d;opencv_core;opencv_dnn;opencv_features2d;opencv_flann;opencv_highgui;opencv_imgcodecs;opencv_imgproc;opencv_ml;opencv_objdetect;opencv_photo;opencv_shape;opencv_stitching;opencv_superres;opencv_video;opencv_videoio;opencv_videostab
-- CERES LIBS: general;/usr/lib/libceres.so
-- 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components: doxygen missing components: dot
-- Found CUDA: /usr/local/cuda-10.0 (found version "10.0") 
-- VideoStitch I/O libs configured to build: avPlugin;jpgPlugin;pamPlugin;pngPlugin;rawPlugin;tiffPlugin;exrPlugin;v4l2Plugin;portaudioPlugin;rtmpPlugin
-- Qt minimum supported version: 5.6
-- Qt FOUND: 5.9.5
-- Studio version: Studio-v2.4.0-win-p1-53-g7bb49a3-master.2021-01-04
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nvidia/stitchEm-build

w-m, 2021-03-15:
Very cool that you got it running on the device! I am not sure how I can support you in debugging the crash from afar.

I'd make sure that stitching procedural inputs is working, just showing the numbers counting upwards in the panoramic output. Should that not be working, I'd check out these OpenGL shader compilation errors.

Then I'd try connecting the camera using a Vahana VR built on x86 and Desktop, until that works properly. Network inputs (https://github.com/stitchEm/stitchEm/pull/75) and outputs (https://github.com/stitchEm/stitchEm/issues/131) may need some bug fixing / adaptation to current ffmpeg version settings.

If the inputs and outputs are working on Desktop, and the stitching with the procedural inputs is working on the device, I'd try to make everything work on the device.

As a tip, it's possible to debug Vahana VR using CMake-suppoting IDEs. On Linux it works quite well to build and debug the project with Qt Creator.

2018boyan, 2021-03-18:
Thank you, I get it. Because our camera could only work on Xavier, maybe using Qt to debug is a good way. How to run stitchEm on Qt creator, should I build a new project and put all (.cpp and .hpp) into it?

w-m, 2021-03-18:
Qt Creator supports CMake directly. Just open the root CMakeLists.txt in Qt Creator, it will create a new project. Under "Projects" -> "[Your Default Kit]" -> "Build" you can then set the CMake parameters you were using above, either manually the table or by putting them in "Initial CMake parameters".

Just put `-DGPU_BACKEND_CUDA=ON -DGPU_BACKEND_OPENCL=OFF -DRTMP_NVENC=OFF -DTEGRA_DEMO=ON -DCUDA_TARGET_ARCH=72 -fPIC` there, not the Ninja generator option, and not the location of the source folder. The Ninja generator can be configured with the compiler and debugger in Preferences -> Kits. But it is not required, you can also use the usually preconfigured `make`.

2018boyan, 2021-03-19:
Thanks for your sincere help! I will try and hope to fix this issue.