stitchEm/stitchEm #133: Apple M1 Silicon Version

uwe22, 2021-02-27:
Is it possible to get a version for M1 Silicon.
Studio v2.4.0. hangs on Big Sur with M1 Silicon.
Many thanks for help.

w-m, 2021-02-28:
I noted that Studio hangs on Big Sur with Intel processors as well, see #134. It's probably helpful to fix this first, before working on Apple Silicon.

I don't have access to an Apple Silicon machine, so I cannot provide binaries for that platform. If you wish to sponsor an Apple Silicon machine for stitchEm development, please let me know :).

Otherwise I'd suggest you build it yourself, it is quite straightforward to build on macOS: 

- using MacPorts dependencies: https://github.com/stitchEm/stitchEm/blob/master/doc/BUILD.md#using-macports
- or using Homebrew: https://github.com/stitchEm/stitchEm/blob/master/.github/workflows/build.yml#L27-L36

The first step to an Apple Silicon binary would be to test, whether all these dependencies of either package manager install natively!

w-m, 2021-03-03:
I've created a new release for Big Sur, v2.4.1: https://github.com/stitchEm/stitchEm/releases/tag/Studio-v2.4.1-macOS

It works well for me on two Intel machines. Please try it out @uwe22, and let me know, whether it works on Apple Silicon as well!

uwe22, 2021-03-05:
You are amazing. I made a quick test. I could reopen my old project well on Big Sur with M1. There was a short information that some settings are updated from my hardware. I could play back in the output tab and render a short sequence to mp4. 

[image]

[image]

w-m, 2021-03-15:
That's wonderful, I'm very glad to hear that! 😃 (Also pretty amazing engineering by Apple to make that work).

I'll leave the ticket open for possible native Apple Silicon support without Rosetta2 in the future.

LiveRock, 2023-03-25:
Hello
I followed the instructions but was stopped at the following when doing the build:

```
-- --- VideoStitch-lib dependencies ---
CMake Error at /opt/homebrew/Cellar/cmake/3.25.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenEXR (missing: OpenEXR_LIBRARY_DIR)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.25.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindOpenEXR.cmake:72 (find_package_handle_standard_args)
  lib/CMakeLists.txt:714 (find_package)
```

My commands:
`cmake -DCREATE_BOX_PACKAGE=OFF -DGPU_BACKEND_CUDA=OFF -DGPU_BACKEND_OPENCL=ON -DMACPORTS=OFF -DQt5_DIR=~/Qt/6.4.3/macos/lib/cmake -G Ninja ..`

I am using Qt6 as I can't find Qt5. I am also using a MacBook Pro m1pro

What did I do wrong here?

w-m, 2023-03-30:
OpenEXR installed from homebrew doesn't work in the latest version, see #147. You can disable that with `-DDISABLE_EXR=O
N` in the CMake call.

The code has not been updated to keep up with the latest API changes of the various libraries. Qt6 will not work, also current ffmpeg version won't work (#148).

Nobody is working on updating the code to use the newer libraries, but pull requests are welcome.