stitchEm/stitchEm #138: Rendering big images

jacktasak, 2021-03-23:
Hi!
I'm quite new to this stuff, but I foud it amazing to simplify my custom VR-generating process.
I quickly managed how to use this software, but when it comes to rendering I reach nothing but failure...
I want to render 11000x5500 output, and no matter what codec or output format I choose, there is render error.
I have quite strong machine - i7, RTX2060 with 32GB RAM. Below I attach log file. Could you help me?

Process is starting.
Process is now running.
Process started.
Found 1 device

Warning: Only one GPU found, ignoring -d.
Selected device 0
[videostitch-cmd] CUDA backend selected
Running on CUDA backend
Scanning 'C:/Program Files/stitchEm/Studio\core_plugins' for plug-ins.
Plugin: found potential plugin "jpg.dll"
Plugin: found potential plugin "png.dll"
Plugin: found potential plugin "tiffPlugin.dll"
Trying to load plugin C:\Program Files\stitchEm\Studio\core_plugins\jpg.dll.
Trying to load plugin C:\Program Files\stitchEm\Studio\core_plugins\png.dll.
Trying to load plugin C:\Program Files\stitchEm\Studio\core_plugins\tiffPlugin.dll.
3 plug-ins loaded from C:/Program Files/stitchEm/Studio\core_plugins.
Scanning 'C:/Program Files/stitchEm/Studio\core_plugins_CUDA' for plug-ins.
Plugin: found potential plugin "av_cuda.dll"
Trying to load plugin C:\Program Files\stitchEm\Studio\core_plugins_CUDA\av_cuda.dll.
1 plug-ins loaded from C:/Program Files/stitchEm/Studio\core_plugins_CUDA.
GPU device: Nvidia GeForce RTX 2060 (64 Bit GPU), global memory size: 6144 MB
Panorama size: 11000x5500
[Status] [audio pipeline configuration] Invalid configuration: No audio processor definition found
[avinput] Using a software decoder
[audiopipeline]: master is set to input "e:\test.mp4"
Will stitch 462 images.
[Status] [GPU] Out of resources: Out of resources. Reduce the project output size and close other applications to free up GPU resources.
[Status] [GPU] Out of resources: Could not allocate GPU memory. Reduce the project output size and close other applications to free up GPU memory.
: 54741 usec
Host allocated memory:
Device allocated memory:
Host CUDA allocated memory:
Input Frames: 0 used (68.34 MB max)
Panorama Frame: MEMORY LEAKING 173.09 MB used (346.18 MB max)
Device CUDA allocated memory:
Colorspace conversion: MEMORY LEAKING 174.56 MB used (349.12 MB max)
Panorama Frame: MEMORY LEAKING 174.56 MB used (174.56 MB max)
Remap Buffer: MEMORY LEAKING 230.79 MB used (461.58 MB max)
StitchOutput: MEMORY LEAKING 230.79 MB used (461.58 MB max)
Process isn't running.
Process exited normally. Exit code 1

w-m, 2021-03-23:
Several things to consider when rendering large outputs:

1) the GPU memory can be the limiting factor, as all of the stitching happens on the GPU, and the GPU memory can't be paged or swapped out. The 6 GB of your GPU should be enough to stitch 11k x 5k, unless you have a large number of gigantic inputs. Have a look at the GPU memory usage information on the bottom of the window. If you're getting close to using half the GPU memory, then use the Batch Stitcher for rendering: Process Tab -> Send to batch, and close Studio before hitting the "Stitch" button to free up GPU memory from having the project opened in Studio.

2) Outputs above 4096x2048 can't be encoded into h264 (#81). Choose MOV container ("Format"), and either Motion JPEG or ProRes as the Codec. (MPEG2 works above 4K, but I think not above 6k or 8k or something like that). If you desire another output format, you'll have to manually reencode the result, e.g. with ffmpeg.

3) Give both of the output dimensions a multiple of 16. Try 11008 x 5504 instead of 11000 x 5000. In theory every even value should work, but in practice some part of the pipeline, e.g. the encoder, might barf if it's not aligned to some block size, so go with the multiple of 16 to be safe.

jacktasak, 2021-03-23:
Thank you for the quick reply!
I did not have to check all your solutions, because the first one helped :) I checked the consumption of dedicated GPU memory and it did not exceed 3.9 of 6GB available. 
Regards,
Jacek

closed 2021-03-25 by w-m