stitchEm/stitchEm #142: Unable to launch videostitch-studio with offscreenAllocator

abhinavkshitij, 2021-04-23:
## Config: 
Linux x86_64, GPU: Internal (Matrox), Four NVIDIA Quadro RTX 8000; Monitor connected by VGA cable to internal graphics card, Nvidia to be used only for compute not rendering, i.e. off-screen rendering. Nvidia driver 465.xx.xx, CUDA 11.3; no Nvidia GL libraries installed; Mesa handles OpenGL. 

## STEPS:
1. Build Docker image from https://github.com/tlalexander/stitchEm/blob/master/docker/user_image.dockerfile
2. Run Docker container: docker run --gpus all -it -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix stitchem/stitchem-cuda-user:latest bash
3. Run /bin/videostitch-studio. GUI launches with warning QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-developer'. 
4. Open sample project '3_Cam_Sample_Project.ptv'. Downloaded from Google drive link. Rename to avoid whitespace in filenames. Changed absolute filepaths of MP4 files in .ptv file to their location within container filesystem.

## ERROR: 
[image]

## TRACEBACK:
FILE: /lib/src/backend/cuda/allocator.cpp
FUNCTION: Potential<PanoOpenGLSurface> OpenGLAllocator::createSpurceSurface(width, height)
LINE 201: FAIL_RETURN(CUDA_ERROR(cudaGraphicsGLRegisterImage(&image, texture, GL_TEXTURE_2D, cudaGraphicsRegisterFlagsSurfaceLoadStore)))

# Issue: VahanaVR defaults to OpenGLAllocator instead of using OffScreenAllocator. 

# Question: 
- Workaround: using videostitch-cmd -i 3_Cam_Sample_Project.ptv -d 0 -p <???> 
     -- What is the right plugin option here?
     -- Not passing -w flag in CLI to disable OpenGL window for display.  Will this force allocator.cpp to use OffAllocator?

# Test:  
 - Passes most of CUDA-based tests with ctest that require copying buffer to device and back.

w-m, 2021-04-26:
`-p` takes multiple directories, where the dynamic I/O plugins are placed. Have a look in the install folder, the names should be `CorePlugins` (for video/image I/O, shared Studio and Vahana VR) and `VahanaPlugins` for the network & capture card stuff.

`-- Not passing -w flag in CLI to disable OpenGL window for display. Will this force allocator.cpp to use OffAllocator?`

Yes that should be the case, unless some overlays are specified in the project file (https://github.com/stitchEm/stitchEm/blob/0f399501d41ab77933677f2907f41f80ceb704d7/lib/samples/sample-02-fullCmd/videostitch-cmd.cpp#L601-L619). "Overlays" in this context should mean inputs that are not true videos or captures, but generated content such as the frame counter input in Vahana VR.

abhinavkshitij, 2021-04-27:
Ok it seems to run now (audio disabled). The output shows CUDA selected as backend but the output lists 0 byte GPU memory usage. Does that mean it is not using CUDA? Also is the time taken for stitching slow? Plus how can I enable multiple GPUs for stitching? 

./videostitch-cmd -d 0 -i ../../../data/sample/3_Cam_Sample_Project.ptv -p core_plugins/
Found 4 devices

Selected device 0
[videostitch-cmd] CUDA backend selected
Running on CUDA backend
Panorama size: 4096x2048
[avinput] Using a software decoder
[avinput] Using a software decoder
[avinput] Using a software decoder
[Status] [audio pipeline configuration] Invalid configuration: No audio input  found
[Status] [audio pipeline configuration] Invalid configuration: No audio input  found
Will stitch 499 images.
[libavoutput] Creating encoder using h264
[audiopipeline] Add output vs-out
[ambisonic] Decoder bad coefficients given for stereo
Stitcher setup: 91 ms
Global computation time: 22527 ms
: 31979 usec
: 49174 usec
[libav-error] failed to rename "x264_2pass.log.temp" to "x264_2pass.log"
Output size: 37.20 MiB
Host allocated memory:
Device allocated memory:
Host CUDA allocated memory:
 Input Bounding boxes: 0 used (16.00 KB max)
 Input Frames: 0 used (142.38 MB max)
 Panorama Frame: 0 used (24.00 MB max)
Device CUDA allocated memory:
 Colorspace conversion: 0 used (24.00 MB max)
 ImageMapping: 0 used (94.92 MB max)
 Input Bounding boxes: 0 used (16.00 KB max)
 Input Frames: 0 used (35.60 MB max)
 Input Mask Interpolation: 0 used (64.00 MB max)
 Input mask interpolation: 0 used (31.64 MB max)
 MaskSetup: 0 used (7.91 MB max)
 Panorama Frame: 0 used (24.00 MB max)
 Remap Buffer: 0 used (64.00 MB max)
 Setup Buffer: 0 used (56.00 MB max)
 StitchOutput: 0 used (64.00 MB max)
 Transform: 0 used (288.14 KB max)
 Unpacking Input Frame: 0 used (35.60 MB max)
 Voronoi Mask Merger: 0 used (120.00 MB max)
 Warped Input Frame: 0 used (96.00 MB max)

w-m, 2021-04-29:
> Ok it seems to run now (audio disabled). The output shows CUDA selected as backend but the output lists 0 byte GPU memory usage. Does that mean it is not using CUDA?

The GPU memory usage is printed after the stitcher is shut down, it's a (somewhat crude) indicator of whether we're leaking any GPU memory. So seeing 0 Bytes used everywhere is good. Stitching still happens on the GPU, with the max memory usage in the parens.

> Also is the time taken for stitching slow?

~40 ms per frame seems ok to me. If you have a powerful GPU than probably the software decoding and encoding is the bottleneck, not the stitching computations.

> Plus how can I enable multiple GPUs for stitching?

Multi-GPU is not supported.

abhinavkshitij, 2021-05-03:
If videostitch-cmd takes only PTV files as inputs, which requires camera calibration data, can it be ever used in a real-time video-stitching application from the cmdline? How can videostitiching from moving cameras? The calibrationimport tool requires PTO file which I believe would require hugin to run side-by-side to constantly generate new calibration data and create new PTV input files to feed into videostitch-cmd. Is there a better way to livestitch from the cmdline (not GUI)?

w-m, 2021-05-03:
It is possible to perform a feature-point-based calibration with the command-line tool. `videostitch-cmd` will then just perform the calibration, write a `.ptv` file, and exit.

To perform the feature-based calibration on the command line, `videostitch-cmd` can be called with `--apply_algos` and a .json file describing the calibration you want to do. I don't think there's documentation in the repo on how the json file is supposed to look, but there are plenty of examples in the [cli_tests](https://github.com/stitchEm/stitchEm/tree/master/tests/cli_tests/etc) folder (`calibration_*.json`). I can try to dig up more documentation should you really be needing that feature.

The camera rig moving is not an issue, as long as the cameras stay fixed relative to each other. Usually you would calibrate your rig once at the start of your session, and stick with that calibration for the whole run.

It's possible to update the calibration from time to time, if needed, in the C++ stitching pipeline. Vahana VR supports that, and I think the Python stitching pipeline for the Orah 4i camera built on top with C++ bindings does support this as well. But when you are in stitching mode with `videostitch-cmd`, it loads a single calibration from the ptv, and stays with that calibration throughout the stitching run. You can not switch to another calibration in the middle of the run, when stitching from the command line.

The code only supports fixed or at least semi-fixed camera rigs. Handheld cameras where the cameras move relatively to each other all the time are not supported. (That would be a tricky problem, as the feature-based calibration runs for several seconds usually, thus can't be updated per-frame).

Please describe your use case in detail, then I can try to help you further.

abhinavkshitij, 2021-05-04:
The calibration_\*.json files have format that cannot be used as an input to videostitch-cmd with -i flag -- there are no "input", "pano" and "merge" fields. I suppose the calibration_\*.ptv files in the cli_test folder are the calibration output files. They contain the "input" So I created a JSON file 3_Cam_Sample_calibration.json from cli_tests/calibration_factory.json, changed tried the following:

`/videostitch-cmd -d 0 --apply_algos calibration -i ../../../data/sample/3_Cam_Sample_calibration.json -p core_plugins/`
`Found 4 devices`

`Selected device 0`
`[videostitch-cmd] CUDA backend selected`
`Running on CUDA backend`
`Error: No 'pano' entry in file: 3_Cam_Sample_calibration.json`
`[Status] [panorama configuration] Invalid configuration: Could not find 'merger' field`
`Error: Cannot parse algos config PTV file: calibration`
`cannot open calibration: No such file or directory`

That did not work. The calibration_*.ptv files contain the fields to input images for calibration, etc. The generated the API documentation from doxygen does not contain any additional material on JSON files for calibration. 

I have a test run of a few cameras that create a 360 view of an enclosed space (say a parking lot) and they move relatively to each other or one of them could get knocked down by wind or loose mounting. So definitely would require calibration from time to time as they pan and rotate around.

w-m, 2021-05-05:
The json file should be supplied as a parameter to `apply_algos`:

`videostitch-cmd --apply_algos <calibration.json> -i file_specifying_paths_to_videos.ptv`

 > So definitely would require calibration from time to time as they pan and rotate around.

Then, as pointed out above, you'll have to do some custom development. If you want to take a deep dive: in Vahana VR this is implemented roughly as following: there's [LiveStitcherController](https://github.com/stitchEm/stitchEm/blob/master/apps/src/videostitch-live-gui/src/videostitcher/livestitchercontroller.cpp) which keeps the state of the stitching pipeline in the Vahana VR GUI. A PanoDefinition is loaded from the .ptv file, and a stitcher is created in the C++ library libvideostitch, similar to the code in `videostitch-cmd`. The LiveStitcherController deals with the user triggering a calibration (which is implemented at [CalibrationOnlineAlgorithm](https://github.com/stitchEm/stitchEm/blob/master/lib/src/calibration/calibrationOnlineAlgorithm.cpp)). This will create an updated PanoDefintion. The updated definition with the new calibration is then handed to the stitcher pipeline through calling [updatePanorama](https://github.com/stitchEm/stitchEm/blob/master/lib/src/core/controller.cpp#L216).

abhinavkshitij, 2021-05-14:
I believe the calibration proceeds in multiple steps, where the output (camera parameters like - yaw, pitch) of one step in the ptv file is fed into the json file for the next step. There are several "_incremental" files available, which point to an incremental process. First I set the parameters to their default value in the JSON file to get an estimate of yaw, pitch in the PTV file. Then repeated it a few times, but found no change in the output despite trying multiple times. As a test case, I tried getting the same yaw, pitch, roll, etc as found in the 3_Camera example, but the final figures are nowhere near the ones seen in the PTV file of your example. So clearly I am missing something. How would one calibrate using the `videostitch-cmd` tool?

closed 2021-07-02 by abhinavkshitij